AltBeacon/android-beacon-library #185: Stackoverflow error

SergiuC10, 2015-05-18:
After updating the library.
I get Stackoverflow error at java.util.Collections

Checking DDMS shows that java.util.Collections are allocated when in writeToParcelable:

  at java.util.Collections$UnmodifiableCollection.iterator(Collections.java:952)	
  at java.util.Collections$UnmodifiableCollection$1.<init>(Collections.java:953)	
  at java.util.Collections$UnmodifiableCollection.iterator(Collections.java:952)	
  at java.util.Collections$UnmodifiableCollection$1.<init>(Collections.java:953)	
  at java.util.Collections$UnmodifiableCollection.iterator(Collections.java:952)	
  at java.util.Collections$UnmodifiableCollection$1.<init>(Collections.java:953)	
  at java.util.Collections$UnmodifiableCollection.iterator(Collections.java:952)	
  at org.altbeacon.beacon.Beacon.writeToParcel(Beacon.java:524)	
  at org.altbeacon.beacon.AltBeacon.writeToParcel(AltBeacon.java:115)	
  at android.os.Parcel.writeParcelable(Parcel.java:1285)	
  at android.os.Parcel.writeParcelableArray(Parcel.java:1984)	
  at org.altbeacon.beacon.service.RangingData.writeToParcel(RangingData.java:62)	
  at android.os.Parcel.writeParcelable(Parcel.java:1285)	
  at android.os.Parcel.writeValue(Parcel.java:1204)	
  at android.os.Parcel.writeArrayMapInternal(Parcel.java:618)	
  at android.os.Bundle.writeToParcel(Bundle.java:1692)	

It crashed in Emulator, also on Galaxy S5, 4.4.4

davidgyoung, 2015-05-18:
Can you please include the library version where you see this crash?

SergiuC10, 2015-05-18:
I am using latest sources from master branch

davidgyoung, 2015-05-19:
A similar issue was reported here:  http://stackoverflow.com/questions/29026487/stackoverflowerror-collectionsunmodifiablelist

Can you reproduce this error?  If so, can you please turn on debugging with beaconManager.setDebug(true), re-run and let me know what you see in LogCat before the crash takes place.

SergiuC10, 2015-05-19:
Yes, it crashes 10-15 minutes after I start the application.

Log: https://goo.gl/6fBXUB

davidgyoung, 2015-05-29:
A few questions:

1. How do you get this to crash in the emulator?  Do you use simulated beacons?

2. Looking at the log you attached, it appears the app with process id 11452 keeps running after the crash.  Does the app actually crash or does it keep running?

3. The log lines do not seem to match the latest source.  Can you tell me the git hash of the source version you are running?

PR #199 (2015-06-01): Fix stackoverflow from re-wrapping collections
Check to see if lists have already been wrapped as an unmodifiableList before re-wrapping them in the getter.  Fixes #185
closed 2015-06-01 by davidgyoung