AltBeacon/android-beacon-library #186: Positive 127 dB RSSI values returned in LeScanCallbacks

ccpmark, 2015-05-18:
We've been encountering a problem where our HTC One Mini phone (running Android 4.4.2) returns a positive 127 RSSI value fairly frequently from Android's LeScanCallback. The callback ordinarily returns RSSI values which correlate well with the true distance from the beacon, but occasionally returns a value of positive 127. 

I was able to trace this all the way back to Android's callback in this line: https://github.com/AltBeacon/android-beacon-library/blob/2d9e3ebeae4269af80a3b6a19a4bad657799c89a/src/main/java/org/altbeacon/beacon/service/scanner/CycledLeScannerForJellyBeanMr2.java#L77. This RSSI spike follows into the distance calculations, affecting all the ranging done through the moving averaging.

Would it be possible to put a catch for RSSI values of exactly 127? I'm sure this phone isn't the only one experiencing this weird glitch.

davidgyoung, 2015-05-18:
@ccpmark, we have had a similar report recently on Stackoverflow: http://stackoverflow.com/questions/30118991/rssi-returned-by-altbeacon-library-127-messes-up-distance

There is actually a fix for this in fbfe4bb289f1ea5b5afb7131e6f7141de31435c9 and I have rolled a beta release that you can try to see if it fixes the problem for you:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3-beta1

The way the fix works is that it will ignore any RSSI report of 127 for ranging purposes *except* the first report from the beacon (It can't really ignore the first report, because  at that point it is the only RSSI value it has).

davidgyoung, 2015-05-18:
Note that the Stackoverflow report was about a LG G3 S having this problem.  So there are now reports from at least two device manufacturers where this is happening.

closed 2015-05-19 by davidgyoung

mozeryansky, 2020-11-07:
The bluetooth standard defines the RSSI value of 127 to mean "none", where the device was unable to do an RSSI reading for that packet.