AltBeacon/android-beacon-library #191: AltBeacon Ranging and Monitoring even when bluetooth is off

ahmadshami, 2015-05-20:
I commented out the bluetooth check popup from the reference app, compiled and installed the app.
1. Launched the app
2. Clicked on Start Ranging button.
3. Nearby beacons are now being seen on the list.
4. Turned bluetooth off.

And surprisingly the ranging continues and does not stop.
How is that even possible? I double checked that the bluetooth is off and yet, the app keeps on ranging for nearby beacons.

Any thoughts?

davidgyoung, 2015-05-20:
Obviously bluetooth is not off.  :)  What model of device are you seeing this on?

ahmadshami, 2015-05-20:
Please have a look:
https://docs.google.com/file/d/0B2PYzXN1SmntUzFsQi03SDRpNmM/edit?usp=docslist_api

I am testing the app with Galaxy S4 running Android 5.0.1

ahmadshami, 2015-05-27:
any updates @davidgyoung?

davidgyoung, 2015-05-27:
I suspect the Bluetooth toggle on the settings screen on the phone is not accurately reflecting the internal state of the operating system.   You might test this hypothesis by adding code like below to your app, and logging the result periodically and seeing what it says when you turn bluetooth on and off:

BluetoothAdapter mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
if (mBluetoothAdapter.isEnabled()) {
  Log.d(TAG, "Bluetooth is enabled");
}
else {
  Log.d(TAG, "Bluetooth is disabled");
}

davidgyoung, 2015-06-01:
Actually there is another report of this happening on the Galaxy S5:

https://github.com/AltBeacon/android-beacon-library/issues/108

It seems that some Samsung devices do not actually disable Bluetooth if disabled by the settings UI if an app is already using it.

Two options:

1. Don't change anything.  Continue scanning for beacons even if the user turns it off on Samsung devices.  Document this as a known issue.
2. Investigate whether it is possible to query whether the user has asked to disable bluetooth on a Samsung device, and if so, stop scanning for beacons at the next checkpoint. 

It would also be possible to make switching behavior between 1 and 2 a configurable setting.

ahmadshami, 2015-06-03:
It makes sense now, thank you!

closed 2015-06-03 by ahmadshami