AltBeacon/android-beacon-library #192: Library is not working on Samsung Galaxy S5 (5.0)

antoniodisanto92, 2015-05-21:
Hello,

I'm trying to use this library with a Samsung Galaxy S5 but seems it doesn't work. I tried also the reference app and the SNAPSHOT library but still the same problem. The library correctly print - in logcat - the it finds in the debug level, but it print nothing in the Activity and it nether start the activity when an iBeacon enter in the range, nor notification. Seems it never enter in the "didEnterRegion" method.

The log still print correctly this:
`05-21 11:34:06.605    1423-1436/org.altbeacon.beaconreference D/BluetoothLeScanner﹕ onScanResult() - ScanResult{mDevice=7C:66:9D:7D:09:08, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=null, mManufacturerSpecificData={76=[2, 21, -21, -17, -48, -125, 112, -94, 71, -56, -104, 55, -25, -75, 99, 77, -11, 36, 0, 1, 0, 1, -59]}, mServiceData={}, mTxPowerLevel=0, mDeviceName=jaalee������������������������], mRssi=-82, mTimestampNanos=3224717421224}`
`05-21 11:34:06.605    1423-1433/org.altbeacon.beaconreference D/ScanRecord﹕ parseFromBytes`
`05-21 11:34:06.605    1423-1433/org.altbeacon.beaconreference D/ScanRecord﹕ first manudata for manu ID`

Any idea?

Thanks,
A.

antoniodisanto92, 2015-05-21:
Ok, i've fixed it. I had to put some custom params.

Sorry for the post!

A.

closed 2015-05-21 by antoniodisanto92

sandeepshabd, 2016-03-29:
How did it work ? I am getting the same error ?

davidgyoung, 2016-03-29:
@sandeepshabd, you may have better luck getting help by posting the details of your problem on a question on Stackoverflow.com including the type of beacon you are using, the mobile device model, the code you are using to set up beacon detection, and any log messages you get that are not as expected.