AltBeacon/android-beacon-library #194: detecting beacons after killing the application android

OmarMahdoui, 2015-05-26:
i'm using the android beacon library for detecting beacon while the app is killed . the problem is that every time i kill the app from the task switcher i have to connect/disconnect my device from the usb cable.
Is there any solution to let the application run after I killed without connect/disconnect every time?

davidgyoung, 2015-05-26:
There are two ways the library will re-launch an app:

1. On power connect/disconnect events
2. On scheduled intervals (5 min by default) based on an alarm.  

The second method is disabled by a known bug in the 2.1 release in Maven/JCenter, but has been fixed in this release:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3-beta4

Please try the above release and see if it works for you.

This may or may not work on all devices.  Some users report in the issue below that some Android Launchers do a force stop of an app when killed from the task switcher.  For ROMs that do this, the app cannot relaunch itself due to Android security restrictions.  There is a test app in the issue linked below you can try to see if your device is affected.

https://github.com/AltBeacon/android-beacon-library/issues/44

OmarMahdoui, 2015-05-26:
Can you help me with the second point:
  On scheduled intervals (5 min by default) based on an alarm.
How to use this feature.

davidgyoung, 2015-05-26:
This is automatic in the 2.3-beta4 version of the library provided you use the RegionBootstrap.   If the app is terminated due to memory pressure or killing from the task switcher, the alarm goes off after five minutes and the app restarts scanning for beacons.    See here on how to use the RegionBootstrap: http://altbeacon.github.io/android-beacon-library/background_launching.html

Now, the question remains whether your device has a customized launcher that does a force stop of an app when killed by the task switcher.  If it does, this relaunching technique may not work at all.

OmarMahdoui, 2015-05-26:
Ok i just try it and all work fine on my Galaxy Grand Prime 4.4.4, my device does not force stop of an app when killed by the task switcher.
Is it possible to customize the Alarm time from 5 minutes to another value?

davidgyoung, 2015-05-26:
The interval is currently configurable by setting the parameters above, but has a limit of being no more often than five minutes.  In other words, you can configure it to happen less often than 5 minutes, but not more.  

The theory behind this limit is that if a background app is killed due to low memory pressure, five minutes gives time for the high memory operation to complete before the app is restarted and starts taking up memory again.  

There is a dedicated issue for this configurability here:  https://github.com/AltBeacon/android-beacon-library/issues/179  Feel free to add you own views.

closed 2015-05-29 by davidgyoung