AltBeacon/android-beacon-library #196: org.altbeacon.beacon.BeaconManager$ServiceNotDeclaredException

Blackimp, 2015-05-27:
I get the following error when i run a Unit Test in my project:

The BeaconService is not properly declared in AndroidManifest.xml.  If using Eclipse, please verify that your project.properties has manifestmerger.enabled=true
org.altbeacon.beacon.BeaconManager$ServiceNotDeclaredException: The BeaconService is not properly declared in AndroidManifest.xml.  If using Eclipse, please verify that your project.properties has manifestmerger.enabled=true

I am using Android Studio with Gradle which has no project.properties file.
Where do I have to set "manifestmerger.enabled=true" instead?

Regards
Julian

Blackimp, 2015-05-27:
This is my current AndroidManifest.xml:

<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.xxx.xxx">

    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />

    <permission
        android:name="com.xxx.xxx.permission.C2D_MESSAGE"
        android:protectionLevel="signature" />

    <uses-permission android:name="com.google.android.c2dm.permission.RECEIVE" />
    <uses-permission android:name="com.xxx.xxx.permission.C2D_MESSAGE" />

    <application
        android:name=".xxxApplication"
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:theme="@style/AppTheme">
        <activity
            android:name=".MainActivity"
            android:label="@string/app_name"
            android:screenOrientation="portrait">
            <intent-filter>
                <action android:name="android.intent.action.MAIN" />

                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>
        <activity
            android:name=".LoginActivity"
            android:label="@string/title_activity_login" />

        <receiver
            android:name="org.jboss.aerogear.android.unifiedpush.gcm.AeroGearGCMMessageReceiver"
            android:permission="com.google.android.c2dm.permission.SEND">
            <intent-filter>
                <action android:name="com.google.android.c2dm.intent.RECEIVE" />
                <action android:name="com.google.android.c2dm.intent.REGISTRATION" />

                <category android:name="com.xxx.xxx.android.test.xxxApp" />
            </intent-filter>

            <meta-data
                android:name="DEFAULT_MESSAGE_HANDLER_KEY"
                android:value="com.xxx.xxx.push.AGMessageListener" />
        </receiver>
    </application>
</manifest>

Blackimp, 2015-05-27:
I just read, that Android Studio automatically merges the AndroidManifest.xml from the Android Beacon Library into my own AndroidManifest.xml, which makes it unnecessary to declare the Beacon Services in my AndroidManifest.xml.

Any idea what could cause this issue?

davidgyoung, 2015-05-27:
Hi, @Blackimp.   I'm happy to help project-specific setup issues with questions posted on Stackoverflow.com, but this forum is reserved for feature requests or repeatable bugs found with the library itself.  

In the interests of time, I will note that a good place to start troubleshooting manifest merging issues is by looking in this file in your build output directory:

      app/build/outputs/apk/manifest-merger-release-report.txt

Please take a look for any errors in that file and post a question on Stackoverflow.com if you need more help.

closed 2015-05-27 by davidgyoung

Blackimp, 2015-05-28:
Hi, @davidgyoung. Thanks for your tip with the release-report.
I moved my question to Stackoverflow: http://stackoverflow.com/q/30499138/4361588.