AltBeacon/android-beacon-library #197: NullPointerException in Samsung Galaxy Note 4 - Android 5.0.1

marcelorbenites, 2015-05-27:
Hi,

I have recurrent NullPointerExceptions with the following stack trace using Samsung Galaxy Note 4 with Android 5.0.1 using version 2.1.3 of android-beacon-library. There was not problem before the update to version 5.0.1.

    java.lang.NullPointerException: Attempt to invoke interface method 'boolean android.bluetooth.IBluetooth.isOffloadedFilteringSupported()' on a null object reference
       at android.bluetooth.BluetoothAdapter.isOffloadedFilteringSupported(BluetoothAdapter.java:1231)
       at android.bluetooth.le.BluetoothLeScanner.isSettingsConfigAllowedForScan(BluetoothLeScanner.java:402)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:126)
       at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:102)
       at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.boolean deferScanIfNeeded()()
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.org.altbeacon.beacon.service.scanner.CycledLeScanner createScanner(android.content.Context,long,long,boolean,org.altbeacon.beacon.service.scanner.CycledLeScanCallback,com.useaurea.bluetooth.BluetoothCrashResolver)()
                                                              void setScanPeriods(long,long,boolean)
                                                              void start()
                                                              void scanLeDevice(java.lang.Boolean)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.void finishScanCycle()()
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.void scheduleScanCycleStop()()
       at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.void run()()
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:145)
       at android.app.ActivityThread.main(ActivityThread.java:5944)
       at java.lang.reflect.Method.invoke(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:372)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

davidgyoung, 2015-05-27:
This is related to an issue in #107 which was fixed in 2.1.2, but if you are experiencing this in a later release there must be a different root cause.  

The null pointer happens because the BluetoothAdapter is null when trying to start scanning.  The funny thing is that an earlier check found it to not be null.  

Is there any circumstance around when this happens?  Is it only when bluetooth is turned off?  Or does it happen all the time?

marcelorbenites, 2015-05-27:
It should be something related to Samsung's implementation of Bluetooth related classes or in Android 5.0.1 source code. Since BluetoothAdapter#getDefaultAdapter() docs states "return the default local adapter, or null if Bluetooth is not supported on this hardware platform.", BluetoothAdapter should never be null for Samsung Galaxy Note 4.

One option would be to always call:

    BluetoothManager bluetoothManager = (BluetoothManager)this.mContext.getApplicationContext().getSystemService("bluetooth");
    this.mBluetoothAdapter = bluetoothManager.getAdapter(); 

And don't keep a mBluetoothAdapter field in Lollipop scanner classes. But it would be just a guess.

marcelorbenites, 2015-05-27:
I have a Service that implements BeaconConsumer and runs in background "forever". We only saw this bug in Samsung Galaxy Note 4, it's a beta tester's phone that we don't have access. Unfortunately we don't have a Samsung Galaxy Note 4 or any Samsung with Android 5.0.1 to test that also. 
It happened three times in a 2 day interval only in Samsung Galaxy Note 4 .

davidgyoung, 2015-05-28:
So it is an intermittent problem?

marcelorbenites, 2015-05-28:
Yes.

davidgyoung, 2015-06-01:
@marcelorbenites, can you please test this snapshot release to see if it solves the problem on your device?  I do not have a device that can reproduce this problem for testing.

https://github.com/AltBeacon/android-beacon-library/releases/tag/fix-gn4-npe-1

marcelorbenites, 2015-06-02:
Ok, I will give a feedback if the bug happens again. I was just wondering
that I keep my service running all the time regardless of bluetooth state.
Would it prevent the bug I check for bluetooth state and start/stop
BeaconManager?

Thanks!

davidgyoung, 2015-06-18:
Fix attempted per https://github.com/AltBeacon/android-beacon-library/pull/200

Since I cannot reproduce please reopen if this does not resolve the problem.

closed 2015-06-18 by davidgyoung