AltBeacon/android-beacon-library #198: didRangeBeaconsInRegion called more then once per BackgroundScanPeriod and finds zero beacons

AlexTrotsenko, 2015-05-29:
I have an issue with Nexus 5 in with scanning in background mode. 

From time-to-time I have `RangeNotifier.didRangeBeaconsInRegion()`  called *more then once* per `BackgroundScanPeriod`. Usually I notified about one result with proper `beacon info`. But then up to 9 notifications without any beacon appear every second. 

This issue is happening with Android 5 API : `beaconManager.setAndroidLScanningDisabled(false);`.

I have tested it without Android 5 API : `beaconManager.setAndroidLScanningDisabled(true);` and looks like it's works without any problems.

Model: `Nexus 5`
Android version: `5.1`
Build Number: `LMY471`
WiFi : `On`

AltBeacon lib version:
- I have following dependency : `compile 'org.altbeacon:android-beacon-library:2+'`.
- In .`../build/intermediates/exploded-aar/org.altbeacon/android-beacon-library/2.1.4/AndroidManifest.xml` I see `android:versionName="2.1.4"`.

For testing following period / delay was used. But actually, I have same result with another periods as well. 

```.Java
            beaconManager.setBackgroundScanPeriod(TimeUnit.SECONDS.toMillis(20));
            beaconManager.setBackgroundBetweenScanPeriod(TimeUnit.SECONDS.toMillis(10));

```

So, without AndroidLScanning I see, that `didRangeBeaconsInRegion` is called every `30 seconds` (20 + 10) :

```
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:01:49.899
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:02:10.062
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:02:30.196
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:03:00.273
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:03:30.223
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:04:00.257
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:04:30.161
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:05:00.204
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:05:30.226
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:06:00.216
```

Here is full log:  https://gist.github.com/AlexTrotsenko/52c6b8bb2ec7e94d7440#file-log_beacons-txt

But with AndroidLScanning I see, that `didRangeBeaconsInRegion` sometimes called almost every second instead of once per `30 seconds` (20 + 10) :

```
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:14:17.967
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:14:22.950
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:14:28.048

........................................ <Somewhere here app have switched to Background >   ............
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:14:33.065
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:14:38.176

..........................................<Start to call didRangeBeaconsInRegion more then once per period> ........
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:15:00.192
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 21351 : 15:15:01.284
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:02.345
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:03.416
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:04.441
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:05.527
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:06.598
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:07.666
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:08.745
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:15:09.824

W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:15:30.256
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:16:00.252
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:16:30.313
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:17:00.201
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 2, 25287, 21351 : 15:17:30.292
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:18:00.210
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:18:30.244
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:19:00.212
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:19:30.276
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:20:00.200

..........................................<Start to call didRangeBeaconsInRegion more then once> ........
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:30.342
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:20:31.464
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:32.487
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:33.539
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:34.585
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:35.627
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:36.650
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:37.748
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:38.816
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:20:39.846

W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion. No new. : 15:21:00.218
W/ALEX_BeaconsDispatcherService﹕ didRangeBeaconsInRegion 1, 25287 : 15:21:30.258
```

Here is full log: https://gist.github.com/AlexTrotsenko/70dfddf67b9063001b61

Please, let me know if you need any additional info.
I read, that Android 5 API is more energy efficient, thus I would like to use them in my app.

Thank you in advance for any help.

davidgyoung, 2015-05-29:
Great tests!  Any chance you can re-run with `BeaconManager.setDebug(true); ` and capture the log output to a gist for the time you are getting the faster callbacks?

davidgyoung, 2015-05-29:
I also wonder if you app is being killed by the operating system and then relaunching itself (potentially reverting to different scan intervals on relaunch.)  It is possible being killed by the OS is only happening when the Android L scans are being used.  To determine if this is the case, you should log when your app launches and track this as well.

AlexTrotsenko, 2015-06-01:
@davidgyoung than you for quick reply. 

1.  I will check if app is killed, but I have a feeling, that it's not:  
Now I call `beaconManager.bind(this);` from my DispatcherService, that is started with `START_NOT_STICKY`. Inside my service I do `new RangeNotifier() {... Log.w("ALEX_"...}` . I assume following:  `new RangeNotifier` is anonymous inner class, thus It should be killed when app and service is killed.

2. > `Any chance you can re-run with BeaconManager.setDebug(true); and capture the log output to a gist for the time you are getting the faster callbacks?`

I have posted "my" logs for cases when I am getting faster callbacks:
```
didRangeBeaconsInRegion 1, 25287 : 15:15:00.192
```

or 

```
didRangeBeaconsInRegion. No new. : 15:20:30.342
```

Just search for these phrases in 2nd gist and you will find corresponded logs from Altbeacon lib as well.

davidgyoung, 2015-06-01:
I understand the issue now.  It actually seems the system is working largely as designed, but that design could use some tweaking.  Here's what's happening:

When you use the Android Beacon Library to look for beacons in the background using the Android L APIs, the BackgroundScanPeriod and BackgroundBetweenScanPeriod fields have special behavior as documented here:  https://altbeacon.github.io/android-beacon-library/battery_manager.html  The idea is that the BackgroundBetweenScanPeriod is ignored when no beacons are visible and scanning is constant using low power APIs.  This way, you get a much faster detection when a beacon first appears -- you don't have to wait for the BackgroundBetweenScanPeriod to be over.  If, on the other hand, beacons are constantly visible, the system reverts to cycling scans on and off with the BackgroundScanPeriod and the BackgroundBetweenScanPeriod in order to save battery.

In the case where beacons have not been visible, but then become visible on Android L, the implementation is actually also ignoring the BackgroundScanPeriod.  It probably should not be ignoring this.  The code in question is in the else statement here:  

https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/service/scanner/CycledLeScannerForLollipop.java#L137

What happens is that when beacons are first detected after a period of not being visible, they are immediately reported up to the RangingAPIs.  Every time the device sees a new beacon advertisement, it immediately reports it up, and this continues for BACKGROUND_L_SCAN_DETECTION_PERIOD_MILLIS (10 seconds), which is why you see 9 ranging callbacks for a beacon advertising at 1 Hz.  If you had a beacon advertising at 10Hz, you would see closer to 100 ranging callbacks.

One way to address this is to do the following:

1. When a beacon is first detected in the background, report the results up the chain exactly as is done now.
2. Note the current time as the start of the background scan cycle, and don't subsequently report results up the chain until the BackgroundScanPeriod has passed.
3. If you have still detected beacons after this second BackgroundScanPeriod ends, start using the BackgroundBetweenScanPeriod to save battery.
4. Get rid of the BACKGROUND_L_SCAN_DETECTION_PERIOD_MILLIS.  Instead of using this time period to revert to using the BackgroundBetweenScanPeriod to save battery, the library would do so after the configurable BackgroundScanPeriod.

With the changes above, and your settings of:

    BackgroundScanPeriod 20
    BackgroundBetweenScanPeriod 10

You would get exactly two callbacks to didRangeBeaconsInRegion after beacons are first detected, separated by 20 seconds.   The first would be as quickly as possible after first beacon detection.  The second would be after the BackgroundScanPeriod. Then you would start getting callbacks every 30 seconds as the BackgroundBetweenScanPeriod starts getting used.

While it would be possible to only provide a single callback to didRangeBeaconsInRegion in this case, it could be problematic from the perspective of a consuming application because the library would have to choose between doing one of two things:

1. Delaying the single callback until the BackgroundScanPeriod had passed.  This delay could be a problem for time critical applications.

2. Only delivering the first beacon seen in a single callback, but delivering it immediately.  This could be a problem if multiple beacons suddenly become visible.

AlexTrotsenko, 2015-06-02:
@davidgyoung thank you for investigating the problem. 

In such cases I usually do following:
1. When new data (beacons) is 1st appeared: I notify consumers about it and add these beacons to the list of "seen beacons".
3. During period I temporary keep tracking of incoming data. I add (*and remove if required by business logic*) new data (beacons) to "seen beacons".
3. At the end of period I just notify consumers one more time with "seen beacons".

Here is more advanced schema:
1. As new data (beacons) is 1st appeared - notify consumers about it. Add them to "processed beacons".
2. Also add these beacons to the list of "seen beacons".
3. During period temporary keep tracking of incoming data: add (*and remove if required by business logic*) new data (beacons) to "seen beacons".
3. At the end of period:
- If there was no new data - do nothing (if "processed beacons" same as "seen beacons"). 
- If there were new beacons (new data) during this time, then just notify consumers with  "seen beacons".

You may consider if this schema is applicable for this case with beacons.