AltBeacon/android-beacon-library #205: How to make sure it's transmitting or not?

datomnurdin, 2015-06-08:
Based on your documentation here, http://altbeacon.github.io/android-beacon-library/beacon-transmitter.html, how to make sure it's transmitting or not the beacon? How I can debug it?

```
Beacon beacon = new Beacon.Builder()
        .setId1("2f234454-cf6d-4a0f-adf2-f4911ba9ffa6")
        .setId2("1")
        .setId3("2")
        .setManufacturer(0x0118) // Radius Networks.  Change this for other beacon layouts
        .setTxPower(-59)
        .setDataFields(Arrays.asList(new Long[] {0l})) // Remove this for beacon layouts without d: fields
        .build();
// Change the layout below for other beacon types
BeaconParser beaconParser = new BeaconParser()
        .setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25");
BeaconTransmitter beaconTransmitter = new BeaconTransmitter(getApplicationContext(), beaconParser); 
beaconTransmitter.startAdvertising(beacon);
```

Please advice. Thank you.

davidgyoung, 2015-06-08:
@datomnurdin, I have just updated the documentation to show an example of how you check the status of starting a transmitter.

http://altbeacon.github.io/android-beacon-library/beacon-transmitter.html

If you get an error code back, you can see what the values mean by looking here:  https://developer.android.com/reference/android/bluetooth/le/AdvertiseCallback.html

closed 2015-06-08 by davidgyoung

datomnurdin, 2015-06-09:
Thank you very much sir.