AltBeacon/android-beacon-library #209: Enter/exit region

woutergoossens, 2015-06-09:
Hi,

First of all. Great library! 

But I have a problem: 

* When the region is entered I start "startRangingBeaconsInRegion" on the beaconManager.
* I have several beacons surrounded on the smartphone.
* But in the logs I see a lot of region status change (enter/exit region) without changing the location of the smartphone or beacons. Shouldn't the status be "entered" until the beacon gets out of range from the smartphone?

What could be the issue?

Thanks in advance,

Wouter 

`06-09 17:37:30.110   6904-11931/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:37:34.484   6904-11963/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:38:00.099   6904-12147/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:38:13.648   6904-12266/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:38:30.092   6904-12380/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:38:34.383   6904-12421/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:39:04.847  12886-12951/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:39:30.163  12886-13168/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:39:34.266  12886-13214/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:40:00.116  12886-13418/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:40:11.322  12886-13508/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:40:30.123  12886-13666/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:40:34.354  12886-13696/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:41:00.119  12886-13883/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:41:10.362  12886-13997/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:41:23.032  14063-14368/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:42:30.108  14063-14749/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:42:41.097  14063-14782/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:43:30.107  14063-15180/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:43:42.656  14063-15268/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:44:00.118  14063-15404/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:44:10.703  14063-15476/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:45:30.134  14063-16093/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:45:40.354  14063-16175/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:46:00.101  14063-16528/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:46:11.103  14063-16613/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:46:30.095  14063-16769/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:46:41.689  14063-16879/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:47:00.097  14063-16905/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:47:13.708  14063-16918/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:47:30.109  14063-16933/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:47:40.627  14063-16937/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:48:00.108  14063-16950/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 0 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null
06-09 17:48:12.636  14063-16954/com.icapps.tokheim.acc.ssl E/LocationService﹕ didDetermineStateForRegion 1 region id1: a0b13730-3a9a-11e3-aa6e-0800200c9a66 id2: null id3: null

`

davidgyoung, 2015-06-09:
This happens when the beacon packets are not detected by the library often enough.  If no packets are detected for 10 seconds, then the an exit condition fires.  There are three common reasons that beacon detections may not happen often enough to prevent exits even though beacons are in the vicinity:

1. The beacon is not transmitting often enough.  Some beacons transmit infrequently to save battery.  Even if a beacon transmits every 5 seconds (0.2 Hz), this problem might happen if one or two subsequent packets are lost due to radio noise.  To prevent his issue, make sure the beacon is transmitting at 1 Hz or faster, or reduce the scan period in the library to match the frequency of transmission.

2. The phone has a WiFi / Bluetooth interference problem.  Some phones like the Moto G, the Moto X and the Nexus 4 don't scan for bluetooth devices properly when WiFi is enabled.  This can cause intermittent cases where no detections can be made.  If you have one of these devices, disable WiFi.

3. The signal is weak.  If the Android device is on the edge of the transmission range for the beacon, only a small percentage of packets will be received.  If there is ever a period of 10 seconds or more where no packet is receive, an exit event will fire.

inbeacon, 2015-06-28:
Same problem here:

after updating from 4.4.4 to android 5.0.1 on a nexus 4. The device is entering/leaving continously, while the beacon has an advertisement interval of 100ms and is a meter away from the device. 
This started right after the update to android 5. Before that, everything was fine.

davidgyoung, 2015-06-28:
@inbeacon, please try the Locate app here:  

https://play.google.com/store/apps/details?id=com.radiusnetworks.locate

And see if you get regular detections or dropouts.  This app is based on the same library.

davidgyoung, 2015-06-28:
@infobeacon, please note also there is a known problem with the Nexus 4 where WiFi and Bluetooth do not work simultaneously.  Please be sure to turn off WiFi on the Nexus 4 before testing the reliability of beacon detections.

fabriziogueli, 2015-11-25:
I have the same problem using a Moto G (2nd generation) with Android 5.0.2.
The beacon has an advertisement interval of 100ms and is near the device but  in the logs I see a lot of region status change (enter/exit region) even if the wifi was turned off. 

I tried with an LG G4 and with a Samsung Galaxy Note 8 and it seems working as expected.

davidgyoung, 2016-01-07:
I suspect this is simply a case of the known issue of WiFi/Bluetooth interference.  Please reopen if this is not the case.

closed 2016-01-07 by davidgyoung