AltBeacon/android-beacon-library #212: NullPointerException in Region.matchesBeacon(Beacon)

mykolaj, 2015-06-10:
Hello!
I've got an issue which doesn't seem to depend on Android version or AltBeaconLib version. Anyways, it is currently confirmed to occur on some of Kyocera and Samsung devices running Android 4.4 and Android 5.0 and AltBeaconLib 2.1.4. Here's a stack trace:

    Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean org.altbeacon.beacon.Region.matchesBeacon(org.altbeacon.beacon.Beacon)' on a null object reference
	at org.altbeacon.beacon.service.BeaconService.matchingRegions(BeaconService.java:462)
	at org.altbeacon.beacon.service.BeaconService.processBeaconFromScan(BeaconService.java:399)
	at org.altbeacon.beacon.service.BeaconService.access$400(BeaconService.java:66)
	at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:439)
	at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:422)
	at android.os.AsyncTask$2.call(AsyncTask.java:288)
	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
	... 3 more

This exception happens in that loop: 
```
           Iterator<Region> regionIterator = regions.iterator();
            while (regionIterator.hasNext()) {
                Region region = regionIterator.next();
                if (region.matchesBeacon(beacon)) {
                    matched.add(region);
                } else {
                    LogManager.d(TAG, "This region (%s) does not match beacon: %s", region, beacon);
                }
            }
```
Of course a quickfix would be to just check if a Region is null and continue the loop if it is. Or is there any purpose for `null` key in `rangedRegionState` and `monitoredRegionState` maps?

davidgyoung, 2015-06-10:
It looks like this is most likely a threading issue and is not specific to a phone model.  Do you have reason to believe it is only an issue with "some Kyocera and Samsung devices" and not other models?

mykolaj, 2015-06-11:
I'm also thinking it is most likely a threading issue. I've just mentioned those phone models because those the phones i'm currently working with. This issue happened on other models too (one LG for sure). So, no. I think there's no reason to believe that this issue happens only on mentioned phones.
I will continue to look to find a solution now when i know that `null` is not for purpose.
By the way, the library is great!

davidgyoung, 2015-06-19:
Proposed fix here: https://github.com/AltBeacon/android-beacon-library/pull/215

mykolaj, 2016-04-26:
Its been almost a year. I think its time to close this issue.

davidgyoung, 2016-04-26:
Indded!  The proposed fix has been merged.

tnortman, 2018-11-09:
I believe this bug is back.

Looks like the method `matchingRegions`, which in previous releases resided in `beacon/service/BeaconService.java` got moved to `beacon/service/ScanHelper.java`
and the null check is no longer there.

I updated my gradle dependency in my app from 
`    api 'org.altbeacon:android-beacon-library:2.9.1'`
to
`    api 'org.altbeacon:android-beacon-library:2.15.2'`
and now every time I launch my app (when beacons are in range), it crashes with the same stack trace as above. I am running Android 8.0 but I don't believe that matters

davidgyoung, 2018-11-09:
Thanks for the report.  I'll look to add an equivalent check in ScanHelper.

davidgyoung, 2018-11-10:
Fixed again in #770

EnesKy, 2019-04-19:
I've got this error at API 26 when i create a Region in onCreate() and using it like global variable.
It might have happens because Region value created before onBeaconServiceConnect() call.

By the way i'm using the latest version. (2.16.2-beta1@aar)

[image]

davidgyoung, 2019-04-19:
Argh!  It looks like my fix for this in #770 was never merged.  I accidentally closed the PR without merging it

davidgyoung, 2019-04-19:
Ok, #770 merged to master now.  closing this issue for the third time!

closed 2019-04-19 by davidgyoung