AltBeacon/android-beacon-library #213: Reference app won't start after phone restart

JaseElder, 2015-06-18:
Hi,
Phone is Sony Xperia M2 D2302 with KitKat 4.4.4

Using iBeacon type beacons, so only change to reference app code is inclusion of appropriate beacon layout, and switching on beacon manager debugging

Reference app works: 
1) when app is in foreground
2) when app is in background
3) when app has been killed from task manager and then charger is attached.
4) when phone as been restarted **and then** charger is attached.

The only time it's not working is when the phone has been restarted, without unplugging/plugging charger.

Here are the logs:

```
06-18 17:37:07.451     960-1250/? I/ActivityManager﹕ Start proc org.altbeacon.beaconreference for broadcast org.altbeacon.beaconreference/org.altbeacon.beacon.startup.StartupBroadcastReceiver: pid=3774 uid=10279 gids={50279, 3003, 3002, 3001}
06-18 17:37:07.711    3774-3774/org.altbeacon.beaconreference D/AndroidProximityReferenceApplication﹕ setting up background monitoring for beacons and power saving
06-18 17:37:07.721    3774-3774/org.altbeacon.beaconreference D/StartupBroadcastReceiver﹕ onReceive called in startup broadcast receiver
06-18 17:37:07.721    3774-3774/org.altbeacon.beaconreference I/BeaconService﹕ beaconService version 2.1.4 is starting up
06-18 17:37:07.721    3774-3774/org.altbeacon.beaconreference D/BluetoothCrashResolver﹕ constructed
06-18 17:37:07.721    3774-3774/org.altbeacon.beaconreference W/BluetoothCrashResolver﹕ Can't read macs from BluetoothCrashResolverState.txt
06-18 17:37:07.731    3774-3774/org.altbeacon.beaconreference D/BluetoothCrashResolver﹕ Read 0 bluetooth addresses
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference D/BluetoothCrashResolver﹕ started listening for BluetoothAdapter events
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference I/CycledLeScanner﹕ This is not Android 5.0.  We are using old scanning APIs
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference D/ModelSpecificDistanceCalculator﹕ Finding best distance calculator for 4.4.4, 18.3.1.B.1.10, D2302, Sony
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference D/AndroidModel﹕ Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Sony;D2302;18.3.1.B.1.10;4.4.4
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference D/AndroidModel﹕ Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Sony;D2302;18.3.1.B.1.10;4.4.4
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference W/ModelSpecificDistanceCalculator﹕ Cannot find match for this device.  Using default
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference D/ModelSpecificDistanceCalculator﹕ Finding best distance calculator for 4.4.4, 18.3.1.B.1.10, D2302, Sony
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference D/AndroidModel﹕ Score is 0 for LGE;Nexus 5;LPV79;4.4.2 compared to Sony;D2302;18.3.1.B.1.10;4.4.4
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference D/AndroidModel﹕ Score is 0 for LGE;Nexus 4;KOT49H;4.4.2 compared to Sony;D2302;18.3.1.B.1.10;4.4.4
06-18 17:37:07.741    3774-3774/org.altbeacon.beaconreference W/ModelSpecificDistanceCalculator﹕ Cannot find match for this device.  Using default
06-18 17:37:07.751    3774-3774/org.altbeacon.beaconreference D/BeaconService﹕ No org.altbeacon.beacon.SimulatedScanData class exists.
06-18 17:37:07.751    3774-3774/org.altbeacon.beaconreference I/BeaconService﹕ binding
06-18 17:37:07.761    3774-3774/org.altbeacon.beaconreference D/BeaconManager﹕ we have a connection to the service now
06-18 17:37:07.771    3774-3774/org.altbeacon.beaconreference D/AppStarter﹕ Activating background region monitoring
06-18 17:37:07.771    3774-3774/org.altbeacon.beaconreference D/AppStarter﹕ Background region monitoring activated for region id1: null id2: null id3: null
06-18 17:37:07.781    3774-3774/org.altbeacon.beaconreference D/BeaconManager﹕ callback packageName: org.altbeacon.beaconreference
06-18 17:37:07.791    3774-3774/org.altbeacon.beaconreference D/BeaconManager﹕ updating background flag to true
06-18 17:37:07.791    3774-3774/org.altbeacon.beaconreference D/BeaconManager﹕ updating scan period to 10000, 300000
06-18 17:37:07.791    3774-3774/org.altbeacon.beaconreference I/BeaconService﹕ start monitoring received
06-18 17:37:07.791    3774-3774/org.altbeacon.beaconreference D/BeaconService﹕ startMonitoring called
06-18 17:37:07.791    3774-3774/org.altbeacon.beaconreference D/BeaconService﹕ Currently monitoring 1 regions.
06-18 17:37:07.791    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ start called
06-18 17:37:07.801    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ starting a new scan cycle
06-18 17:37:07.801    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ starting a new bluetooth le scan
06-18 17:37:07.801    3774-3774/org.altbeacon.beaconreference D/BluetoothAdapter﹕ startLeScan(): null
06-18 17:37:07.961    3774-3785/org.altbeacon.beaconreference D/BluetoothAdapter﹕ onClientRegistered() - status=0 clientIf=5
06-18 17:37:07.971    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Waiting to stop scan cycle for another 1100 milliseconds
06-18 17:37:07.971    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Scan started
06-18 17:37:07.971    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Set scan periods called with 1100, 0 Background mode must have changed.
06-18 17:37:07.971    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ We are not in the background.  Cancelling wakeup alarm
06-18 17:37:07.981    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ cancel wakeup alarm: null
06-18 17:37:07.981    3774-3774/org.altbeacon.beaconreference I/BeaconService﹕ set scan intervals received
06-18 17:37:07.991    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Set scan periods called with 10000, 300000 Background mode must have changed.
06-18 17:37:08.001    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ We are in the background.  Setting wakeup alarm
06-18 17:37:08.011    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ cancel wakeup alarm: null
06-18 17:37:08.021    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Set a wakeup alarm to go off in 300000 ms: PendingIntent{419285f0: android.os.BinderProxy@41928580}
06-18 17:37:08.971    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Waiting to stop scan cycle for another 98 milliseconds
06-18 17:37:08.981    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ cancel wakeup alarm: PendingIntent{419285f0: android.os.BinderProxy@41928580}
06-18 17:37:08.991    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Set a wakeup alarm to go off in 300000 ms: PendingIntent{419297a8: android.os.BinderProxy@41929738}
06-18 17:37:09.091    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Done with scan cycle
06-18 17:37:09.091    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ stopping bluetooth le scan
06-18 17:37:09.091    3774-3774/org.altbeacon.beaconreference D/BluetoothAdapter﹕ stopLeScan()
06-18 17:37:09.091    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Normalizing between scan period from 300000 to 150895
06-18 17:37:09.091    3774-3774/org.altbeacon.beaconreference D/CycledLeScannerForJellyBeanMr2﹕ Waiting to start next bluetooth scan for another 150895 milliseconds
06-18 17:37:09.091    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ cancel wakeup alarm: PendingIntent{419297a8: android.os.BinderProxy@41929738}
06-18 17:37:09.101    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Set a wakeup alarm to go off in 300000 ms: PendingIntent{4192ae28: android.os.BinderProxy@4192adb8}
06-18 17:37:10.101    3774-3774/org.altbeacon.beaconreference D/CycledLeScannerForJellyBeanMr2﹕ Waiting to start next bluetooth scan for another 149882 milliseconds
06-18 17:37:10.111    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ cancel wakeup alarm: PendingIntent{4192ae28: android.os.BinderProxy@4192adb8}
06-18 17:37:10.111    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Set a wakeup alarm to go off in 300000 ms: PendingIntent{4192c318: android.os.BinderProxy@4192c2a8}
06-18 17:37:11.121    3774-3774/org.altbeacon.beaconreference D/CycledLeScannerForJellyBeanMr2﹕ Waiting to start next bluetooth scan for another 148871 milliseconds
[... cut ...]
06-18 17:37:11.131    3774-3774/org.altbeacon.beaconreference D/CycledLeScanner﹕ Set a wakeup alarm to go off in 300000 ms: PendingIntent{4192d460: android.os.BinderProxy@4192d3f0}
06-18 17:37:12.111      960-970/? I/ActivityManager﹕ Killing 3774:org.altbeacon.beaconreference/u0a279 (adj 15): empty #17
06-18 17:37:12.131     960-1441/? A/ActivityManager﹕ Service ServiceRecord{42636878 u0 org.altbeacon.beaconreference/org.altbeacon.beacon.service.BeaconService} in process ProcessRecord{4272d818 3774:org.altbeacon.beaconreference/u0a279} not same as in map: null
```

davidgyoung, 2015-06-18:
@JaseElder, this sounds like the same issue as here:  https://github.com/AltBeacon/android-beacon-library/issues/44

This may be a device-specific problem, so I'd really appreciate it if you could clone, build and run the test app in this repo I created:  https://github.com/davidgyoung/power-connect-tester, then follow the instructions in the project readme, then report back as to what you see.  

This should take no more than 15 minutes if you have an Android Studio dev environment.  And it would really help in tracking down this issue.

JaseElder, 2015-06-18:
Ah, forgot to say - I installed and ran that yesterday, and get the expected output.
ACTION_POWER_DISCONNECTED and 
ACTION_POWER_CONNECTED

on foreground and background, and after app quit from task manager.

The issue I'm experiencing seems to be for a slightly different use case though. If I restart the device, and then disconnect/connect the charger, the reference app is fine.
It's only in the case where I restart the device, and **don't** touch the charger that it doesn't start monitoring.

davidgyoung, 2015-06-19:
For the log shown, it appears the beacon service is scanning.  does this log really show the problem case right after device power on?

Also, when you run the Power Connect Tester app, did you try rebooting the phone and seeing if it displays "ACTION_BOOT_COMPLETED"?

JaseElder, 2015-06-19:
Ran a new test to double check and now it works perfectly!
Not sure if your latest commit fixed it (I wasn't getting an NPE myself), or if it was a transient problem with my device, (or if I was doing something stupid like not waiting long enough) but it appears to be all go!
Thanks! 
BTW, we're using your library here: https://www.kickstarter.com/projects/juanjimenez/homies/description

davidgyoung, 2015-06-19:
Glad to hear the problem is gone, @JaseElder.  Good luck on your kickstarter!

closed 2015-06-19 by davidgyoung