AltBeacon/android-beacon-library #219: NullPointerException in Samsung Galaxy S5 - Android 5.0 in CycledLeScannerForLollipop

marcelorbenites, 2015-06-26:
Hi,

I have a NullPointerException with the following stack trace on Samsung Galaxy S5 with Android 5.0 using version 2.3 of android-beacon-library. My implementation has a Service that implements BeaconConsumer and runs in background "forever" monitoring the environment for Beacons with a short interval.

    java.lang.NullPointerException: Attempt to invoke virtual method 'android.os.Looper    android.os.Handler.getLooper()' on a null object reference
        at android.os.Parcel.readException(Parcel.java:1546)
        at android.os.Parcel.readException(Parcel.java:1493)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.unregisterClient(IBluetoothGatt.java:905)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.stopLeScan(BluetoothLeScanner.java:262)
        at android.bluetooth.le.BluetoothLeScanner.stopScan(BluetoothLeScanner.java:151)
        at com.useaurea.beacon.service.scanner.CycledLeScannerForLollipop.deferScanIfNeeded()
        at com.useaurea.beacon.service.scanner.CycledLeScanner.createScanner()
                                                              setScanPeriods
                                                              start
                                                              scanLeDevice
        at com.useaurea.beacon.service.scanner.CycledLeScannerForLollipop$1.run()
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:145)
        at android.app.ActivityThread.main(ActivityThread.java:5832)
        at java.lang.reflect.Method.invoke(Method.java)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1399)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1194)

davidgyoung, 2015-06-26:
This appears to be an Android bluetooth bug reported here:
    https://code.google.com/p/android/issues/detail?id=160503

The only way to protect against this would be to catch a NullPointerException when stopping scan in the line of code below:

     getScanner().stopScan((android.bluetooth.le.ScanCallback) getNewLeScanCallback());

marcelorbenites, 2015-06-26:
Will you create a patch release?

davidgyoung, 2015-06-27:
Does this NPE happen often enough that you will know for sure if a build fixes it?  If so, please try this release and report back on what you see.

https://github.com/AltBeacon/android-beacon-library/releases/tag/catch-start-stop-npe

davidgyoung, 2016-01-07:
closed due to inactivity

closed 2016-01-07 by davidgyoung

st-f, 2016-10-11:
Got that issue on a Nexus 5X with Nougat:

> 10-11 22:39:38.850 21044-21044 E/CycledLeScannerForLollipop: Scan Failed
10-11 22:39:38.866 21044-21044 D/BluetoothAdapter: STATE_ON
10-11 22:39:40.891 21044-21044 D/BluetoothAdapter: STATE_ON
10-11 22:39:40.891 21044-21044 D/BluetoothLeScanner: could not find callback wrapper
10-11 22:39:40.896 21044-21044 D/BluetoothAdapter: STATE_ON
10-11 22:39:40.905 21044-23288 D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=18 mClientIf=0
10-11 22:39:40.935 21044-21044 D/BluetoothAdapter: STATE_ON
10-11 22:39:40.942 21044-21044 D/BluetoothAdapter: STATE_ON
10-11 22:39:40.947 21044-22001 D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=18 mClientIf=0
10-11 22:39:44.989 21044-21044 D/BluetoothAdapter: STATE_ON
10-11 22:39:44.999 21044-21044 E/CycledLeScannerForLollipop: Scan Failed

Works again after a reboot. It happens quite randomly.

davidgyoung, 2016-10-11:
@st-f, does this cause a crash with `java.lang.NullPointerException: Attempt to invoke virtual method 'android.os.Looper    android.os.Handler.getLooper()' on a null object reference` as described in this issue, or do you just see the "ScanFailed" log lines you mention in your comment?

st-f, 2016-10-12:
Thanks for the quick reply - it doesn't crash but I can't detect any beacons after it happens the first time. I was trying short scan period (1000 and less) and short in between scan time (50 to 200ms), maybe that's related?

st-f, 2016-10-12:
Strangely enough it just happened I restarted the device... But it works fine after I reinstall the app

st-f, 2016-10-12:
Enabled debug (deprecated, but it works) :

> 10-12 01:24:06.974 29010-29010 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 90518 milliseconds
10-12 01:24:07.975 29010-29010 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 89517 milliseconds
10-12 01:24:08.976 29010-29010 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 88516 milliseconds
10-12 01:24:09.976 29010-29010 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 87515 milliseconds
10-12 01:24:10.979 29010-29010 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 86514 milliseconds
10-12 01:24:11.980 29010-29010 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 85512 milliseconds
10-12 01:24:12.982 29010-29010 D/CycledLeScannerForLollipop: Waiting to start full Bluetooth scan for another 84510 milliseconds

st-f, 2016-10-12:
It's now stuck into a loop; it pauses at 300s when the countdown is finished, allowing a few scans to be done in the meanwhile

st-f, 2016-10-12:
Sorry this wasn't related to this issue actually. Not sure if that would be any help but I also found how to resolve the issue. I was using setSampleExpirationMilliseconds with a value smaller than the scan period. This was causing the "CycledLeScannerForLollipop: Scan Failed"