AltBeacon/android-beacon-library #221: IllegalArgumentException: start > end.

gotama, 2015-06-29:
So I have released my app and I am only dealing with beacons of my own UUID, which are not out in production yet.  But I have gotten over 700 crashes that are failing silently within the alt beacon library but it is creating a lot of noise in my reporting system.

We shouldn't be detecting any beacons as there are no beacons with my UUID in the wild.

I am struggling to reproduce this, so Ill paste my stack trace and hopefully someone can help me understand why we the ScanProcessor is trying to process beacons that are not of my BeaconParsers UUID??

Stack Trace
_________________________________
0	java.lang.RuntimeException: An error occured while executing doInBackground()
1		at android.os.AsyncTask$3.done(AsyncTask.java:300)
2		at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
3		at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
4		at java.util.concurrent.FutureTask.run(FutureTask.java:242)
5		at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
6		at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
7		at java.lang.Thread.run(Thread.java:841)
8	Caused by: java.lang.IllegalArgumentException: start > end
9		at org.altbeacon.beacon.Identifier.fromBytes(Identifier.java:125)
10		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:421)
11		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:340)
12		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:439)
13		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:430)
14		at android.os.AsyncTask$2.call(AsyncTask.java:288)
15		at java.util.concurrent.FutureTask.run(FutureTask.java:237)
16		... 3 more
17	java.lang.IllegalArgumentException: start > end
18		at org.altbeacon.beacon.Identifier.fromBytes(Identifier.java:125)
19		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:421)
20		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:340)
21		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:439)
22		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:430)
23		at android.os.AsyncTask$2.call(AsyncTask.java:288)
24		at java.util.concurrent.FutureTask.run(FutureTask.java:237)
25		at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
26		at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
27		at java.lang.Thread.run(Thread.java:841)

gotama, 2015-06-29:
Also my Beacon parser code snippet:
mBeaconManager.getBeaconParsers().add(new BeaconParser().
                setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));

davidgyoung, 2015-06-29:
It appears that malformed beacon packet transmissions are causing a decoding to crash when the malformed packet ends in the middle of one of the beacon identifiers.   Fixing this will require a small update to the library.  In the meantime, you can safely ignore this exception (although I realize that may be easier said than done if it makes noise in your reporting system) as the library will restart scanning automatically after the exception. 

There is nothing wrong with your beacon layout string (which I have redacted to keep intellectual property out of this open source repo -- apologies!)  Understand that even though you have not deployed any of your beacons, the library still has to decode any beacon packets it sees in order to see if it matches one of your identifiers.

gotama, 2015-06-29:
Hey David,

Thanks for getting back to me, I am in the process of trying to figure out how to reduce the noise on our logging system but as you advised this has some complexity.

I would like to furnish you with what ever data you need in order to assist in a speedy fix for this issue.  We are using crittercism as our reporting system.  So i can paste the stack trace for this error which since the 26th of June 2015 has logged over 700 crashes.  

Crittercism entry:
Error
java.lang.IllegalArgumentException known 
An error occured while executing doInBackground()
Last Occurred
29-Jun-15 05:42:18 PM
First Occurred
26-Jun-15 05:35:14 AM
Times Unsymbolicated
782
Total Occurrences
782
Users Affected
309

Fullstack trace:
Stack Trace
_________________________________
0	java.lang.RuntimeException: An error occured while executing doInBackground()
1		at android.os.AsyncTask$3.done(AsyncTask.java:300)
2		at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
3		at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
4		at java.util.concurrent.FutureTask.run(FutureTask.java:242)
5		at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
6		at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
7		at java.lang.Thread.run(Thread.java:841)
8	Caused by: java.lang.IllegalArgumentException: start > end
9		at org.altbeacon.beacon.Identifier.fromBytes(Identifier.java:125)
10		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:421)
11		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:340)
12		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:439)
13		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:430)
14		at android.os.AsyncTask$2.call(AsyncTask.java:288)
15		at java.util.concurrent.FutureTask.run(FutureTask.java:237)
16		... 3 more
17	java.lang.IllegalArgumentException: start > end
18		at org.altbeacon.beacon.Identifier.fromBytes(Identifier.java:125)
19		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:421)
20		at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:340)
21		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:439)
22		at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:430)
23		at android.os.AsyncTask$2.call(AsyncTask.java:288)
24		at java.util.concurrent.FutureTask.run(FutureTask.java:237)
25		at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
26		at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
27		at java.lang.Thread.run(Thread.java:841)

Threads
_________________________________
Thread: Thread-789
0	java.lang.Object.wait(Native Method)
1	java.lang.Object.wait(Object.java:401)
2	com.path.android.jobqueue.JobManager$2.getNextJob(JobManager.java:441)
3	com.path.android.jobqueue.executor.JobConsumerExecutor$JobConsumer.run(JobConsumerExecutor.java:198)
4	java.lang.Thread.run(Thread.java:841)

Thread: Thread-787
0	java.lang.Object.wait(Native Method)
1	java.lang.Object.wait(Object.java:401)
2	com.path.android.jobqueue.JobManager$2.getNextJob(JobManager.java:441)
3	com.path.android.jobqueue.executor.JobConsumerExecutor$JobConsumer.run(JobConsumerExecutor.java:198)
4	java.lang.Thread.run(Thread.java:841)

Thread: AsyncTask #2
[... 116 lines cut ...]
GT-I9500 - 82
SM-N9005 - 70
SM-G900H - 66
HTC One_M8 - 61
SM-G925F - 49
SM-N900 - 35
SM-G920F - 33
D6603 - 28
D5803 - 23
Other - 192

System Versions:
5.0 - 37
5.0.1 - 83
5.0.2 - 163
4.3 - 16
4.4.4 - 52
5.1.1 - 15
4.4.2 - 474

Please let me know if there is anything I can assist with.

Regards,

James

davidgyoung, 2015-06-30:
Fixed in https://github.com/AltBeacon/android-beacon-library/pull/224

Upon further investigation, I don't believe this is a mobile device-specific issue.  It appears to be the result of beacons that (either sometimes or all the time) actually transmit a packet that is too short to be properly formatted.    The change in the linked pull request will make it so these packets are silently ignored vs. causing the parsing thread to crash and making for noisy logs.

gotama, 2015-07-02:
Thank you David,will let you know if this issue persists after update.

Good work :)

davidgyoung, 2015-07-07:
This is in the 2.3.2 release.

closed 2015-07-07 by davidgyoung