AltBeacon/android-beacon-library #222: Unable to detect Beacons on Micromax A190

kiledar2006, 2015-06-30:
Hello I am using AltBeacon library sample app, when i start ranging, it's not showing any beacon on Micromax A190 model, but it's working fine on Samsung, HTC, Lava & Motorola devices.

can u please tell me why it's not working on Micromax.

davidgyoung, 2015-06-30:
On the Micromax, please ensure BLE is enabled, set `beaconMangager.setDebug(true)` and collect a LogCat excerpt for 10 seconds where the app is not detecting beacons.   Please put a link to that LogCat excerpt as a Gist in this issue.

joginder89, 2015-08-04:
I have the same issue and this is link 
https://gist.github.com/joginder89/f641567b64a3da1ea4e5

davidhelms, 2015-08-04:
This model does not appear to come equipped with Bluetooth 4.0 support, which is a prerequisite

http://www.micromaxinfo.com/mobiles/smartphones/canvas/Canvas-HD-Plus-A190

davidgyoung, 2015-08-05:
I agree, it looks like the specs for the Micromax A190 says it only has Bluetooth 3.0.  This is unusual for a modern phone, but means that this won't work with the library.

closed 2015-08-05 by davidgyoung