AltBeacon/android-beacon-library #229: NPE with BeaconLayout "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"

HanSolo, 2015-07-06:
I've updated to version 2.3.1 and have problems with the BeaconLayout "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25". As soon as I use the given BeaconLayout I get the following RuntimeException:

07-06 09:04:46.083  30544-30689/eu.hansolo.healthtrack E/AndroidRuntime﹕ FATAL EXCEPTION: pool-4-thread-1
    Process: eu.hansolo.healthtrack, PID: 30544
    java.lang.RuntimeException: An error occured while executing doInBackground()
            at android.os.AsyncTask$3.done(AsyncTask.java:300)
            at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
            at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
            at java.util.concurrent.FutureTask.run(FutureTask.java:242)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
            at java.lang.Thread.run(Thread.java:818)
     Caused by: java.lang.NullPointerException: Attempt to write to field 'java.util.List org.altbeacon.beacon.Beacon.mIdentifiers' on a null object reference
            at org.altbeacon.beacon.BeaconParser.fromScanData(BeaconParser.java:496)
            at org.altbeacon.beacon.AltBeaconParser.fromScanData(AltBeaconParser.java:59)
            at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:446)
            at org.altbeacon.beacon.service.BeaconService$ScanProcessor.doInBackground(BeaconService.java:437)
            at android.os.AsyncTask$2.call(AsyncTask.java:288)
            at java.util.concurrent.FutureTask.run(FutureTask.java:237)
            at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
            at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
            at java.lang.Thread.run(Thread.java:818)

With a BeaconLayout of "m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25" and different Beacons it works in 2.3.1

davidgyoung, 2015-07-06:
Thanks for this report.  I have put in a change to make the BeaconParser more tolerant of errors in decoding in #230, and rolled this into a 2.3.2 release.  Please let me know if you see other problems.

davidgyoung, 2015-07-06:
Apologies, I had to redact part of the issue text, as it exposes a proprietary beacon layout which cannot be shown in this open source repository.

HanSolo, 2015-07-06:
Ok, with 2.3.2 it doesn't crash any more :)
But now I can't see any beacon at all (no AltBeacon and no iBeacon), I saw that I now get
"Can't read macs from BluetoothCrashResolverState.txt" which might be related to something else...not sure.

davidgyoung, 2015-07-06:
Can you please add the line:

beaconManager.setDebug(true);

And capture a LogCat excerpt where your beacon is not detected.  The log may be long so you can link to a Gist if needed.

HanSolo, 2015-07-06:
Ok, just to give you an idea what I'm doing here. I have a mixed setup of AltBeacons and iBeacons and would like to monitor and range both of them. With the 2.3.2 version it seems the iBeacons will be detected fine but my app doesn't seem to see the AltBeacons. I've added to Parsers,

iBeacon:
beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));

AltBeacon (tried 2 approaches):
beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));

beaconManager.getBeaconParsers().add(new AltBeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));

It seems the lib has problems with detecting the manufacturer id "beac" here...just a thought.

Please find the gist here: https://gist.github.com/HanSolo/385e052170bfa1c5294e

davidgyoung, 2015-07-06:
In the gist, which beacon parser expression is active?

HanSolo, 2015-07-06:
This one was only with the following parser:
beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));

Just to make sure...they should run both right? Or can I only scan either iBeacons or AltBeacons ?

HanSolo, 2015-07-06:
To make sure it was the right parser, here one more gist with only using the following parser:

beaconManager.getBeaconParsers().add(new AltBeaconParser().setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25"));

https://gist.github.com/HanSolo/a6a0ae655e990bd4828c

davidgyoung, 2015-07-06:
Both will run, simultaneously, yes.  

However, looking at the Gist, I don't see that there is a transmitter active that is sending an AltBeacon packet.   If you look in the log you will see a line like below each time a bluetooth advertisement is detected:

07-06 15:36:27.626  21269-21475/eu.hansolo.healthtrack D/BeaconParser﹕ Processing pdu type FF: 0201041bff4c000215a7ae2eb71f004168b99ba749bac1ca6400010001bb6200000000000000000000000000000000000000000000000000000000000000 with startIndex: 5, endIndex: 30

I don't see any lines where there is the two byte signature of an AltBeacon: beac

Can you confirm that you actually do have a device transmitting an AltBeacon packet?

HanSolo, 2015-07-06:
Yep, I'm using blukii beacons in AltBeacon mode and checked it with the Locate app (see screenshot). The interesting thing is that these beacons worked with the exact same parser with version 2.1.4 until this morning. 
[image]

HanSolo, 2015-07-06:
Here is another screenshot that shows that the beacon is sending in AltBeacon format...
[image]

HanSolo, 2015-07-06:
Any results here? Still doesn't work...

davidgyoung, 2015-07-06:
Sorry, I cannot reproduce this.  Can you try to get a LogCat capture with lines that start with:

      This is not a matching Beacon advertisement.

This would indicate a non-matching beacon parser.

If I can get a sample that shows this happening, I might be able to figure out what is going on.  Alternately, if you could reproduce with the reference application that would be helpful, too.

davidgyoung, 2015-07-07:
I have verified that the BeaconParser with the proper layout will parse the AltBeacon advertisement byte sequence shown on the Bluetooth LE Scanner screenshot.  The following test passes:

```
    @Test
    public void testParseProblematicBeaconFromIssue229() {
        org.robolectric.shadows.ShadowLog.stream = System.err;

        byte[] bytes = hexStringToByteArray("0201061bffe000beac7777772e626c756b692e636f6d000100010001abaa000000");
        BeaconParser parser = new BeaconParser();
        parser.setBeaconLayout("m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25");

        Beacon beacon = parser.fromScanData(bytes, -55, null);
        assertNotNull("beacon should be parsed", beacon);
    }
```

This suggests to me that the problem you are experiencing is not a parser problem, but may have something to do with the way your app sets up ranging or configures its beacon layouts.

HanSolo, 2015-07-07:
Ok, will give it another try. Just wondering because the exact same code worked before :(

davidgyoung, 2015-07-07:
Two things to clarify:

1. You do not need to add any BeaconParser for AltBeacon.  It is built-in and active by default.
2. In the second Gist I cannot see any bluetooth at all.  It may be helpful if you can do a second LogCat capture without filtering for the app's process id.  That would allow me to see the raw bluetooth logging information like below

07-06 16:25:31.857    1067-1089/eu.hansolo.healthtrack D/BluetoothLeScanner﹕ onScanResult() - ScanResult{mDevice=DE:7F:51:93:09:CC, mScanRecord=ScanRecord [mAdvertiseFlags=4, mServiceUuids=null, mManufacturerSpecificData={76=[2, 21, -89, -82, 46, -73, 31, 0, 65, 104, -71, -101, -89, 73, -70, -63, -54, 100, 0, 1, 0, 2, -69, 85]}, mServiceData={}, mTxPowerLevel=-2147483648, mDeviceName=null], mRssi=-75, mTimestampNanos=7026431159419}

HanSolo, 2015-07-08:
Ok, I've tried all that and it seems that as soon as I switch the beacons to AltBeacon mode my app doesn't recognize them any more ( I really don't know why because it worked like mentioned before with 2.1.4). So I've switched the beacons to iBeacon mode and that works perfectly. Don't ask me why it doesn't see the beacons in AltBeacon mode but I'm fine now using iBeacon mode.

davidgyoung, 2015-07-10:
@HanSolo I think I figured out what is causing this problem.   The AltBeacons you are using have a manufacturer code of 0x00e0 which is the Bluetooth SIG identifier for Google.   (Note this is a different  field from the 0xbeac AltBeacon beacon type code) The filter on Android 5.0's BLE scanning APIs is not expecting this code.

Are these off the shelf beacons or something custom you made?

HanSolo, 2015-07-10:
Ok, that would explain this behavior. The beacons are off the shelf beacons that you can find here: http://www.blukii.com

davidgyoung, 2016-01-07:
Fixed in 2.3.3
https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3.3

closed 2016-01-07 by davidgyoung