AltBeacon/android-beacon-library #231: update 2.3.1 seems to detect empty beacon list

Onheiron, 2015-07-06:
When synced gradle in my project to v2.3.1 I could no longer detect beacons in my office.

didRangeBeaconsInRegion is called, but beacons list is empty.

davidgyoung, 2015-07-06:
OK ase try with the 2.3.2 release.  If you still have problems, please include any custom beacon parser expression with the issue, add the line beaconManager.setDebug(true) and attach a LogCat excerpt.  That will help find the problem.

davidgyoung, 2015-07-10:
@Onheiron, this may be related to the issue described in #235 

Can you please answer the following:

* What type of beacons you are using?  Are they iBeacon transmitters or AltBeacon transmitters?  Who is the manufacturer?
* What type of mobile device are you using?

davidgyoung, 2016-01-07:
Resolved in 2.3.3 
https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3.3

closed 2016-01-07 by davidgyoung

andretietz, 2016-11-23:
Happens to me on 2.9.1:
[image]

ealvaradogsg, 2017-06-08:
Hi Team, I have the same problem, i work with Estimote and we have the location proximity beacons. I use the 2+ version in my gradle. 

[image]

[image]

This issue have any solve in this community? 

My device: Huaweii p8 LTE: Android 5.1

Regards, thanks for all