AltBeacon/android-beacon-library #234: No beacons detected on MotoE2

SergiuC10, 2015-07-09:
With the latest update to CycledLeScannerForLollipop no beacons are detected on Moto E2 (5.0.2)

I've managed to solve it changing 
```
getScanner().startScan(scanFilters, settings, callback);
```

in startScan to 

```
getScanner().startScan(filters, settings, callback);
```

chuckurbis, 2015-07-09:
I also have been having problems when using a MotoE2 with all versions after 2.3 when trying to use a range notifier.

davidgyoung, 2015-07-10:
@SergiuC10  and @chuckurbis, can you please tell me if you are using iBeacon or AltBeacons?  If you are using AltBeacons, which manufacturer provided them?

davidgyoung, 2015-07-10:
The core issue of this is addressed in #235.  However, I would still like to understand why the MotoE2 had this problem.  Were you using unusual AltBeacons with the library?  Or do Android 5.x BLE filters simply not work on the Moto E2?

chuckurbis, 2015-07-10:
I am using EM MICROELECTRONIC EMBC01 and also the iPhone Locate app. None of the app settings or the physical beacon were recognized.

davidgyoung, 2015-07-10:
Thanks, @chuckurbis, that is very helpful.

To confirm, you are saying you cannot detect an iBeacon transmission from the Locate App on an iPhone with 2.3.x?  We're you able to detect the iPhone's transmission prior to 2.3?

chuckurbis, 2015-07-10:
2.3 seems to work. I've rolled back to that version in the project I'm working on and all seems well. Versions beyond that are where the issue comes up.

davidgyoung, 2016-01-07:
This issue was resolved in 2.3.3

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3.3

closed 2016-01-07 by davidgyoung