AltBeacon/android-beacon-library #237: updating android-distance json

eckoj433, 2015-07-13:
The json file for model specific distance calculations really need to get updated (http://data.altbeacon.org/android-distance.json). I have seen many people contribute with measurements, but they never seem to get added, so there are still only models for nexus 4 and 5. Especially in newer android models this is noticable, the galaxy s5 and s6 generations (and other flagship models of same generation) will get 3.5 meters when the actual distance is around 15-20 meters. 

On galaxy S3 and older the calculations seem pretty accurate, but for S5 we need to multiply the results by 5 to even get usable values.

davidgyoung, 2015-07-13:
@eckoj433, I completely agree with the need to update this.  The only reason for the lack of updates is not enough time.  I have been looking for a volunteer to help augment this database for awhile.  If you would like to help, or know somebody who would, please let me know.

eckoj433, 2015-07-13:
unfortunetly we do not have the greatest divercity of devices in our office, but i could certainly help out with the models that i do have access to (Lg G3, galaxy s3-s4-s5, xperia z3).

davidgyoung, 2016-01-07:
closed due to inactivity

closed 2016-01-07 by davidgyoung