AltBeacon/android-beacon-library #240: locate beacon crashs on nexus 9 build LRX22L

wsyzxcn, 2015-07-16:
here is the error log:
07-16 15:02:11.445: E/AndroidRuntime(12366): Process: com.radiusnetworks.locate, PID: 12366
07-16 15:02:11.445: E/AndroidRuntime(12366): java.lang.NullPointerException: Attempt to invoke virtual method 'int java.lang.Long.intValue()' on a null object reference
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at org.altbeacon.beacon.BeaconTransmitter.startAdvertising(BeaconTransmitter.java:153)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at org.altbeacon.beacon.BeaconTransmitter.startAdvertising(BeaconTransmitter.java:142)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at com.radiusnetworks.locate.TransmitterActivity.startAdvertising(TransmitterActivity.java:487)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at com.radiusnetworks.locate.TransmitterActivity$1$1.run(TransmitterActivity.java:135)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at android.os.Handler.handleCallback(Handler.java:739)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at android.os.Handler.dispatchMessage(Handler.java:95)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at android.os.Looper.loop(Looper.java:135)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at android.app.ActivityThread.main(ActivityThread.java:5221)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at java.lang.reflect.Method.invoke(Native Method)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at java.lang.reflect.Method.invoke(Method.java:372)
07-16 15:02:11.445: E/AndroidRuntime(12366): 	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)

davidgyoung, 2015-07-16:
Thanks for the report.  We'll have a fix up shortly.

davidgyoung, 2015-07-16:
Fixed in 40087fb829f1627f4529fce2f02d9d8b12503fa7.  A new version of the Locate app that includes this fix, Locate 2.5.1 has been uploaded to the Play Store and should be available shortly.

closed 2015-07-16 by davidgyoung