AltBeacon/android-beacon-library #241: Beacons not reported unless Android L scanning disabled

bluetooth-mdw, 2015-07-16:
I have what seems to be a similar problem to Issue 63, with previously working code on my Nexus 5 running 5.1.1 Build LMY48B. I'm using version 2.3.3 of the AltBeacon library

Disabling wi-fi had no affect on the issue for me.

Problem: Beacons were detected and reported the first scan and then rarely if at all after that.

Here's log output from my application showing this:

```
07-16 11:31:59.781    8625-8668/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 2 beacons
07-16 11:32:09.854    8625-8712/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:32:19.926    8625-8739/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:32:30.051    8625-8912/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:32:41.122    8625-9004/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:32:52.134    8625-9015/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:33:03.145    8625-9021/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:33:14.155    8625-9023/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:33:25.172    8625-9025/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:33:36.166    8625-9027/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:33:47.145    8625-9029/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:33:58.160    8625-9045/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:34:09.158    8625-9047/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:34:20.119    8625-9049/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:34:31.156    8625-9052/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:34:42.148    8625-9054/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:34:53.114    8625-9077/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:35:04.149    8625-9097/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:35:15.172    8625-9099/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:35:26.254    8625-9101/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:35:37.161    8625-9103/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:35:48.180    8625-9191/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:35:59.214    8625-9213/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:36:10.135    8625-9215/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:36:21.172    8625-9218/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:36:32.146    8625-9220/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:36:43.133    8625-9222/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:36:54.189    8625-9242/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:37:05.118    8625-9244/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:37:16.137    8625-9531/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:37:27.107    8625-9572/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
07-16 11:37:38.176    8625-9574/? D/.GetYourBeacOn? didRangeBeaconsInRegion() reporting on 0 beacons
```
Adding
```
beaconManager.setAndroidLScanningDisabled(true);
```
has fixed the problem however and now I see:

```
07-16 12:51:46.570   9222-10030/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:51:56.650   9222-10133/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:52:06.719   9222-10201/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:52:16.789   9222-10292/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:52:26.867   9222-10362/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:52:37.078   9222-10432/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:52:48.114   9222-10507/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:52:59.106   9222-10602/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
07-16 12:53:10.128   9222-10680/? D/.GetYourBeacOn﹕ didRangeBeaconsInRegion() reporting on 2 beacons
```
as expected.

I wrote a simple application to test the native Android L scanning APIs and they are working correctly from what I can see. Here's the key code:

```
    public void onScanButton(View view) {
        // check bluetooth is available on on
        if (mBluetoothAdapter == null || !mBluetoothAdapter.isEnabled()) {
            Intent enableBtIntent = new Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE);
            startActivity(enableBtIntent);
            return;
[... 153 lines cut ...]
07-16 12:45:28.102    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0200-0300
07-16 12:45:28.132    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:28.239    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:28.345    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:28.445    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:28.478    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0200-0300
07-16 12:45:28.768    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:28.870    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0200-0300
07-16 12:45:28.878    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:28.980    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:29.251    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0200-0300
07-16 12:45:29.301    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:29.405    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:29.729    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:29.837    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:29.943    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:30.050    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:30.152    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:30.261    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:30.363    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
07-16 12:45:30.399    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0200-0300
07-16 12:45:30.469    6096-6096/com.bluetooth.mwoolley.altbeaconscanner I/AltBeaconScanner? AltBeacon: 00000000000000000000000001000000-0400-0500
```

Let me know if I can help further.

davidgyoung, 2015-07-16:
@bluetooth-mdw, I believe the issue is that the hardware filter the library uses to save battery is not matching your beacons.  The example code you wrote disables these filters, which is probably the difference.  (` filters = new ArrayList<ScanFilter>();`) A few questions:

* When Android L scanning is enabled using the library, do you see the failed detections when the app is in the foreground or the background, or both?
* What is the manufacturer of the AltBeacon you are using?  Does it really use the manufacturer code FFFF? The manufacturer code are the two bytes that come before the BEAC beacon type code in the advertisement:   

0201061BFFFFFFBEAC0000000000000000000000000100000004000500C50000000000000000000000000000000000000000000000000000000000000000

bluetooth-mdw, 2015-07-16:
Hi David

I only tested in foreground so far.

The manufacturer is me :-) I'm using an Intel Edison running a node.js app for one beacon and a Bluegiga DBBLE board running a script I wrote for the other, hence I chose FFFF for a fake manufacturer.

regards

Martin

davidgyoung, 2015-07-16:
A couple of thoughts:

* Can you try using a manufacturer id of 0x0118 (Radius Networks)?  The Android L scan filters built in to the library are automatically set up to use this id.  Radius Networks authorizes the use of its manufacturer code in AltBeacon advertisements, and will allow you to match the filter built into the library.  Because there are 65536 different manufacturer combinations, it is simply not possible to include every combination in the hardware filter -- there just aren't enough of them.  (Hardware filters require using the manufacturer id as part of the filter.)  A similar issue exists with iBeacon on iOS devices, which is why the Apple manufacturer code 0x004c is always used there, regardless of who manufactures the beacon.

* The scan filters are designed to only be used in the background to save battery, not the foreground.  The fact that you say you are not getting detections in the foreground suggest to me that the background flag is being improperly set in the library.  Have you written any special code in your app to set the foreground/background mode?  Are you using the `BackgroundPowerSaver`, or are you calling `beaconManager.setBackgroundMode()`?

bluetooth-mdw, 2015-07-16:
Will do, that's useful information, thanks.

I'm not using BackgroundPowerSaver and I've not called setBackgroundMode.

I'm doing all my beacon detection in an Application which has this class declaration:

```
public class GyboApplication extends Application implements BootstrapNotifier, RangeNotifier, BeaconConsumer {
```

I'll try changing the manufacturer ID and report back.

cheers

Martin

bluetooth-mdw, 2015-07-16:
Success!

With my home made beacons advertising with manufacturer ID 0x0118, I get a consistent series of calls to didRangeBeaconsInRegion with the expected number of beacons detected and without having disabled Android L Scanning.

Thanks for this.... learned something useful.

regards

Martin

bluetooth-mdw, 2015-07-16:
fyi the 0xFFFF value was selected from the official Bluetooth SIG list of company identifiers:

https://www.bluetooth.org/en-us/specification/assigned-numbers/company-identifiers

davidgyoung, 2015-07-16:
@bluetooth-mdw, would you be willing to share your source code?   I'd still like to test to see why this didn't work in the foreground.  You may have uncovered a bug in the library.  If you would be willing to share it privately as a zip file, you can send it to me at david@radiusnetworks.com

closed 2015-07-22 by davidgyoung