AltBeacon/android-beacon-library #242: didRangeBeaconsInRange most of the times returns 0

ImMathanR, 2015-07-21:
I have an issue similar to #163 Most of the time the didRangeBeaconsInRange callback returns 0 beacons. If I use foreground scan period 10000ms the response is some what 0. But I have to use 1000ms or 2000s. But what I am getting is 8 out of 10 times its returning 0. But its not device specific. I have tested in 2 devices. 

Mobile - Nexus 7 - 2013 , Lenovo a7000 
Beacon advertise by 1hz

davidgyoung, 2015-07-21:
@ImMathanR, can you please provide the following information:

* Do you see the same behavior if using the Locate app? https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en
* What library version are you using?
* What kind of beacons are you using, and are they transmitting AltBeacon, iBeacon or some other format?

ImMathanR, 2015-07-22:
In Locate app its working better. 
Library version 2.5
Its iBeacon. 

Apologize I had given wrong information. The Beacons advertising frequency is 2hz and 3hz. 
Now I am not updating any foreground scan periods. I am using the default scan period of 1100ms which I guess.

My scenario is I have to notify exit if the beacon is approximately 13 to 15 meters and have to show entry if the range is below 10 meters. As per your reply for this thread its not reliable for my scenario. Do I have to rely on RSSI.

davidgyoung, 2015-07-22:
@ImMathanR, please understand reliable ranging functionality is dependent on getting multiple packet samples in each interval.  For best results, 10Hz is recommended, and 1 Hz is a bare minimum.  Are you sure the beacons are transmitting at 2-3 Hz, or is it every 2-3 seconds?

Under typical conditions (it varies based on the noisiness of the radio environment, the receiving device, and the distance to the beacon), only 80-90% of beacon packets are detected by the phone.  So in order to get a detection every ranging cycle (default 1100 ms), you need to guarantee that a packet will be detected every 1100ms.

Whether you rely on RSSI or the getDistance() calculation that is based on RSSI probably does not matter.  The most important thing for this to work reliably is you need to make your beacons transmit  at a higher advertising frequency.

ImMathanR, 2015-07-22:
I am using 2 beacons which broadcasting every 2 and 3 beacons. Now I had tested in Samsung Grand 2(Os - 4.4.2). And it worked as what I expected. But It's not working in my nexus and lenovo which is basically lollipop device.

davidgyoung, 2015-07-22:
One point with the Nexus 7 2013 edition is that Bluetooth and WiFi do not work properly when both are enabled at the same time.  You should test on this device with WiFi disabled.  I recommend that folks prompt users to disable Wifi if you detect it is turned on with one of these devices in a beacon-based app.

ymc-sise, 2015-08-20:
I have discovered the same behaviour on my nexus 7 2013 edition with Android 5.1.1(LMY47V). The beacons in the didRangeBeaconsInRegion are empty. Bluetooth and WiFi have to be active at the same due to REST calls when a beacon signal is arriving.

On my Samsung Galaxy A5 Android 5.0.2 (LRX22G) everything works as expected.

davidgyoung, 2015-08-20:
Unfortunately with Bluetooth and WiFi both active on the Nexus 7 2013 edition, this is the behavior you can expect to see.  A hardware limitation mrans the device simply does not detect Bluetooth packets reliably when WiFi is active.

lhunath, 2015-10-01:
In my experience, this is not related to WiFi.  Can we re-open this until ImMathanR has confirmed that disabling WiFi resolves at least his ranging issue?  

In my experience, both the Locate and my own app have issues detecting nearby beacons reliably (they disappear and re-appear, only showing for a fraction of the didRangeBeaconsInRegion calls when I expect them to be there for all of them, the beacon is sitting next to the device, doesn't move and is detected properly by an adjacent iPhone and is also detected properly by other apps on the Android store such as the Estimote and iBeaconDetector app on the same Android device with WiFi enabled without any issues.

lhunath, 2015-10-01:
In addition, disabling and re-enabling Bluetooth on the Android devices causes the beacon to be detected almost immediately, and then it disappears again.

lhunath, 2015-10-02:
I should point out that in my experience, increasing the foreground poll period from 1100 to eg. 4000 ms pretty much resolved the issue for me.  I'm unsure why, since foreground polling should happen with a between period of 0, which I would expect means the poll period itself should only have an effect on how often the polled beacons are parsed.  Since this is not the effect I am seeing, I expect there may be a bug in android-beacon-library in this area.

davidgyoung, 2015-10-02:
@lhunath, do you know how frequently your beacon is advertising?  Any chance you can set `beaconManager.setDebug(true)` then post a LogCat capture over a period where you see the problem?

lhunath, 2015-10-02:
Debugging output at various foreground poll periods and beacon advertisement intervals:
http://stuff.lhunath.com/poll-period-1100-advertisement-100.txt
http://stuff.lhunath.com/poll-period-1100-advertisement-2000.txt
http://stuff.lhunath.com/poll-period-4000-advertisement-100.txt
http://stuff.lhunath.com/poll-period-4000-advertisement-2000.txt 

A small note about my application-level logging in response to altbeacon's callbacks:

If you filter that on AutourBeaconManager, you'll get what my application has determined as a result of what altbeacon is reporting, eg.
in: id1: b9407f30-f5f8-466e-aff9-25556b57fe6d id2: null id3: null, ranged 3 beacons
- [b9407f30-f5f8-466e-aff9-25556b57fe6d, 33726, 48447] (RSSI: -87 / TX: -81, distance: 1.24m, data: [4]), model: {AutourBeacon: uid=225291, version_id=574718, name=ottawa-35}
- [b9407f30-f5f8-466e-aff9-25556b57fe6d, 39884, 11650] (RSSI: -86 / TX: -81, distance: 1.19m, data: [4]), model: {AutourBeacon: uid=225292, version_id=562404, name=ottawa-36}
- [b9407f30-f5f8-466e-aff9-25556b57fe6d, 42039, 26915] (RSSI: -73 / TX: -81, distance: 0.35m, data: [4]), model: {AutourBeacon: uid=224947, version_id=535997, name=ottawa-31}

Indicating that in the region defined by the given id1 and an empty id2 and id3, altbeacon reported 3 beacons in didRangeBeaconsInRegion, and then it describes the beacons.

When it says "Beacon gone", it's because altbeacon hasn't reported that beacon in more than poll period * 2 + 1s time and so we consider the beacon to no longer be nearby.
When it says "Beacon expired", it's because altbeacon hasn't reported that beacon in more than poll period * 3 + 1s time and so we expire the beacon from the UI entirely.

davidgyoung, 2016-01-07:
This thread started with a question from @ImMathanR which appears to me to be related to  the known issue of WiFi/Bluetooth interference on the Nexus 7.  @lhunath commented later, with what looks to me like a different issue of an infrequently advertising beacon.  For best results, a beacon must advertise more often than the scan interval (1100 ms by default).  If you have an infrequently advertising beacon, you need to decrease lengthen the scan interval.  This appears to me to be confirmed by @lhunath based on his comments and attached logs.

Since two issues have been confused in this thread, if either of you would like to continue the discussion on this, please open a new issue with the specifics of what you are seeing.

closed 2016-01-07 by davidgyoung

lhunath, 2016-01-07:
I would like to point out that when the beacon advertises at an interval of 100ms and the poll period is set to 1100ms, I would not expect to see the beacons to not get picked up for 2 or more periods, like my logs indicate.  That seems indicative of a software bug rather than a poll/advertisement period or other configuration issue.

davidgyoung, 2016-01-07:
Fair point, @lhunath.  I did not notice this in that log.  Could you please open a new issue specifically with that case, and include both the full log and an annotation of it showing the timestamps where you don't see detections but would expect to?   Please include your device model, os version, library version and whether or not wifi was turned on during the test.