AltBeacon/android-beacon-library #248: Use PendingIntent and AlarmManager / JobScheduler for background monitoring/ranging

danielesegato, 2015-07-22:
Hi,

this is a feature request / enhancement.

This is very similar to a comment I left in another bug (#151) but I tough it deserved its own issue.

if you run a service in background and want it to consistently keep running all the time you have to use wake lock permission to avoid the device going to sleep.
This is really bad anyway! Because that's how your device save the most of its battery while you don't use it. (With M this become enforced and I would be surprised if the current implementation kept working on it).

If you do NOT use wake lock the system can kill your service any time: usually when it has low resources and need to free some. This means your service can go down and not come up again until the user open the app or reboot the device.

An always running service is a bad thing.

I think the library should use Alarms Manager or new Job Sheduler to check for beacons. This is either more efficient in term of battery usage and remove the need of having to keep your service always up and running.

The other similar problem is the need of keeping a static reference to the callback (listener) somewhere in your app. If the system kills your app because of low resource that instance is gone.

The proper way of handling this, in my opinion, is exactly how the Google FusedLocation API works. It provide 2 manner of registering for events: a listener, meant to be used from the UI / while the user operate with your activity; and a PendingIntent, meant to be used for background notifications.

danielesegato, 2015-07-23:
Something to support my argument: never write an always running background service:

http://stackoverflow.com/questions/2566350/how-to-always-run-a-service-in-the-background
http://www.androidguys.com/2009/09/09/diamonds-are-forever-services-are-not/
https://groups.google.com/forum/#!topic/android-platform/nU0e3aoBlSY

the last one contains comments from an Android Framework developer (Dianne Hackborn) explicitly stating an always running service is a bad practice that no one should use.

The fact is that the system will kill your service.
There's no "IF". It's "WHEN". And when that happen your service will go down and not come back.

This is absolutely WANTED behavior because you do not want to drain the device battery so much.
The problem is that at some point your app will stop getting background notification of beacon getting in range unless the user regularly open it.

davidgyoung, 2015-07-23:
Points taken.  The library currently does use an AlarmManager to reschedule scans in the future, but these are essentially used to kick off a long-running service.  It may be possible to avoid starting up the service in some cases and rely completely on the AlarmManager.  

One challenge to doing this is the design of the bluetooth scanning API on Android 5, which is a key way to saving battery on newer Android devices.   These API provide a method callback when a bluetooth device is discovered matching a hardware filter.  But they won't send an Intent when a matching hardware device is found.  So code must be running in the background to receive the callback, hence the need for a long-running service.

danielesegato, 2015-07-23:
@davidgyoung yes I saw how the API in Android 5 works... And it seems like it is meant to be used with an app open more then a background service.

I understand this is really challenging and I think it may also be worth opening a bug to them about this.

If the bug is backed by a library like AltBeacon it may come to the attention of the right persons.

policybased, 2015-09-17:
Hi,

I agree with  danielesegato.

I don't see a problem with the design of the bluetooth scanning API on Android 5. There is already a background and a foreground mode in the Altbeacon Lib. In the foreground mode it can still work like it is running right now. But in the background mode it should wake up the BeaconService with an alarm. So when the BeaconService is woken up by the AlarmManager or JobScheduler (API21) it should start the CycledLeScanner (bluetooth scanning API) as long as the background scan period is set to. During this time the bluetooth callbacks will be called. When the background scan period is expired then it should stop the CycledLeScanner and finish the cycle work like processExpiredMonitors and then it should stop the BeaconService (stopSelf()), so that it can go to sleep again. During this time the service should not be killed or stopped. Android will not kill this service if there is not a low memory issue. StopSelf just work if there was a startService call. This means the monitoring should not done with the bind / unbind concept.

Where I see the design problem how an application has to handle the monitored beacons. This should not be done with a bind and unbind concept. This should be done similar like the Google GeofencingApi. An application should register each monitoring region with a PendingIntent from the application service class (Or just the service class name to invoke a intent. PendingIntent are for external processes.), which handles the exit and entry triggers. The callback to the application would be just a startService(monitorState.getIntent(). There is no need for this RegionBootstrap class and BootstrapNotifier interface.
There should be just a WakefulBroadcastReceiver (AlarmReceiver class) which handles the alarms in the onReceive method. In the onReceive method there should just be a call of the BeaconService (startWakefulService) and so the BeaconService can call stopSelf.

1. CycledLeScanner should not be responsible for the running time. It should be just responsible to do a bluetooth scan and return the data by a callback.

2. The BeaconService should be responsible for the scanning time.

3. There should be persistent monitored regions (like shared preferences) that they do not have always to be registered again, when the application is restarted. By the way right now they lose the transition state. If a region has been entered the application will get an entry trigger. When the application exit or gets killed after a restart there is again an entry trigger or if the user moves and gets out of this monitored region he will never get an exit trigger.

cheers

dopry, 2018-06-22:
@danielesegato, any progress on this? I'd be happy to review any code you develop. I'm working on https://github.com/petermetz/cordova-plugin-ibeacon and would be happy to review and test upstream PRs.

danielesegato, 2018-06-22:
@dopry I never worked on this, I was an user of the library and just posted a feature request for @davidgyoung (which is the owner of this library). I no longer works with beacon (and hope not to, don't really a fan of the technology).

davidgyoung, 2018-06-22:
Several of these proposed changes have already been incorporated into the library as part of other pull requests.

* Performing scans with the JobScheduler instead of a long-running service was introduced in 2.12 (#484) and is now the default on Android 8+
* Monitored region persistence was introduced in version 2.8 (#315) and improved in #405 

The main item discussed above that is not implemented is:

>> An application should register each monitoring region with a PendingIntent from the application service class (Or just the service class name to invoke a intent. PendingIntent are for external processes.), which handles the exit and entry triggers.

The reason this is not implemented is because the library is not a standalone app installed on the phone like Google Play Services.  It is bundled with a user application and it starts and stops running along with the parent app.  it is therefore not possible for the library to trigger the app to start by sending it an Intent.  

As most of these changes have been implemented as described above, I am closing this issue.  We can re-open if folks think the remaining part that was no implemented is somehow useful.

closed 2018-06-22 by davidgyoung

dopry, 2018-06-22:
@davidgyoung, thanks for the update.

@danielesegato, would it be possible to connect out of band to discuss your experience with beacons?