AltBeacon/android-beacon-library #250: How do I initialize the region at runtime?

danielesegato, 2015-07-23:
Hi,

the documentation say I need to initialize the library in the Application onCreate().

But I have the data needed to initialize it at runtime.

Specifically at runtime (after a remote http service call) I know:
- IF beacon tracking should be started at all
- which regions to monitor

Furthermore I may need to turn it off completely afterward in the lifecycle of the app.

What's the correct way with the current version of the library to handle this use case?

davidgyoung, 2015-07-23:
@danielesegato, please use StackOverflow.com for questions on how to use the library as this forum is reserved for feature requests and reports of bugs within the library itself.

You can see s similar question here:

http://stackoverflow.com/questions/30097586/how-can-i-refresh-my-regionbootstrap-android-beacon-library

If that doesn't fully answer your question, please feel free to open a new one.

closed 2015-07-23 by davidgyoung

danielesegato, 2015-07-23:
Fair enough.
That didn't replied all my question so I created a new one:
http://stackoverflow.com/questions/31588667/how-can-i-start-disable-background-monitoring-for-beacons-at-runtime-with-altb