AltBeacon/android-beacon-library #256: Altbeacon intergration

ccoulton, 2015-08-05:
Hello, I'm with Qwasi and we are currently trying to integrate your altbeacon library with our android open source SDK.

Basically I'm creating a location manager that handles geofences, and beacons, and when Implement the interface onto the manager and bind it i'm getting a null pointer back to the application context.

This is using Android studio

davidgyoung, 2015-08-05:
Hi, @ccoulton, I'd be happy to help look at this, but would need to see the code.  Perhaps you can put this on a branch in your GitHub repo, then create a new issue there describing the problem and linking to the line of your code where it is causing the problem.  You can then link to that issue from here and I will take a look.

ccoulton, 2015-08-05:
Ok I've created a commit with the issue and created a pull request, but cannot create an public issue on that repo.  I have however created a fork with the updates, including this issue https://github.com/ccoulton/android-sdk/issues/1, I basicly took the application demo and put the portions of on resume and onPause in the ActivtyLifecycelcallbacks in AppManager, and put the consumer/ranging functions in the Location manager

davidgyoung, 2015-08-05:
Closing this issue and commenting on the other one.

closed 2015-08-05 by davidgyoung