AltBeacon/android-beacon-library #260: [AltBeacon with BT A2DP]

daniel75kim, 2015-08-10:
Hi, David

I am getting bluetooth crash when using bluetooth headset (streaming) at the same time as using AltBeacon library: 

The issue is reproducible using :
   
   (1) Device : LG G3 (Android : 4.4.4)
   
   (2) Project Environments :
   
       (a) android-beacon-library project
       
       <manifest
       package="org.altbeacon.beacon"
       android:versionCode="1"
       android:versionName="2.5.1" >
       
       
       (b) aapt project
       
       package org.altbeacon.beacon;		
			 public final class BuildConfig {
				   public final static boolean DEBUG = true;
			 }
   
   (3) Steps to reproduce :
   
       Run the AltBeacon reference application is running in foreground and use A2DP bluetooth streaming is running in the background. After a while bluetooth streaming  stops(disconnects) for a while and restarts(reconnect)
   
   (4) Crash Frequency : at least once every 10 minutes, sometimes every 2~3 minutes
        
   (5) Logs : the following log is from with Altbeacon reference application (library v 2.5.1) running and with A2DP bluetooth streaming application (Spotify)
            
		    08-10 12:07:20.129 D/bt-btif (22533): post cmd type:1, size:0, h:7, 
				08-10 12:07:20.129 I/bt-btif (22533): HAL bt_gatt_callbacks->client->scan_result_cb
				08-10 12:07:20.129 D/BtGatt.GattService(22533): onScanResult() - address=D0:8A:9C:CD:EE:72, rssi=-75
				08-10 12:07:20.129 D/BtGatt.btif(22533): btif_gattc_upstreams_evt: Event 4096
				08-10 12:07:20.129 I/bt-btif (22533): HAL bt_gatt_callbacks->client->scan_result_cb
				08-10 12:07:20.129 D/BtGatt.GattService(22533): onScanResult() - address=D3:9F:82:34:6F:BA, rssi=-63
				08-10 12:07:20.129 D/BtGatt.btif(22533): btif_gattc_upstreams_evt: Event 4096
				08-10 12:07:20.129 I/bt-btif (22533): HAL bt_gatt_callbacks->client->scan_result_cb
				08-10 12:07:20.129 D/BtGatt.GattService(22533): onScanResult() - address=E5:A4:44:54:86:1D, rssi=-85
				08-10 12:07:20.129 D/BtGatt.btif(22533): btif_gattc_upstreams_evt: Event 4096
				08-10 12:07:20.129 D/BtGatt.btif(22533): btif_gattc_add_remote_bdaddr device added idx=11
				08-10 12:07:20.129 F/libc    (22533): Fatal signal 11 (SIGSEGV) at 0x00000000 (code=1), thread 22549 (BTIF)
				
				
				08-10 12:07:20.239 I/DEBUG (  277): Build fingerprint: 'lge/g3_open_hk/g3:4.4.2/KVT49L.D858HK10f/D858HK10f.1413522899:user/release-keys'
				08-10 12:07:20.239 I/DEBUG (  277): Revision: '10'
				08-10 12:07:20.239 I/DEBUG (  277): pid: 22533, tid: 22549, name: BTIF  >>> com.android.bluetooth <<<
				08-10 12:07:20.239 I/DEBUG (  277): signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 00000000
				08-10 12:07:20.239 W/NativeCrashListener(  974): Couldn't find ProcessRecord for pid 1530015793     

        
	Full logs with AltBeacon debugging enabled :
       
        https://www.dropbox.com/sh/b6umg1os7f346g9/AAD-wC88Vj9DeqlCSgcZVi0ba?dl=0   
	

This seems to be similar to other errors reported with lescan :    
        - http://stackoverflow.com/questions/22048721/bluetooth-share-has-stopped-working-when-performing-lescan
	- http://codegur.com/22048721/bluetooth-share-has-stopped-working-when-performing-lescan
							
			 
BTW, I have another test device Moto G  Moto G (Android4.4.4) and this issue does not occur on it.

davidgyoung, 2015-08-10:
Thanks for the detailed report.  This does appear to be a firmware/driver issue with the LG G3 4.4.4 build.  I suspect that simply doing a bluetooth LE scan at the same time as streaming causes this intermittent problem (or perhaps the act of starting or stopping scanning).  Other than a fix from Android or LG (which seems unlikely, given that the OS has moved on from 4.4.4) the only way to avoid this might be to detect if streaming is active (not sure this is possible) and making the Android Beacon Library  not do scans.

daniel75kim, 2015-08-11:
Thank you for the comment.
If this issue is related to firmware/driver(Broadcom Bluetooth IC) issue of LG G3 4.4.4 build, it seems to be 50:50 to assume that this issue will be resolved by Android 5.x Lollipop upgrade. 
Other than the firmware/driver perspective, is there any chance that AltBeacon library running with Android 5.x Lollipop APIs is able to resolve this issue ?

davidgyoung, 2015-08-18:
Yes, upgrading to 5.x would replace the firmware, so this may resolve the issue.

closed 2015-08-18 by davidgyoung