AltBeacon/android-beacon-library #265: getBluetoothName() is always null

atiffarrukh, 2015-08-17:
I am trying to get the device name for the iBeacon I am using (Wellcore iBeacon). I am able to get all the other data except for the 'getBluetoothName()', its always null.

Currently using this beacon layout
"m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"
Also tried following
"m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25" (Doesn't detect beacon)
"m:0-3=0215,i:4-19,i:20-21,i:22-23,p:24-24" (Doesn't detect beacon)
"m:2-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24" (Gets all data except for device name)
"m:0-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24" (Gets all data except for device name)
"m:2-3=4c000215,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25" (Gets all data except for device name)

davidgyoung, 2015-08-18:
The bluetooth name has nothing to do with the beacon layout.  It populated automatically by the operating system if the BLE device sends out a scan response containing a friendly name.  Because you are not seeing the name for the layouts that work, that means the device probably isn't transmitting it.

Apologies, I have to redact some parts of the layouts you show as they contain proprietary information that must be kept out of this open source repo.

atiffarrukh, 2015-08-18:
But I do get the device name using other BLE Scanners that are available in Google Play. Though I get the name after "bonding" with BLE.

davidgyoung, 2015-08-18:
Connecting to the beacon over GATT is another way to read its name, yes.  Understand, however, that scanning for bluetooth beacons is connectionless, reading only the advertisements that are broadcast by the beacon. That is why using the library a scan response transmitted by the beacon is generally the only way to see the name populated.

atiffarrukh, 2015-08-18:
So, I have to make GATT connection in order to get its name, and other data like manufacturer etc? Right?

davidgyoung, 2015-08-19:
correct

closed 2015-08-19 by davidgyoung