AltBeacon/android-beacon-library #266: Locate Beacon app returns different from checkTransmissionSupported() result

and7ey, 2015-08-17:
When I try to transmit an iBeacon with Locate Beacon app, I get an error

    Failed to start beacon. Too many advertisers - error code 2.

But when I test BLE peripheral mode support on the same device with

    int result = BeaconTransmitter.checkTransmissionSupported(getBaseContext());

it returns zero, which is equal to `BeaconTransmitter.SUPPORTED`. 

The device used is Sony Z3 Compact (D5803, Android 5.1.1 - 23.4.A.0.546), in accordance with the Devices List (http://altbeacon.github.io/android-beacon-library/beacon-transmitter-devices.html) D5803 with older Android versions (5.0.2) don't support it.

Why do I get `BeaconTransmitter.SUPPORTED` then?

davidgyoung, 2015-08-17:
Two possibilities:

1. There is another app that has already started transmitting and the device supports only one active transmission at a time.  It's possible this could be an earlier attempt to start transmitting with your app, and it just wasn't stopped properly.  Toggling bluetooth off/on or reinstalling your app may fix this.  (I've seen this before.)

2. There is something wrong with the device such that it cannot transmit even though it reports it can.  Since other folks have reported this device being able to transmit (per the list you mention) this is a less likely possibility.

closed 2015-08-17 by davidgyoung

and7ey, 2015-08-17:
Thanks for prompt reply.

That's interesting - the following returns `false` on the same device in my app - 

    mBluetoothAdapter.isMultipleAdvertisementSupported()
    mBluetoothAdapter.isOffloadedFilteringSupported()
    mBluetoothAdapter.isOffloadedScanBatchingSupported()

Looks like your library uses the same approach to identify BLE peripheral mode support. Strange that it returns 0. Though that this is the library-related issue.

davidgyoung, 2015-08-18:
It is important to understand that just because isMultipleAdvertisementSupported() returns false does not mean that the device cannot advertise at all -- it means either that it cannot advertise at all *OR* that it can only transmit one advertisement at a time.  Many Motorola devices (Moto X, Moto G) behave this way.  The way you are supposed to check if advertising is supported at all is to get an instance of BluetoothLEAdvertiser. If the call to get an instance fails, then it is not supported.

and7ey, 2015-08-19:
Since this is not android-beacon-library issue, I've raised further question at StackOverflow - http://stackoverflow.com/questions/32092902/why-ismultipleadvertisementsupported-returns-false-when-getbluetoothleadverti