AltBeacon/android-beacon-library #269: Android Marshmallow Exception

gotama, 2015-08-27:
Hi,

We have been running AltBeacon on the Android M preview build on a Nexus 5, and have noticed this exception.  

08-26 20:05:50.106    9010-9010/? E/BluetoothLeScanner﹕ application registeration exception
   android.os.DeadObjectException
           at android.os.BinderProxy.transactNative(Native Method)
           at android.os.BinderProxy.transact(Binder.java:503)
           at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerClient(IBluetoothGatt.java:873)
           at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegisteration(BluetoothLeScanner.java:255)
           at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:147)
           at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:106)
           at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.deferScanIfNeeded(CycledLeScannerForLollipop.java:105)
           at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:165)
           at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:261)
           at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:234)
           at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:230)
           at android.os.Handler.handleCallback(Handler.java:739)
           at android.os.Handler.dispatchMessage(Handler.java:95)
           at android.os.Looper.loop(Looper.java:148)
           at android.app.ActivityThread.main(ActivityThread.java:5415)
           at java.lang.reflect.Method.invoke(Native Method)
           at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:725)
           at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:615)
08-26 20:05:50.120    9010-9010/? E/CycledLeScannerForLollipop﹕ Scan Failed
08-26 20:05:50.120    9010-9010/? E/CycledLeScannerForLollipop﹕ Scan Failed

If you require any more information let me know... Running the latest altbeacon library 2+

Regards,

James

davidgyoung, 2015-08-30:
@gotama, under what circumstances do you get this exception?  Does it happen immediately?  Does it happen only when the app is in the background?

gotama, 2015-09-04:
It happens immediately on every scan.

davidgyoung, 2015-09-08:
Can you please try this with the reference application here?  https://github.com/AltBeacon/android-beacon-library-reference

I have attempted to reproduce this issue under Android M 6.0 Build Number MPA44 on the Nexus 9, but I do not see the same exception.

gotama, 2015-09-09:
Hi David,

Ok i ran the reference app on the device and same issue is there.

Build Number MPZ79M on Nexus 5

I dont have to do anything, as i start the app up exception is shown in the logcat:

09-09 09:01:26.027  18322-18322/org.altbeacon.beaconreference D/BluetoothAdapter﹕ STATE_ON
09-09 09:01:26.027  18322-18322/org.altbeacon.beaconreference D/BluetoothLeScanner﹕ could not find callback wrapper
09-09 09:01:26.028  18322-18322/org.altbeacon.beaconreference D/BluetoothAdapter﹕ STATE_ON
09-09 09:01:26.029  18322-18322/org.altbeacon.beaconreference E/BluetoothLeScanner﹕ application registeration exception
    android.os.DeadObjectException
            at android.os.BinderProxy.transactNative(Native Method)
            at android.os.BinderProxy.transact(Binder.java:503)
            at android.bluetooth.IBluetoothGatt$Stub$Proxy.registerClient(IBluetoothGatt.java:873)
            at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.startRegisteration(BluetoothLeScanner.java:255)
            at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:147)
            at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:106)
            at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop.startScan(CycledLeScannerForLollipop.java:182)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scanLeDevice(CycledLeScanner.java:186)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.finishScanCycle(CycledLeScanner.java:261)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner.scheduleScanCycleStop(CycledLeScanner.java:234)
            at org.altbeacon.beacon.service.scanner.CycledLeScanner$1.run(CycledLeScanner.java:230)
            at android.os.Handler.handleCallback(Handler.java:739)
            at android.os.Handler.dispatchMessage(Handler.java:95)
            at android.os.Looper.loop(Looper.java:148)
            at android.app.ActivityThread.main(ActivityThread.java:5415)
            at java.lang.reflect.Method.invoke(Native Method)
            at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:725)
            at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:615)
09-09 09:01:26.029  18322-18322/org.altbeacon.beaconreference E/CycledLeScannerForLollipop﹕ Scan Failed
09-09 09:01:26.029  18322-18322/org.altbeacon.beaconreference E/CycledLeScannerForLollipop﹕ Scan Failed

davidgyoung, 2015-10-13:
Are you still seeing this on the release version of Android 6?  I have not been able to reproduce.

gotama, 2015-10-13:
We don't have a marshmallow build any more, so unable to test again. I will be running the app on marshmallow devices soon hopefully next month some time.

Will keep you posted, with regards to to your questions I will get back to you on that.

Regards 
James

Wrywulf, 2015-10-28:
Could this issue be related: https://code.google.com/p/android-developer-preview/issues/detail?id=2935

davidgyoung, 2015-10-28:
@Wrywulf, this does look like the core issue.  Since they have market it as fixed, I'm closing this ticket now.  If somebody reproduces with a release build of Android 6 we can reopen.

closed 2015-10-28 by davidgyoung

gotama, 2015-10-28:
Yes David, seems it was only happening on the M Preview build.  Not seeing it on any of the Marshmallow devices at this time

Wrywulf, 2015-10-28:
Well, I'm seing it ('it' being the DeadObjectException) on an LG G3 running API 21, so even though it may be fixed in the native bluetooth service in the Marshmallow release, a workaround should perhaps be investigated.