AltBeacon/android-beacon-library #271: RoadMap

yodutouf, 2015-08-31:
Is there any roadmap document?
I am ready to contribute, but I need to get more visibility to convince my company.

davidgyoung, 2015-09-02:
Hi, @yodutouf.  Thanks for the offer to contribute!  We're always looking for folks who can help.

While we don't have a formal roadmap document here are the upcoming plans for development with the library:

* Android M Preparation (target October 2015)
  - Test background detection on Android M and adjust as needed for OS changes.
  - Support runtime permission model changes coming in Android M.
* Model Distance Calculation Rework
  - Switch to Path Loss Distance Calculations per https://github.com/AltBeacon/android-beacon-library/pull/251
* Background Detection Redesign per https://github.com/AltBeacon/android-beacon-library/issues/151
* Support multiple AltBeacon manufacturer codes per https://github.com/AltBeacon/android-beacon-library/issues/229
* Bug fixes (other open issues: https://github.com/AltBeacon/android-beacon-library/issues?q=is%3Aopen+is%3Aissue)

If you are interested in helping with one of these, feel free to contact me at david@radiusnetworks.com

davidgyoung, 2016-01-07:
closing due to inactivity

closed 2016-01-07 by davidgyoung