AltBeacon/android-beacon-library #272: Eddystone not returning correct url

stefansgit, 2015-08-31:
I have a Kontakt.io beacon broadcasting an URL http://google.com. The UrlBeaconUrlCompressor uncompress returns http://google.comedu.comKontakt. If I use UrlBeaconUrlCompressor to compress  http://google.com and then uncomress it works fine. I think the problem is when I compress the number of bytes the result is exaclty the same as the size of the string (e.g 7 bytes) while the beacon.getId1() returns 17 bytes every time.

The following code is used:

if (beacon.getServiceUuid() == 0xfeaa && beacon.getBeaconTypeCode() == 0x10) {
                // This is a Eddystone-URL frame
                String url = UrlBeaconUrlCompressor.uncompress(beacon.getId1().toByteArray());
}

A reason to suspect error in UrlBeaconUrlCompressor and not Kontaktis.io protocol is that the chrome app physical web gives the correct URL-string.

davidgyoung, 2015-08-31:
Thanks for this report, @stefansgit.  I suspect this is being caused by the scan response with the "Kontakt" name suffixed to the scan data.  

In order to get this fixed, it would be very useful if you could turn on low level debugging in the library with `BeaconManager.setDebug(true);`  Then scan for this url again and attach the result of the log to this issue.

SilkSama, 2015-09-22:
Hi @davidgyoung,
I have the same problem but with estimote beacon.
My beacon broadcast the url http://www.google.com and the UrlBeaconUrlCompressor returns http://www.google.com.info/.eduEST.net/.net/

Thank you in advance for your help

SilkSama, 2015-09-22:
here the identifier built with the UrlBeaconUrlCompressor compress method : 0x00676f6f676c652e6672
here the identifier detected by the library : 0x00676f6f676c652e667204094553540303
So the beacon identifier is not correctly returns by the library when it scans beacons.
As @stefansgit says the id returned has always 17 bytes.

SilkSama, 2015-09-24:
I confirm with an url encoded on 17 bytes, there is no problem. The id returns by the library after decoding corresponds to the url associated with the beacon.

davidgyoung, 2015-09-24:
These reports indicate a problem with the BeaconParser.  It should never return an identifier longer than 17 bytes (although fewer are legal for a URL beacon)

@SilkSama, @stefansgit, I have been unable to reproduce the cause of this problem.  Two things would be very helpful in figuring this out:

1. Please post the `BeaconParser` layout expression you are using to parse the URL beacon.
2. Please turn on debugging with `beaconManager.setDebug(true);` then capture a line from logcat that shows the byte sequence of the URL beacon causing the problem.  It should looks something like below.  Please post this line.  (Note:  you will know it is the proper line if it contains the URL beacon signature byte sequence of aafe10)

09-23 19:39:19.470  3766  3802 D BeaconParser: Processing pdu type FF: 0201060303aafe0d16aafe10e70102030405060708090a0b0c0d0e0f0102030405060708090a0b0c0d0e0f00000000000000000000000000000000000000 with startIndex: 5, endIndex: 22

davidgyoung, 2015-09-24:
I think I see the problem.  The beacon layout needs to have a "v" suffix on it to make the identifier length be variable.  This suffix was inadvertently missing from the documentation page here: http://altbeacon.github.io/android-beacon-library/eddystone-how-to.html  I will fix that.

So instead of: "s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20"

Please use: "s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20v"

Let me know if this solves the issue.

SilkSama, 2015-09-24:
Hi @davidgyoung,
it solves the problem ;)
Thank you for your help.

davidgyoung, 2015-09-24:
I have added common public layout strings in this commit https://github.com/AltBeacon/android-beacon-library/commit/01019700107021fb8dc7f4284cb5b0b6cb279a78  as constants so it is easier for folks to use the correct layout strings.  So now you will be able to do:

     parser.setBeaconLayout(BeaconParser.EDDYSTONE_URL_LAYOUT);

closed 2015-09-24 by davidgyoung