AltBeacon/android-beacon-library #275: Samsung s4 - BootstrapNotifier works even when bluetooth is off

navigator2904, 2015-09-03:
Hello! 
I have no idea how is that possible. I'm testing the app on the Nexus 5 and Galaxy s4. 
Nexus 5 works as expected, but Galaxy s4 behaves very weird. It runs 5.0.1 and BootstrapNotifier works even when bluetooth is off. 
Besides that it frequently drops out from the region, but I put my beacon on the top of the phone. 
But still exit/enter region events occurred every 10-15 seconds. 

Any advice is aapreciated

davidgyoung, 2015-09-03:
Do you see the same if you use an off the shelf beacon detector that uses this library like Locate? 
 https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en&referrer=utm_source%3Dgoogle%26utm_medium%3Dorganic%26utm_term%3Dlocate+beacon+play+store&pcampaignid=APPU_1_5EjoVeiMKoiS-wHb452oCA
Does it detect beacons on the S4 with Bluetooth off?

Does the beacon detected drop out periodically?

davidgyoung, 2015-10-13:
Closing due to inactivity.

closed 2015-10-13 by davidgyoung