AltBeacon/android-beacon-library #276: Moto G (XT1079) with 4G LTE, Android 5.0.2: Beacon Transmission interval resumed to default ~1Hz after one minute of 10Hz transmission

jamescfli, 2015-09-05:
Dear David,

The situation I encountered with AltBeacon was that when I set my BeaconTransmitter to send beacons with highest frequency, i.e.

    mBeaconTransmitter.setAdvertiseMode(AdvertiseSettings.ADVERTISE_MODE_LOW_LATENCY);    
    // should be with ~10Hz

It did work in the first place by sending 10 beacons / sec, but after a while (about one minute later) my mobile went back to send ~1 beacon per second (equivalent to *AdvertiseSettings.ADVERTISE_MODE_LOW_POWER*). I need to restart my App in order to get back to the low latency beacon transmission. Just for the record, my mobile is MOTO G (XT1079) with Android Version 5.0.2.

Your advice is highly appreciated. Thanks.

Sincerely,
James

davidgyoung, 2015-09-05:
Thanks for this report, @jamescfli .  I have not able to reproduce this using the procedure below.  Can you please try to use the same procedure and tell me what you see?  The main thing would be to test using a copy of the Locate Beacon app from the Play Store, which uses the same API you describe above.

Tested with Beacon Locate app from the Google Play Store on:

Model number: Moto G
Android Version: 5.0.2
Build Number: LXB22.99-36

I transmitted an Eddystone-UID frame and left it running for approximately one hour.  I used an iPod touch to pick up the advertisements and count them.  After one hour, I was still seeing multiple advertisements per second.  (I almost never saw 10 advertisements, but I usually saw several -- this is to be expected as I was in a noisy bluetooth environment in Radius Networks HQ).

jamescfli, 2015-09-05:
Thanks David for your quick response. I will double-check it.

In addition, what Advertise_Tx_Power_Level did you apply for your test?
- AdvertiseSettings.ADVERTISE_TX_POWER_HIGH -56 dBm @ 1 meter with Nexus 5
- AdvertiseSettings.ADVERTISE_TX_POWER_LOW -75 dBm @ 1 meter with Nexus 5
- or AdvertiseSettings.ADVERTISE_TX_POWER_MEDIUM -66 dBm @ 1 meter with Nexus 5?

BTW, the description from Line 99 to 101 in [BeaconTransmitter.java](https://github.com/AltBeacon/android-beacon-library/blob/33203e7db7944cea9270fe08bf74c50adf9475f8/src/main/java/org/altbeacon/beacon/BeaconTransmitter.java)

*AdvertiseSettings.ADVERTISE_MODE_LOW_LATENCY 10 Hz*
*AdvertiseSettings.ADVERTISE_MODE_LOW_POWER 1 Hz*

seems to be contradicted with the second table in [README.md](https://github.com/AltBeacon/altbeacon-transmitter-android/blob/master/README.md) for [AltBeacon Transmitter project](https://github.com/AltBeacon/altbeacon-transmitter-android)

ADVERTISE_MODE_LOW_LATENCY	approx *1 Hz*
ADVERTISE_MODE_LOW_POWER	approx *10 Hz*

Could you please clarify which one is correct? Thanks.

davidgyoung, 2015-09-05:
For the test I did, I left the Locate settings on its defaults:

    AdvertiseSettings.ADVERTISE_MODE_LOW_LATENCY
    AdvertiseSettings.ADVERTISE_TX_POWER_HIGH

You are correct about the inconsistency in the documentation for the AltBeacon Transmitter project.  This project is based on the first Android L Preview release, and the documentation is correct for that version of the APIs.  Google fixed this in the release versions of Android 5.  I have added notes to the documentation that hopefully makes this clear.  Bottom line:  the documentation of the settings in BeaconTransmitter for this project are correct.  `AdvertiseSettings.ADVERTISE_MODE_LOW_LATENCY` makes a device transmit at 10 Hz.

jamescfli, 2015-09-08:
Hi David,

Thanks for your further clarification. I have tried the same program on Samsung S6; it works as expected. So I believe your code is fine. The issue would be on my Moto G (XT1079) then. Obviously, it did not follow instructions from 

    BeaconTransmitter#setAdvertiseMode()

and

    BeaconTransmitter#setAdvertiseTxPowerLevel()

My guess is something different has been done by the OEM which changed the behavior of Bluetooth LE Adapter.

closed 2015-09-15 by jamescfli