AltBeacon/android-beacon-library #277: BeaconLayoutException with BeaconLayout "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24,d:-4--3"

NabilNoaman, 2015-09-06:
Hi All ,

I am trying to get battery status of our own beacon which is transmitting in iBeacon mode.

As shown below Battery Status byte(from index -4 to -3)  is defined before company identifier "4C00" (index 0).

![Image of Payload]
(https://lh5.googleusercontent.com/-cKf1geITGOFMPW0At0FLTBfh_Ow5xBfSoXWxnlX2zzrR1Ig0MibHAHaz-Z4s3X-NMzuRrOgxE-sn5KuP-5OPwN3-OFXVf1oQVI=w2162-h1112)

I got next exception :  
Caused by: org.altbeacon.beacon.BeaconParser$BeaconLayoutException: Cannot parse beacon layout term: d:-4--3

How can i define my BeaconLayout to be able to get bytes with negative index ?
Or Can I get bytes before company identifier "4C00" without changing BeaconLayout  and How to get it ?

Thanks

davidgyoung, 2015-09-06:
Beacon layout expressions don't support accessing information at negative offsets.  I don't think it makes sense to do so -- the only information before the manufacturer code in manufacturer advertisements is:

```
PDU Length (1 byte)
Flags PDY Type (1 byte)
Flags (1 byte)
PDU Length (1 byte)
Manufacturer type (value 0xff, 1 byte)
```

closed 2015-09-06 by davidgyoung

NabilNoaman, 2015-09-07:
Thanks David ,

It seems the image is not shown here is the [link](https://lh4.googleusercontent.com/x0DMzf8Y9dSjLB_vESJ8zKkkOPTxJO4GBXt0AxyEzQb80WhnjazxYj6GmM9tPK9wr2hSBnZng5IKBm0xjGdeBf_5HLDxwA=w2560-h1228)

1. Can I get direct access to manufacturer advertisements from the AltBeacon library ?

2. Can I get access to Bluetooth advertisement payload ( manufacturer advertisements + Apple specific iBeacon information.) from the library ? 

If I could't get access to manufacturer advertisements ,  I appreciate if you guide me how to get it as shown using [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en).

Note: If i removed the data field from BeaconLayout ```
 "m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"
``` library works perfect.

Thanks

davidgyoung, 2015-09-07:
If you want raw access to the advertisement data, you can override the `BeaconParser` class, and put custom code in the `fromScanData` method.  You then have to register your custom `BeaconParser` class with the `BeaconManager`.  You can see an example of this here:  https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/AltBeaconParser.java

Apologies, but I had to redact your comment because the BeaconLayout expression you show is proprietary.  As an open source project, we cannot publish details of other companies' intellectual property in this forum without permission.

NabilNoaman, 2015-09-08:
Thanks David appreciate your help :)