AltBeacon/android-beacon-library #280: OutOfMemoryError from onLeScan

daniel75kim, 2015-09-10:
I was running App that included AltBeacon library(ver2.5.1) with the following environments and encountered the fatal exception : OutOfMemoryError. 

Device : GalaxyS5 (SM-G900F)
Android Version : 4.4.2
AltBeacon library version : 2.5.1

App is running with background and foreground with the following cycle. 
- mBeaconManager.setBackgroundScanPeriod(1500);
- mBeaconManager.setForegroundScanPeriod(1500);
- mBeaconManager.setBackgroundBetweenScanPeriod(5*1000);   
- mBeaconManager.setForegroundBetweenScanPeriod(5*1000); 

Could you share your opinion about if it is already known issue and happened in specific devices
or if there is any workaround to resolve it. 

* Fatal exception dump : 

E/JavaBinder( 1963): *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
E/JavaBinder( 1963): java.lang.RuntimeException: Out of memory
E/JavaBinder( 1963): 	at android.os.Binder.execTransact(Binder.java:422)
E/JavaBinder( 1963): 	at dalvik.system.NativeStart.run(Native Method)
E/JavaBinder( 1963): Caused by: java.lang.OutOfMemoryError: pthread_create (stack size 16384 bytes) failed: Try again
E/JavaBinder( 1963): 	at java.lang.VMThread.create(Native Method)
 E/JavaBinder( 1963): 	at java.lang.Thread.start(Thread.java:1029)
E/JavaBinder( 1963): 	at java.util.concurrent.ThreadPoolExecutor.addWorker(ThreadPoolExecutor.java:920)
E/JavaBinder( 1963): 	at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1327)
E/JavaBinder( 1963): 	at android.os.AsyncTask.executeOnExecutor(AsyncTask.java:590)
E/JavaBinder( 1963): 	at org.altbeacon.beacon.service.BeaconService$1.onLeScan(BeaconService.java:304)
E/JavaBinder( 1963): 	at org.altbeacon.beacon.service.scanner.CycledLeScannerForJellyBeanMr2$2.onLeScan(CycledLeScannerForJellyBeanMr2.java:80)
E/JavaBinder( 1963): 	at android.bluetooth.BluetoothAdapter$GattCallbackWrapper.onScanResult(BluetoothAdapter.java:2116)
E/JavaBinder( 1963): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:84)
E/JavaBinder( 1963): 	at android.os.Binder.execTransact(Binder.java:404)
E/JavaBinder( 1963): 	... 1 more
D/BluetoothAdapter( 1984): onScanResult() - Device=DE:25:5A:B5:DE:E2 RSSI=-76
D/BtGatt.btif(21049): btif_gattc_upstreams_evt: Event 4096
D/BtGatt.btif(21049): btif_gattc_add_remote_bdaddr device added idx=6
D/BtGatt.GattService(21049): onScanResult() - address=1C:1A:C0:7F:CB:EF, rssi=-84
D/BluetoothAdapter( 1963): onScanResult() - Device=1C:1A:C0:7F:CB:EF RSSI=-84
W/libc    ( 1963): pthread_create failed: couldn't allocate 1040384-byte stack: Out of memory
E/dalvikvm( 1963): pthread_create (stack size 16384 bytes) failed: Try again
E/Binder  ( 1963): Caught an OutOfMemoryError from the binder stub implementation.
E/Binder  ( 1963): java.lang.OutOfMemoryError: pthread_create (stack size 16384 bytes) failed: Try again
E/Binder  ( 1963): 	at java.lang.VMThread.create(Native Method)
E/Binder  ( 1963): 	at java.lang.Thread.start(Thread.java:1029)
E/Binder  ( 1963): 	at java.util.concurrent.ThreadPoolExecutor.addWorker(ThreadPoolExecutor.java:920)
E/Binder  ( 1963): 	at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1327)
E/Binder  ( 1963): 	at android.os.AsyncTask.executeOnExecutor(AsyncTask.java:590)
E/Binder  ( 1963): 	at org.altbeacon.beacon.service.BeaconService$1.onLeScan(BeaconService.java:304)
E/Binder  ( 1963): 	at org.altbeacon.beacon.service.scanner.CycledLeScannerForJellyBeanMr2$2.onLeScan(CycledLeScannerForJellyBeanMr2.java:80)
E/Binder  ( 1963): 	at android.bluetooth.BluetoothAdapter$GattCallbackWrapper.onScanResult(BluetoothAdapter.java:2116)
E/Binder  ( 1963): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:84)
E/Binder  ( 1963): 	at android.os.Binder.execTransact(Binder.java:404)
E/Binder  ( 1963): 	at dalvik.system.NativeStart.run(Native Method)

daniel75kim, 2015-10-13:
The same issue happened in Android version 5.0 with the same device (GalaxyS5 : SM-G900F).

davidgyoung, 2015-10-13:
@daniel75kim, can you please try to reproduce this issue with the reference application?  https://github.com/AltBeacon/android-beacon-library-reference

If you can get a crash log with that app it will help eliminate anything specific to the two apps you mention.  One possibility, of course, is that something non-beacon related in the app is using all the memory, then the beacon scanning is the last thing that pushes it over the edge.  If you can reproduce the problem with the reference app it will eliminate this possibility.

davidgyoung, 2016-01-07:
closing due to inactivity

closed 2016-01-07 by davidgyoung

punishermalade, 2016-12-06:
Hi, I'm facing an OutOfMemory error as well. I do agree with the conclusion that something is filling the memory and probably not related to AltBeacon. 

Maybe this stack trace might be useful.

```
Android: 5.0
Android Build: LRX21R.A1447078419
Manufacturer: LGE
Model: LG-D855
Thread: main-1
CrashReporter Key: B320F89A-783B-4984-AB6B-3A9F3DCA5CEE4D79F2ED
Start Date: Mon Dec 05 17:12:16 GMT+08:00 2016
Date: Mon Dec 05 17:18:10 GMT+08:00 2016

java.lang.OutOfMemoryError: pthread_create (1040KB stack) failed: Try again
	at java.lang.Thread.nativeCreate(Native Method)
	at java.lang.Thread.start(Thread.java:1063)
	at java.util.concurrent.ThreadPoolExecutor.addWorker(ThreadPoolExecutor.java:920)
	at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1327)
	at android.os.AsyncTask.executeOnExecutor(AsyncTask.java:590)
	at org.altbeacon.beacon.service.BeaconService$1.onLeScan(BeaconService.java:354)
	at org.altbeacon.beacon.service.scanner.CycledLeScannerForLollipop$4.onScanResult(CycledLeScannerForLollipop.java:289)
	at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:330)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:135)
	at android.app.ActivityThread.main(ActivityThread.java:5272)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:909)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:704)
```

tekinalper, 2017-02-22:
Manufacturer: Samsung
Model: SM-A700F
Version: 4.4.4

```
Fatal Exception: java.lang.OutOfMemoryError: pthread_create (stack size 16384 bytes) failed: Try again
       at java.lang.VMThread.create(VMThread.java)
       at java.lang.Thread.start(Thread.java:1029)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.<init>(CycledLeScanner.java:67)
       at org.altbeacon.beacon.service.scanner.CycledLeScannerForJellyBeanMr2.<init>(CycledLeScannerForJellyBeanMr2.java:18)
       at org.altbeacon.beacon.service.scanner.CycledLeScanner.createScanner(CycledLeScanner.java:94)
       at org.altbeacon.beacon.service.BeaconService.onCreate(BeaconService.java:201)
       at android.app.ActivityThread.handleCreateService(ActivityThread.java:2941)
       at android.app.ActivityThread.access$1900(ActivityThread.java:174)
       at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1407)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:146)
       at android.app.ActivityThread.main(ActivityThread.java:5756)
       at java.lang.reflect.Method.invokeNative(Method.java)
       at java.lang.reflect.Method.invoke(Method.java:515)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1291)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1107)
       at dalvik.system.NativeStart.main(NativeStart.java)
```

punishermalade, 2017-02-22:
This error was happening because the scan rate was every 1 second. Try reducing the number of time the scan happens and you should not having this error anymore.

tekinalper, 2017-02-22:
ScanPeriod is 6600 miliseconds and BetweenScanPeriod is 400 msecs in my case.