AltBeacon/android-beacon-library #282: Stuck in Bluetooth crash recovery

davidgyoung, 2015-09-14:
Issue moved from outdated Library repo: https://github.com/RadiusNetworks/android-ibeacon-service/issues/38#issuecomment-140187586 as reported by @jessolo

> I am getting this error on Galaxy Note 3 and beacon detection is not starting. This is the error I am getting. What is the issue with this? I am using AltBeacon version 2.1. Has this been fixed in later releases?
>
> 09-14 15:44:21.533 4488-4488/com.cvs.launchers.cvs W/BeaconService﹕ Skipping scan because crash recovery is in progress.

davidgyoung, 2015-09-14:
@jessolo, can you please attach a longer LogCat excerpt (5 min minimum) from the time of the Bluetooth crash until you see this log entry?  I'd appreciate the help as this issue is specific to the state of the device and is not easy to reproduce.  

Also, can you please tell me your device OS version?

jessolo, 2015-09-14:
I am always getting the same error. Android version 4.3. Pls find the detailed stack trace.
09-14 17:49:24.409  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:24.409  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:25.009    1134-1134/? D/STATUSBAR-NetworkController﹕ refreshSignalCluster: data=-1 bt=false
09-14 17:49:25.039    1134-1134/? D/STATUSBAR-IconMerger﹕ checkOverflow(360), More:false, Req:false Child:3
09-14 17:49:25.169      300-300/? E/SMD﹕ DCD ON
09-14 17:49:25.519  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:25.519  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:26.479     689-1957/? D/SSRMv2:Monitor﹕ SIOP:: AP = 300, Prev AP = 300, Duration = 10157
09-14 17:49:26.629  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:26.629  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:27.729  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:27.739  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:28.169      300-300/? E/SMD﹕ DCD ON
09-14 17:49:28.679      689-923/? D/BatteryService﹕ update start
09-14 17:49:28.689      689-923/? D/BatteryService﹕ level:78, scale:100, status:2, health:2, present:true, voltage: 4128, temperature: 267, technology: Li-ion, AC powered:false, USB powered:true, Wireless powered:false, icon:17303563, invalid charger:0, online:4, charge type:1, current avg:450
09-14 17:49:28.689      689-863/? D/BatteryService﹕ Sending ACTION_BATTERY_CHANGED.
09-14 17:49:28.699      689-864/? D/CMASMessage﹕ [update] mShowingCMASText is false
09-14 17:49:28.709      689-864/? D/CMASMessage﹕ [update] mShowingCMASText is false
09-14 17:49:28.709    1134-1134/? D/STATUSBAR-BatteryController﹕ onReceive() - ACTION_BATTERY_CHANGED
09-14 17:49:28.709    1134-1134/? D/STATUSBAR-BatteryController﹕ onReceive() - BATTERY_STATUS_CHARGING:
09-14 17:49:28.739    1549-1549/? I/BusinessExecutor﹕ registered Task#671 : BatteryEventSource$1
09-14 17:49:28.739    1549-1627/? I/BusinessExecutor﹕ Task#671 : #run() called
09-14 17:49:28.739    1549-1627/? I/BusinessExecutor﹕ Task#671 : #run() ended
09-14 17:49:28.739    1134-1134/? D/STATUSBAR-IconMerger﹕ checkOverflow(360), More:false, Req:false Child:3
09-14 17:49:28.749  27808-27841/? D/HeadsetStateMachine﹕ Disconnected process message: 10
09-14 17:49:28.759    1134-1134/? D/STATUSBAR-PhoneStatusBar﹕ mBrightnessEnablebySettings = true mBrightnessEnablebyBattery = true mBrightnessEnablebyDisableFlag = true
09-14 17:49:28.839  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:28.849  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:29.019     689-1957/? D/SSRMv2:Monitor﹕ SIOP:: AP = 302 (read only)
09-14 17:49:29.949  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:29.959  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:31.059  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:31.059  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:31.169      300-300/? E/SMD﹕ DCD ON
09-14 17:49:31.559     689-1957/? D/SSRMv2:Monitor﹕ SIOP:: AP = 302 (read only)
09-14 17:49:32.039    1134-1134/? D/STATUSBAR-NetworkController﹕ refreshSignalCluster: data=-1 bt=false
09-14 17:49:32.079    1134-1134/? D/STATUSBAR-IconMerger﹕ checkOverflow(360), More:false, Req:false Child:3
09-14 17:49:32.169  27142-27142/appId D/BluetoothAdapter﹕ stopLeScan()
09-14 17:49:32.169  27142-27142/appId W/BeaconService﹕ Skipping scan because crash recovery is in progress.
09-14 17:49:32.259    1723-4404/? W/System.err﹕ org.json.JSONException: Value <HEAD><TITLE>Invalid of type java.lang.String cannot be converted to JSONObject
09-14 17:49:32.259    1723-4404/? W/System.err﹕ at org.json.JSON.typeMismatch(JSON.java:111)
09-14 17:49:32.259    1723-4404/? W/System.err﹕ at org.json.JSONObject.<init>(JSONObject.java:158)
09-14 17:49:32.259    1723-4404/? W/System.err﹕ at org.json.JSONObject.<init>(JSONObject.java:171)

davidgyoung, 2015-09-15:
@jessolo, since I cannot reproduce the issue, I have prepared an ad-hoc build for testing on your device.  This build will timeout on a bluetooth crash recovery process after 60 seconds if it does not complete successfully.

In order to test this build with your app, please download [this build](https://github.com/AltBeacon/android-beacon-library/releases/download/2.5.1-brttest/android-beacon-library-2.5.1-brttest.aar) and configure your project to use it instead of the main release.  In order to do so, you’ll need to:

Step 1. Make a app/libs directory inside your project, and put the android-beacon-library-2.5.1-brttest.aar file there.

Step 2. Make your app/build.grade file include a flatDir entry like this:

```
 repositories {
     mavenCentral()
     flatDir {
       dirs 'libs'
     }
  }
```

Step 3. Make your app/build.grade file use this specific version of the library:

```
    dependencies {
        //compile 'org.altbeacon:android-beacon-library:2+@aar'
        compile 'org.altbeacon:android-beacon-library:-2.5.1-brttest@aar'
    }
```

davidgyoung, 2015-10-13:
@jessolo, did you have a chance to test the adhoc build?

davidgyoung, 2016-01-07:
closing due to inactivity

closed 2016-01-07 by davidgyoung