AltBeacon/android-beacon-library #284: galaxy s6 and region scan

DarkAulius, 2015-09-17:
with samsung galaxy s6  after some time that the application is in the background no longer detects the enter region and exit region.
any solution?

thanks

davidgyoung, 2015-09-17:
Can you tell me what Library version you are using?    There was a fix in 2.3 for an issue where Samsung devices limited Alarms causing similar behavior.

https://github.com/AltBeacon/android-beacon-library/issues/184

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.3

DarkAulius, 2015-09-18:
2.5.1 i've read fix at 2.3 but does'n work

davidgyoung, 2015-09-18:
Can you please:

1. Provide the OS version of your Galaxy S6
2. Set `beaconManager.setDebug(true)`, then capture a LogCat excerpt from the time a beacon is last detected by your app to the time a detection is incorrectly missed.  This will be a big file, so it may be best to paste it into a Gist or upload it to a server elsewhere.

davidgyoung, 2015-10-13:
@DarkAulius, any luck capturing the log exceprt?

davidgyoung, 2016-01-07:
closing due to inactivity

closed 2016-01-07 by davidgyoung