AltBeacon/android-beacon-library #288: Necessity of using an unmodifiableList after a consumer has been bound

matthewmichihara, 2015-09-18:
I just hit a `ConcurrentModificationException` the other day, presumably because I tried to call `BeaconManager.getInstanceForApplication().getBeaconParsers().add(..)` after someone had binded to the beacon manager.

I noticed that at https://github.com/AltBeacon/android-beacon-library/blob/master/src/main/java/org/altbeacon/beacon/BeaconManager.java#L260-L264, `getBeaconParsers` returns an unmodifiable list if a consumer has already been bound. I understand there are threading concerns around iterating over beacon parsers if people can add / remove from that list at any time, but would it be possible to use something like a [CopyOnWriteArrayList](http://developer.android.com/reference/java/util/concurrent/CopyOnWriteArrayList.html) here instead to safely iterate while preventing `ConcurrentModificationException`s?

matthewmichihara, 2015-09-18:
I'm also not sure the `isAnyConsumerBound` guard there is useful, given that I could call `getBeaconParsers` ahead of time to get the raw list, bind, and then modify the beacon parser list with no consequences, sidestepping the `unmodifiableList` protection.

davidgyoung, 2015-09-18:
@matthewmichihara, these are good points.  Modifying this list at runtime has always been a sore point with this API.  I think the `CopyOnWriteArrayList` sounds like a better alternative to what we have now.  If you'd like to try out making this change and testing it, I'd be interested in seeing a pull request.

matthewmichihara, 2015-09-18:
:+1: will try to get something up soonish

commit 6d29771b (2015-09-23): Use CopyOnWriteArrayList for beacon parser list

BeaconManager used to either return a regular ArrayList or
unmodifiableList depending on whether any consumers were bound to the
BeaconManager. This unnecessarily caused
ConcurrentModificationExceptions if used incorrectly, and was easily
bypassed. Replace with the simpler CopyOnWriteArrayList.

Fixes #288
closed 2015-09-23 by davidgyoung