AltBeacon/android-beacon-library #289: Lollipop long period scan breaks phone's Bluetooth

jbarmanet, 2015-09-19:
In order to test if my beacons signal was stable enough, I let them on contact all night long with 2 phones, one on Android 4.4, the other on Android 5.1.1
Below is my scanning service code... 

In the morning the Android 4.4 was still scanning successfully the beacons.
But the Android 5.1.1 stopped after 8 consecutive hours, and I have this repeating in the logs : 

    09-19 06:42:11.914: E/CycledLeScannerForLollipop(11355): Scan Failed
    09-19 06:42:18.964: I/MonitorService(11355): didRangeBeaconsInRegion, number of beacons detected = 0
    09-19 06:42:18.970: E/CycledLeScannerForLollipop(11355): Scan Failed
    09-19 06:42:26.059: I/MonitorService(11355): didRangeBeaconsInRegion, number of beacons detected = 0

Worst than this : the Bluetooth of the Android 5.1.1 device is broken... I can no longer activate it / deactivate it, and neither of the connected devices (headset etc.) can connect to it. 
Even if I restart my App. 
Even if I uninstall my App (just to be sure...)... 

So at the end it looks like it broke the Bluetooth of this phone... and the Bluetooth came back only after rebooting the device. 

Is there any precaution to take when scanning continuously beacons ? (for example maybe interrupting sometimes the scan could make it more stable ?? I have no idea what is playing behind the scene here...) 

Thanks for your feedback

    public class MonitorService extends Service implements BeaconConsumer 
    {
	
		protected static final String TAG = "MonitorService";
	    private BeaconManager beaconManager;
		
		
		@Override
	    public void onCreate() 
	    {
		    super.onCreate();
		
			beaconManager = BeaconManager.getInstanceForApplication(this);
		    beaconManager.getBeaconParsers().add(new BeaconParser().
		            setBeaconLayout("m:2-3=0215,i:4-19,i:20-21,i:22-23,p:24-24"));
		
		    beaconManager.setForegroundScanPeriod(5000l);
		    beaconManager.setBackgroundScanPeriod(5000l);
		    beaconManager.setForegroundBetweenScanPeriod(1100l);
		    beaconManager.setBackgroundBetweenScanPeriod(1100l);
		
			setupBeaconManager();
	    }
	
	
	    private void setupBeaconManager()
	    {
		    if (!beaconManager.isBound(this))
		        beaconManager.bind(this);
	    }
	
	    
	    private void unsetBeaconManager()
	    {
		    if (beaconManager.isBound(this))
		    {
		        beaconManager.unbind(this);
				try 
		        {
		            beaconManager.stopRangingBeaconsInRegion(new Region("apr", null, null, null));
		        } 
		        catch (RemoteException e) 
		        {  
		            Log.i(TAG, "RemoteException = "+e.toString());
		        }
		    }
	    }
	
	
	    @Override
	    public void onBeaconServiceConnect() 
	    {
		    beaconManager.setRangeNotifier(new RangeNotifier() {
		        @Override 
		        public void didRangeBeaconsInRegion(Collection<Beacon> beacons, Region region) 
		        {
		            Log.i(TAG,"didRangeBeaconsInRegion, number of beacons detected = "+beacons.size());
		        }
		    });
		
		    try 
		    {
		        beaconManager.startRangingBeaconsInRegion(new Region("apr", null, null, null));
		    } 
		    catch (RemoteException e) 
		    {  
		        Log.i(TAG, "RemoteException = "+e.toString());
		    }
	    }
	
	    @Override
	    public void onDestroy() 
	    {
		    unsetBeaconManager();
			super.onDestroy();
	    }
    }

davidgyoung, 2015-09-19:
This does sound like a problem with the device you mention on 5.1.1.  Can you please tell us the make, model and build version?

The library already pauses between scans -- for 1100ms after every 5000ms spent scanning using your settings. So I doubt a longer pause would help.  This problem may be caused by a bug in the Bluetooth firmware on that device.

You could try disabling the 5.x BLE APIs on the problematic  device to see if that helps.

jbarmanet, 2015-09-19:
The buggy device is a Nexus 6, Android 5.1.1, build LMY48M 
As suggested I'll make the same test next night with : 

    beaconManager.setAndroidLScanningDisabled(true);

And will let you know...

jbarmanet, 2015-09-19:
Is there any smart way to reset completely the AltBeacon's BeaconManager attached to the Service ? 
My thought is that if it is regularly re-initialized, this could reduce quite much the probability of a BTLE crash on the device (?)

davidgyoung, 2015-09-19:
There is no way to reset the BeaconManager as it is a singleton and will not be destroyed without stopping the app.  You can restart the BeaconService.  If you unbind all BeaconConsumer instances, it will be destroyed.  Then you can rebind.  You might see if this helps.

Another thing you could try is programmatically turning Bluetooth off and back on after unbinding as described above.  Android allows this with the same Bluetooth Admin privilege the library needs to scan.

davidgyoung, 2015-10-13:
@jbarmanet, did you try setting  `beaconManager.setAndroidLScanningDisabled(true);`?  What were the results?

jbarmanet, 2015-10-14:
Yes David, I just retried this night with your latest lib version 2.6.1. And I'm still on Android 5.1.1.
The results are much better with setAndroidLScanningDisabled(true) indeed. 
It loses the beacons in less than an hour with L enabled, but it could last a bit more than 8 hours with Android L disabled... But even once lost it restarted gently... 

I should make some more tests to let you know if what I'm saying here is reliable, because I added some tricks in the app to avoid "fake losses" as much as possible... so I need to alter the code in order to get more "raw" info about this... I will let you know asap

jbarmanet, 2015-10-15:
One more feedback,

I have a strange behavior, first time I have this problem (is that to say it is related to the deactivation of AndroidLScanning ?).
It looks like we came back to the first issue, but with new errors in the LogCat, see, I have continuous : 

    10-15 07:55:14.055: D/BluetoothAdapter(4913): stopLeScan()
    10-15 07:55:14.055: D/BluetoothAdapter(4913): startLeScan(): null 

And my App cannot anymore interact with Bluetooth, it looks like it has been broken.
I can manually deactivate the BT in the phone's params, but I can no longer reactivate it, even manually... I thought (while writing those lines) that I had to restart the phone, but the Bluetooth just came back suddenly, reactivating your AltBeacon classes !

So, something my break the bluetooth at a system level... but once it comes back it restarts... 
Sorry if it looks messy, I just wrote it as it came... it is indeed a bit confusing...

jayshildave, 2015-12-01:
Even on Nexus 6 and Nexus 6P running MarshMallow we observed similar issue. The worst part was even restarting did not seem to solve the problem we have to manually do a factory reset. Do i update the logs they are same as jbarmanet.

davidgyoung, 2015-12-01:
@jayshildave,  It may be useful to capture a Logcat excerpt that is not filtering for your application, but shows everything on the phone, then try to do a bluetooth scan from Android Settings and see what shows up in the logs.

jayshildave, 2015-12-02:
@davidgyoung , thanks for the response and sorry for replying late. We had reset the device and are now eagerly waiting for the crash. As soon as i have the crash will post the logs of device. 
Couple of things we do which may have a impact.
1. We are running the scan period for foreground as well as background somewhere between 15 to 30 sec. Which i understand is way to fast but its because of business use case we have where its required to scan at that rate both in foreground and background.
2. We also applied the  setAndroidLScanningDisabled(true) in the code while putting it this time. 

As soon as i have more logs would paste here.

jayshildave, 2015-12-09:
@davidgyoung Finally after 7 days of usage the device crashed again and we found it was similar to issue report on google at https://code.google.com/p/android/issues/detail?id=191831 The issue doesn't look to be AltBeacon issue instead is the google firmware which on continuous BLE scans tends to break the Bluetooth of the device and similar to issue #26 Hopefully google will fix it. Will update this issue in either case so people are aware.

paulpv, 2016-01-07:
@davidgyoung I've been watching that code.google.com 191831 issue for months.
Can you think of any way to get traction on this issue so that someone in side Android/AOSP/Bluedroid can fix this?
This is going to be a major problem, given the supposed tsunami of IoT/Beacons.

davidgyoung, 2016-01-07:
There were also lots of reports of this in #308

davidgyoung, 2016-01-07:
Looks like there is a fix, but it is not released yet:

> Project Member #26 dnaga...@google.com
The development team has fixed the issue that you have reported and it will be available in a future build.

https://code.google.com/p/android/issues/detail?id=191831

paulpv, 2016-01-07:
Yup! That was great news I saw earlier today!

I'll try to post a message here if I see the fix in the wild.
Anyone else that sees the fix in the wild, please let me know at pv@swooby.com.
Thanks!

SilkSama, 2016-03-17:
it seems to be fixed in the March update
https://code.google.com/p/android/issues/detail?id=191831#c64

SamuelY, 2016-05-02:
The issue seems not found in Lollipop only.
My samsung S4 & motorola phones also can duplicate this issue
After samsung S4 reboot, the bluetooth and wifi still crash....

sangamangreinin, 2016-05-03:
I can reproduce this issue in motorola.

Berger89, 2016-05-04:
Same issue on the oneplus one and android 6.0.1

davidgyoung, 2016-05-04:
Quick recap here:  Google reports this issue is reported fixed for 6.x devices as of 6.0.2.:  https://code.google.com/p/android/issues/detail?id=191831

You'll have to wait for that update to get it.  Unfortunately, those of you with  5.x are probably out of luck.

Berger89, 2016-05-05:
Solution for probleme "BluetoothLeScanner: could not find callback wrapper" with Android 6:

http://developer.radiusnetworks.com/2015/09/29/is-your-beacon-app-ready-for-android-6.html

aderbas, 2016-11-16:
https://developer.android.com/training/permissions/requesting.html

zosocanuck, 2016-11-27:
I an reproduce the error: "BluetoothLeScanner: could not find callback wrapper" on a Nexus 5 build number M4B30X.

A factory reset does not fix the issue.

davidgyoung, 2016-11-27:
@zosocanuck, I don't think that seeing "could not find callback wrapper" in the logs is necessarily a problem.  This issue is about something different, which is indicated by seeing "Scan Failed" in the logs.

In my experience, seeing "could not find callback wrapper" in the logs often does not indicate a problem at all.   If you are not detecting beacons there are many, many possible causes.  @Kokscola mentions one possible cause above, but again there are many others.

I am closing this issue as the root cause was fixed by Google in 6.0.2.

closed 2016-11-27 by davidgyoung

am2222, 2017-05-06:
Does any one has this problem with android 7?
I am on sumsung s7 with android 7 and have faced this error
`05-06 14:55:21.194 14740-14740/org.altbeacon.beaconreference E/CycledLeScannerForLollipop: Scan Failed`

rickul, 2018-09-05:
I'm having the same issue:
`E/CycledLeScannerForLollipop: Scan failed: app cannot be registered`
on HTC U11 life with Android 8.0.0. But I realized this issue on several phones of different vendors and with different Android versions.

Its happening when scanning a long time in the background. After some research, this issue seems to be present for a long time. Isn't there any workaround for it?

davidgyoung, 2018-09-05:
I am locking this issue as it has become a magnet for similar but distinct issues.  The specific issue that opened this was resolved in an Android bug fix.  For similar cases, please open a new issue.