AltBeacon/android-beacon-library #291: Cannot use in Foreground Service

StephanGs, 2015-09-24:
Is there any way to use the AltBeacon library in a foreground service?

We are trying to use the library with the ranging feature in a foreground service. This service should keep running after the app is killed (e.g., when swiping the app out of the task switcher). However, we get the following exception and the service terminates.

```
09-23 18:45:04.480     535-1212/? W/ActivityManager﹕ Exception when destroying service com.my.application /org.altbeacon.beacon.BeaconIntentProcessor
    android.os.DeadObjectException
            at android.os.BinderProxy.transactNative(Native Method)
            at android.os.BinderProxy.transact(Binder.java:496)
            at android.app.ApplicationThreadProxy.scheduleStopService(ApplicationThreadNative.java:950)
            at com.android.server.am.ActiveServices.bringDownServiceLocked(ActiveServices.java:1698)
            at com.android.server.am.ActiveServices.bringDownServiceIfNeededLocked(ActiveServices.java:1608)
            at com.android.server.am.ActiveServices.stopServiceTokenLocked(ActiveServices.java:568)
            at com.android.server.am.ActivityManagerService.stopServiceToken(ActivityManagerService.java:15160)
            at android.app.ActivityManagerNative.onTransact(ActivityManagerNative.java:925)
            at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:2208)
            at android.os.Binder.execTransact(Binder.java:446)
```

Any ideas are welcome.
Is this a bug or is this by design?

davidgyoung, 2015-09-24:
When you kill an app with the task switcher, all services stop along with the Application.  This is expected behavior.

I think this exception is a red herring.  As mentioned above, It is normal for the app to terminate when you kill it from the task switcher.  In this case, the exception is just thrown when the app terminates for some reason of the state of the app at the time, but it doesn't change the fact that the app terminates as expected.

The way the Android Beacon Library continues scanning after an app terminates is through Alarms.  A five minute alarm is continually kept active during background scanning.  If the app terminates unexpectedly, the Alarm goes off, calling the library's broadcast receiver and restarting it.  For this to work, the Application onCreate method must restart beacon scanning by using the RegionBootstrap class.  This should work for your use case as well -- although you may want to start up your own scanning service in the Application onCreate method.  You can read more about how to set this up here:  http://altbeacon.github.io/android-beacon-library/background_launching.html

StephanGs, 2015-09-24:
Hello David,

Thanks for the speedy reply, but I believe there is a misunderstanding.  

I am sorry for using the term "killing". Swiping out an app from the recent app list does not kill the app (explaining the exact behavior is beyond the scope of this post and my knowledge – for details you may check out http://android.stackexchange.com/questions/19987/what-actually-happens-when-you-swipe-an-app-out-of-the-recent-apps-list  for details).

Android supports so-called foreground services. Such a foreground service continues to run in the background – even if the app itself was swiped out of the recent task list. Good examples of such foreground services are the soundcloud music player (and a lot of other players) or the facebook app. All such apps continue to play music (or provide notifications). 

We are trying to implement exactly such a service with the AltBeacon library. If the library IS NOT embedded in our project everything works fine (end the service keeps running). If the AltBeacon library is embedded the service stops with the mentioned error. 

Thanks again.

davidgyoung, 2015-09-24:
When you say Foreground service, am I correct in assuming you are referring to this:  http://developer.android.com/reference/android/app/Service.html#startForeground%28int,%20android.app.Notification%29?

If so, my guess is that the problem is that the Android Beacon Library starts a non-foreground service  to scan for beacons and there may be an issue with using foreground services and background services simultaneously in the same app.  

If this is the issue, you're breaking new ground here -- I'm not aware of anybody who has tried using the library this way.  You might try manually starting the library's `BeaconService` as a foreground service before the `BeaconManager` has a chance to do so, and see if this works.  If you can figure out a way to get this working, I'd be glad to see a pull request that sets an optional way to run the `BeaconService` as a foreground service.

StephanGs, 2015-09-24:
Thanks,

yes – that is exactly what we are interested in.

Starting the BeaconService as an additional foreground service is not an option because the number of foreground services is limited by Android (seems the maximum is 3).

We have also tried another approach to keep the service running by starting our service in a separate process. But this approach was not successful in our first try. We experienced a RuntimeException from the binder stub implementation (`<Class not found when unmarshalling: org.altbeacon.beacon.service.StartRMData>`), which suggests that the AltBeacon service is not started in the same process as our service. However, we need to verify this. Maybe there is a way to force the AltBeacon service to run on the same process as our service?

A solution would be great, because currently it seems that the AltBeacon library can’t be used in a way which avoids service restarts.

davidgyoung, 2015-09-24:
To clarify, it is possible to use the library in a way that doesn't automatically restart the service -- you just have to remove the StartupBroadcastReceiver manifest entry (easier than it sounds.)  While this is a fine solution for foreground-only apps, it won't help your case, because what you want is for the beacon service not to be stopped when the user closes the app from the task switcher.  

I agree that running the BeaconService in a separate process will probably not work.  This was its initial design, but we moved away from that because a shared service between multiple apps was not tenable for security and versioning reasons.  Since we have shifted from this design, there are a number of configuration options that have been created that are based on singleton instances that require that the client's BeaconManager and the BeaconService be in the same process.

It's too bad you have already hit the limit on foreground services -- that would be the easy option.  Right off I can't think of any alternatives that don't require significant redesign of the BeaconService.

aldoborrero, 2015-11-03:
@davidgyoung any concrete clues on how to accomplish this? Well first of all we should remove all references to the singleton, right?

davidgyoung, 2015-11-04:
Yes, you would need to remove all static configuration options in the BeaconManager and a few other classes and find a different way to share them between multiple processes.

lionscribe, 2016-09-20:
I have posted a solution on StackOverflow on how to run your beacon service in a separate process. See http://stackoverflow.com/a/39583516/2661303 for details.

ClaudioINc, 2016-10-11:
@lionscribe could you share what the service class would look like, i can't comment on stackoverflow.

hirenpatelazilen, 2017-03-25:
@davidgyoung  and @StephanGs , Alt beacon library is working fine in foreground and background mode. My application can not scan the beacons in kill state but I want to detect the beacons if application is in kill state. Can you please let me know any work around for the kill state?

Thank you,

davidgyoung, 2017-03-26:
@hirenpatelazilen,  There is an experimental build with support for running the service in its own process which does support the kill state here:  https://github.com/AltBeacon/android-beacon-library/pull/479

Please give it a try and let me know if it solves your issue.

davidgyoung, 2020-04-09:
An included foreground service has been added to the library.  See [here](https://altbeacon.github.io/android-beacon-library/foreground-service.html).

closed 2020-04-09 by davidgyoung