AltBeacon/android-beacon-library #294: didRangeBeaconsInRegion() returns 0 beacons while trying to detect Eddystone URLs

EricYuen, 2015-09-30:
My app that use to successfully detect Eddystone URLs no longer works. I've reviewed issues #84, #163 and #242 but they have failed to give me a solution.

Device: Nexus 9
I've tried using AltBeacon library version v2.5 and v2.6 both yield same results
I'm using Radius Beacon USB and it's advertising Eddystone URL (and AltBeacon, but I've tried turning that off).
I've tried the Locate App and verified that the Eddystone URL is being detected:
TxPower :-61
RSSI: -48

On launch of the app it displays which layouts it's trying to parse:
D/BeaconParser﹕ Parsing beacon layout: s:0-1=feaa,m:2-2=10,p:3-3:-41,i:4-20v
D/BeaconParser﹕ Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
I'm using BeaconParser.EDDYSTONE_URL_LAYOUT

When I receive the didRangeBeaconsInRegion() callback, there are no Eddystone URL beacons in the collection. It does however get called a second time detecting the AltBeacon. If I turn off advertising of AltBeacon or clear the BeaconParser before adding the Eddystone URL layout, it stops detecting the AltBeacon as expected, but it still doesn't detect the Eddystone URL.

Any ideas or leads would be greatly appreciated, thanks.

davidgyoung, 2015-09-30:
Thanks, for this report, @EricYuen.  I have indeed found a problem, but it was introduced only in version 2.6.  Are you certain this didn't work with 2.5?

I have reproduced the problem on 2.6.  Basically a region with three null identifiers won't match an Eddystone-UID or Eddystone-URL beacon, but a Region with fewer identifiers will.  If this is indeed the problem, then you should be able to fix it with the 2.6 version of the library by changing your Region definition.

Please change:

        Region region = new Region("myRegion",
                null, null, null);

To:
        
        ArrayList<Identifier> identifiers = new ArrayList<Identifier>();
        identifiers.add(null);
        Region region = new Region("myRegion", identifiers);

Note that this new region definition will *ONLY* match beacons with a single identifier (e.g. Eddystone URL)

Please let me know if this works.  If it does, this is a good indication that the fix I have put in #296 will resolve your issue, and I can roll a new release.

EricYuen, 2015-09-30:
Thanks David! That solved the problem. Fortunately I only need to detect Eddystone URLs.
Secondly, I was mistaken in saying that the problem occurs on 2.5 too. I just made a mistake in gradle settings, I retested it and it works on 2.5.

closed 2015-09-30 by EricYuen