AltBeacon/android-beacon-library #301: didRangeBeaconsInRegion and empty list beacons

evansgelist, 2015-10-13:
nexus 5
android 6.0

I get an empty list beacons when call didRangeBeaconsInRegion. 
This happened after the upgrade to Android 6.0. Before that, everything worked FINE.
Add a line 
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
in the manifest, but not helped

davidgyoung, 2015-10-13:
@evansgelist, this is the first report of this problem I have heard of on the Nexus 5 with Android 6.0.  Can you please try three things:

1. Go to Settings -> Apps -> AltBeacon Loc, then check to see what shows up under Permissions.    A screenshot would be helpful.

2. Go to Settings -> Bluetooth, and look for available devices.  Do you see any?

3. Change your app so it sets `beaconManager.setDebug(true);` then capture a logcat excerpt when it is supposed to be detecting beacons (but isn't.)  The capture will be long, so you probably need to post it to a server (a Gist will work well) and link to it from here.

evansgelist, 2015-10-14:
Ok
Please check
1. https://dl.dropboxusercontent.com/u/20178650/Screenshot_20151014-100329.png
2. https://dl.dropboxusercontent.com/u/20178650/Screenshot_20151014-100624.png
3. Log  - https://gist.github.com/evansgelist/a1927973d746b2cb1a91

davidgyoung, 2015-10-14:
Wow.  That log file indicates that bluetooth LE scanning starts successfully, but just does not receive any callbacks.  The Android bluetooth stack simply thinks it doesn't see any devices on the Nexus 5 on Andorid 6.0.  This is very concerning -- I worry that this OS build version has broken Bluetooth LE scanning.

One other thing to try:  In your own app, can you please try adding the following line to see if this enables detections again:

        beaconManager.setAndroidLScanningDisabled(true);

This will force the library to use the old 4.x Bluetooth scanning APIs instead of the new 5+ scanning APIs.  Perhaps only the 5+ scanning APIs are broken.

evansgelist, 2015-10-14:
Nothing changed. I made a new log. Please see
https://gist.github.com/evansgelist/a195afb8ac415f2f64a3

davidgyoung, 2015-10-14:
This is really bad news, guys.  It suggests that BLE scanning is completely broken on at least one Nexus 5 Android 6 build.  Below is the table of know build numbers for this device and OS version.  Can you please comment on which build number you have so I can update the table below.

| Model | Build number | OS Version | BLE Scanning works? |
|-------------------|-----------|---------|---------|
| LGE Nexus 5 | MPA44G | 6.0 |
| LGE Nexus 5 | MPA44I | 6.0 |
| LGE Nexus 5 | MPZ44Q | M |
| LGE Nexus 5 | MRA58K | 6.0 |
| LGE Nexus 5	| MRA58L	 | 6.0 |

ankitgupta100, 2015-10-14:
Broken on MRA58K.

evansgelist, 2015-10-14:
I have the same build number
https://dl.dropboxusercontent.com/u/20178650/Screenshot_20151014-233126.png

ankitgupta100, 2015-10-15:
David, is there any temporary/permanent solutiion to this problem, as I can't proceed with testing anymore on my device. Will posting on android forums help?

if12b062, 2015-10-15:
I have the same build number (MRA58K) and it seems that not only does the BluetoothLeScanner#startScan() method need the Location runtime permission, but also that the Location service is enabled.
There seems to be the same issue (unnecessary permission) when trying to scan for WIFIs which was reported [here](https://code.google.com/p/android-developer-preview/issues/detail?id=2964).
To test this behaviour you could download a BluetoothLeScanner application (i.e. [this](https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan&hl=en)) and try scanning with location disabled and enabled.

davidgyoung, 2015-10-15:
@if12b062, can you clarify what you mean by "the Location service is enabled"?

Great news that we have one report of scanning working on MRA58K.  

In case you mean Settings -> Location, I have it turned OFF on my Nexus 9 with 6.0 build MPA44I, and it detects beacons just fine.

ankitgupta100, 2015-10-15:
@davidgyoung I guess @if12b062 is talking about a possible solution only for build MRA58K on Nexus 5. I tried now with Locate app by turning location service on and the scanning works fine.

davidgyoung, 2015-10-15:
@ankitgupta100, this is good news.  Can you please clarify precisely what you did to turn location service on?

ankitgupta100, 2015-10-15:
Settings -> Location -> On (manually, not via any app). Seems like a temp solution, this will drain the battery and doesn't make sense why it should be ON for BLE scanning.

davidgyoung, 2015-10-15:
Interesting.  If you go back and turn it off, does it immediately stop beacons from being detected on the Nexus 5 MRA58K?  (For the record, I have tried the same on a Nexus 5 with Android 5.0 and a Nexus 9 with Android 6.0 and both keep detecting beacons with the location service disabled.)

ankitgupta100, 2015-10-15:
Yes, the beacons are immediately stopped detecting on turning it off.

davidgyoung, 2015-10-15:
@evansgelist, does turning on Location Services fix it for you?

if12b062, 2015-10-15:
Yes, turning on means in the settings (as @ankitgupta100 did it) or via the quicksettings.

evansgelist, 2015-10-15:
>>>@evansgelist, does turning on Location Services fix it for you?
Yes, now it works. But it's strange.

davidgyoung, 2015-10-20:
I am closing this issue since we have a resolution.  But this behavior does not seem to be consistent across Android devices, and the requirement on the Nexus 5 with Android 6 to have location services turned on to do bluetooth scans is not a documented Android requirement.  It is unclear whether it is a bug or if it is an intended security restriction.

closed 2015-10-20 by davidgyoung

axierjhtjz, 2015-11-09:
Just for the sake of the documentation, it only shows beacons nearby on Nexus 5 with Android 6.0 on MRA58N build when location services are enabled too.

ankitgupta100, 2015-11-10:
Even with the recent update, this issue is not fixed by Android team.

stanleychg, 2016-01-04:
Issue still is not fixed. Requires the location services to be turned on. It also seems that the onBatchScanResults callback is never called.

Device: Nexus 5
Android V: 6.0.1
Build Number: MMB29K

ankitgupta100, 2016-01-04:
@stanleychg The issue is fixed from 6.0.1 update and it does not require location service to be turned on now. I have the same build as yours.

stanleychg, 2016-01-04:
Ah, my mistake. Thank you for the quick reply.

heypiotr, 2016-01-05:
Just tested on my Nexus 5, Android 6.0.1 (build MMB29K), and I still need Location enabled to detect beacons.

davidgyoung, 2016-01-05:
Yes, the Locate app in the play store still targets pre-SDK 23,  which explains why it does not need Location services on.

acaliaro, 2016-01-23:
@davidgyoung sorry if I disturb you. I would like to know if your library monitoring region also if i press "power off" on my device. I don't "shut down" the device, it's only in "sleep mode" (I don't know if on Android it's the right name)

Thanks

davidgyoung, 2016-01-24:
@acaliaro, this question is a bit off topic for this thread.  Could you please open a new question on StackOverflow.com?  That is the preferred forum for library support questions.

4ntoine, 2016-02-08:
i'm having similar issue (http://stackoverflow.com/questions/35266845/how-to-fix-android-ble-scan-failed-feature-unsupported), any idea on how to fix it?

AlexLardschneider, 2016-02-21:
The same issue happens on a S5 (G900F), beacon scanning only works with location permission and location services enabled.

maorhadad, 2016-04-03:
LGE nexus 5 MMB29V must GPS enabled to find ble.

davidgyoung, 2016-04-03:
To clarify: the location services switch in Android settings is not just for enabling GPS.  It also enables getting location data from cell tower and WiFi.  And as this thread suggests, some Android builds also require it to be turned on to detect BLE advertisements (including beacons).  This makes some sense as location information can be inferred from beacons.  But it would be nice if the behavior were consistent across Android builds and documented by Google.

andreaciri, 2016-05-02:
"Broken" on Nexus 5 MMB29S (6.0.1).

niteshthakrar, 2016-05-02:
FYIO: same problem as above on Moto G ("rd Generation) with Android v6.0    works if location turned on!

davidgyoung, 2016-05-02:
To clarify, Android has changed as of 6.0 to require that Location be turned on in settings in order for an app to do BLE scanning or search for BLE beacons.  The fact that this has been inconsistently enforced on different builds does not change the fact that this is the new requirement.  

Please consider it a requirement for all apps targeting SDK 23 on Android 6+ devices to have location turned on in order to detect beacons.  The "bug", if there is one, is that some Android 6 builds do not require this.

rameen, 2016-07-05:
@davidgyoung .Is google fixing or it will remain as enforced.Is the problem with Nexus 5 devices only with Android 6.
It is replicated in my Nexus 5 with build number M8974A with 6.0.1

davidgyoung, 2016-07-05:
This is intended behavior on Android 6.  Many device models enforce that
location be enabled to do BLE scanning, and are supposed to do so.  Those
devices that do not require this are the outliers and are the source of
confusion.