AltBeacon/android-beacon-library #303: Library unable to detect beacons of custom parsers after Marshmallow update

ankitgupta100, 2015-10-14:
My app uses altbeacon lib and is unable to detect custom parsers since I updated the OS to Marshmallow. There is no error in logcat too. I tried detecting them from Locate app too from a couple of Nexus 5 devices, but still not detecting. Are there any changes required ?

evansgelist, 2015-10-14:
Perhaps the same problem
https://github.com/AltBeacon/android-beacon-library/issues/301

ankitgupta100, 2015-10-14:
Looks so. Have tried out the same methods mentioned in that post. Same observations here too.

ankitgupta100, 2015-10-14:
Devices are shown in Settings -> Bluetooth, but not detected in my app and Locate app. My app has FINE_LOCATION PERMISSION and all required Bluetooth permissions.

davidgyoung, 2015-10-14:
@ankitgupta100,  I wonder if this problem is caused by an upgrade to 6.0 with apps that were already  installed for Android 5.0.  Can you try uninstalling the Locate app and re-installing it, to see if it starts detecting?

davidgyoung, 2015-10-14:
This is a duplicate issue to #301.  Please add subsequent comments there.

closed 2015-10-14 by davidgyoung