AltBeacon/android-beacon-library #304: Using a different process for beacon scanning

yodutouf, 2015-10-14:
For me, it's an important limitation today of the library :
If I kill the application, the beacon scanning process is stopped as well (ok the scanning restart when I plug my phone)
It's a critical point for launching a notification from background when a user enters into a beacon region.

I know, It's major breaking change, that oblige to re-think the communication between the beacon service and the beaconManager and many other architecture points but the benefit for the user experience is great.

I read some comments into the issues about. 
I see that an argument is the beaconService versionning. 

I see two solutions that Android permits 

## A- Using a different process does not oblige to use the same service version

Imagine you get : 
### 1- An application A, with the following service manifest configuration 
    
    <service android:enabled="true"
            android:exported="false"
            android:process="com.altbeacon.service"
            android:label="beacon"
            android:name=".service.BeaconService"
            />
###  2- An application B, with the following service manifest configuration 
    
    <service android:enabled="true"
            android:exported="false"
            android:process="com.altbeacon.service"
            android:label="beacon"
            android:name=".service.BeaconService"
            />
            
### If the service is started as following by the library :
    
    Intent intent = new Intent(context, BeaconService.class);
    context.startService(intent);

This will conduct to get two BeaconService instances running into the same process "com.altbeacon.service"
So, even if application A and application B are not using the same version of the beaconService it won't be a problem because each application is creating its own BeaconService instance (into the "com.altbeacon.service" process)
This implies as well that each application could continue to personnalize the scanning frequency for exemple.

The main limitation I see is : if the user kills the com.altbeacon.service process, all the applications using the library will stop receiving beacon scanning informations.

## B- Using a different process and only one BeaconService instance

I think it could help about the scanning process optimization even if it has many consequences like less configuration possible (scanning frequency for exemple),  main architecture rewrite because the Regions could come from various applications.

### 1- Service configuration to mutualize BeaconService

If the application A and application B get the following serice manifest configuration :

     <service android:enabled="true"
            android:exported="true"
            android:process="com.altbeacon.service"
            android:label="beacon"
            android:name=".service.BeaconService"
    >
        <intent-filter>
            <action android:name="com.altbeacon.service.beaconService" />
        </intent-filter>
    </service>
    
And if the library is starting the BeaconService as following : 
    
    context.startService(new Intent("com.altbeacon.service.beaconService"));
    
The beaconService will run into the "com.altbeacon.service" process and this instance will be share by all the applications calling context.startService(new Intent("com.altbeacon.service.beaconService"));

But what if the application A and B is using a different version of the library

### 2- Service versionning 

So because the BeaconService is lauched using an intentFilters we could imagine to version the intent-filter :
    
     <intent-filter>
            <action android:name="com.altbeacon.service.beaconService_v12" />
    </intent-filter>

And so, into application A using the version 12 of altBeacon, the altBeacon BeaconManager will call **context.startService(new Intent("com.altbeacon.service.beaconService_v12"))** and into the application B using the version 14 of altBeacon, the altBeacon BeaconManager will call **context.startService(new Intent("com.altbeacon.service.beaconService_v14"))**

What you think about these solutions ?
What you think about implementing it starting from the solution A (that breaks only the way to communicate between beaconService and beaconManager) ?

davidgyoung, 2015-10-14:
How would you propose the independent beacon scanning process be installed and started?  It could be installed as a separate app, but I don't think end users would do that.

yodutouf, 2015-10-14:
No, what I propose works exactly like the current library. 
The beaconService stays included with the applications that integrate the library.

It's just the manifest configuration of the BeaconService that changes to launch it in a different process (android:process option) and the way the library starts the service **startService(new Intent(context, BeaconService.class)) or startService(new Intent("com.altbeacon.service.beaconService"))** if you want to use the same Service instance over several applications.

davidgyoung, 2015-10-14:
Sorry, I still don't understand.  If the service is bundled with the application, what would be the practical benefit of being able to access a service instance bundled with a different application?

yodutouf, 2015-10-14:
Perhaps I miss a point, and I see an optimization where there is no.
I will try again what is in my mind.

My first point : running the beaconService in a different process to avoid loosing scanner detection when application is killed. 

Second point :  With the current version of the library if the application A and application B includes the altBeacon library, each one will start a beaconService. 
If application A is on foreground, application B is on background, the application B won't take benefit of the fact that application A is scanning for BLE at high rate, does it ? (perhaps I miss the point here)
More globally, if application A and B share the same beaconService, several resources could be shared (for exemple some beacon analysing process) which could save some battery and memory on the phone user.

If I summarize less there is beaconService running on the User Phone, better it is for the user Phone performance, isn't it ?

davidgyoung, 2015-10-14:
On point 1, I don't know for sure that running in a new process will affect what happens when you kill the task with the task switcher.  Have you tried this to see?

On point 2, The problem with allowing different applications to use the same service instance (even if suffixed by version) is security.  How can you trust that the service bundled with another app really is doing what you expect it to do?  I could make an app and publish it to Google Play that exposes something called "com.altbeacon.service.beaconService_v12" that refuses to detect beacons with identifiers known to be used by a certain app and thereby disable its functionality.

When originally designing the scanning service, I had thought of making it a shared service.  But I could never come up with a solution to the problem discussed in point 2.  This was also discussed in this issue:  https://github.com/RadiusNetworks/android-ibeacon-service/issues/22

yodutouf, 2015-10-15:
Thanks for your answer,

Point 1: I was wrong, when the users kill the application, the beaconService was killed even if running into a separate process.
Running service in a separate process help to run it longer because the service process while have a less memory print than the full application and so when android need memory, the service process has less probability to be killed.

However, the BeaconService could implement onStartCommand() and return START_STICKY
    
    public int onStartCommand(Intent intent, int flags, int startId) {
       
        return START_STICKY;
    }

In that case, the service is restarted after 5 seconds after the application is killed. (This time I fully check)

Point 2 : I see the security problem, I was focus on the version problem. I don't see efficient solutions for now but I keep it mind

davidgyoung, 2016-01-07:
Closing due to inactivity.  I am still open to the ideas presented in this discussion, but I don't see the obvious benefits versus what is implemented now.

closed 2016-01-07 by davidgyoung

aldoborrero, 2016-04-20:
Hi @yodutouf and @davidgyoung I was thinking the same long time ago.

I see a real benefit of using the process tag inside the library. But first citing what android says with process tag:

> The name of the process where the service is to run. Normally, all components of an application run in the default process created for the application. It has the same name as the application package. The <application> element's process attribute can set a different default for all components. But component can override the default with its own process attribute, allowing you to spread your application across multiple processes.
If the name assigned to this attribute begins with a colon (':'), a new process, private to the application, is created when it's needed and the service runs in that process. If the process name begins with a lowercase character, the service will run in a global process of that name, provided that it has permission to do so. This allows components in different applications to share a process, reducing resource usage.

This in the end will be translated to what @yodutouf has said before: "less memory consumption" as the main process for the application will start to grow memory as long the user keeps using the app (think about the allocation of images, or whatever). The more memory the process consume, the high probability android will kill the app when is not in foreground.

In the memory section (http://developer.android.com/training/articles/memory.html) we can found the next:

> An example of when multiple processes may be appropriate is when building a music player that plays music from a service for long period of time. If the entire app runs in one process, then many of the allocations performed for its activity UI must be kept around as long as it is playing music, even if the user is currently in another app and the service is controlling the playback. An app like this may be split into two process: one for its UI, and the other for the work that continues running in the background service.
> You can specify a separate process for each app component by declaring the android:process attribute for each component in the manifest file. For example, you can specify that your service should run in a process separate from your app's main process by declaring a new process named "background" (but you can name the process anything you like)
>Before you decide to create a new process, you need to understand the memory implications. To illustrate the consequences of each process, consider that an empty process doing basically nothing has an extra memory footprint of about 1.4MB, as shown by the memory information dump below.
> This memory footprint for an empty process is fairly significant and it can quickly grow as you start doing work in that process. For example, here is the memory use of a process that is created only to show an activity with some text in it.
> Additionally, when running more than one process, it's more important than ever that you keep your code as lean as possible.

So I could see a benefit of using it.

Also if you implement the BeaconManager inside another service and you use the startForeground method, Android will raise priority of the service and will try to honor to not to kill the service (as long as the user don't swipe the app from the recents, in that case it will be restarted if your service has the sticky flag).

I would gladly want to discuss this more deeply @davidgyoung and to decide if is a good approach to move :)