AltBeacon/android-beacon-library #305: Bug in Region ID matching

philipp-fischer, 2015-10-16:
Hi,
I tried to match a single-id beacon and didn't succeed with a region of three null-ids. It only worked when I created a region with just one Identifier.

In short, this didn't match:

    Region region = new Region("com.example.myapp.boostrapRegion", null, null, null);

while this does:

        ArrayList<Identifier> idlist = new ArrayList<Identifier>(1);
        idlist.add(null);
        Region region = new Region("com.example.myapp.boostrapRegion", idlist, null);

Minimal example for non-working matching although it should (all region IDs are null):

    Region region = new Region("com.example.myapp.boostrapRegion", null, null, null);

    Beacon beacon = new Beacon.Builder()
            .setId1("533").setId2("533").setId3("533")
            .setManufacturer(0x4c)
            .setTxPower(-59)
            .build();

    Log.d(TAG, "Match region " + region + " with beacon " + beacon + " = " + region.matchesBeacon(beacon));

davidgyoung, 2015-10-19:
What library version are you using?  I can't reproduce the issue you are seeing with library version 2.6.  I created a unit test with the second example (shown below) and it passes. 

        Region region = new Region("com.example.myapp.boostrapRegion", null, null, null);

        Beacon beacon = new Beacon.Builder()
                .setId1("533").setId2("533").setId3("533")
                .setManufacturer(0x4c)
                .setTxPower(-59)
                .build();
        assertTrue("Beacon should match region with all null identifiers", region.matchesBeacon(beacon));

philipp-fischer, 2015-10-20:
Sorry my example was wrong.
You need to create a beacon with only one id. Only call `setId1("553")` but not for the other IDs. Then the test should fail. I'm using version 2.6 as well.

davidgyoung, 2015-10-20:
@philipp-fischer, I'm still seeing a unit test pass using the code below that has one identifier for the beacon:

    testMatchesSingleIdentifer	0.003s	passed

    @Test
    public void testMatchesSingleIdentifer() {
        Region region = new Region("com.example.myapp.boostrapRegion", null, null, null);

        Beacon beacon = new Beacon.Builder()
                .setId1("533")
                .setManufacturer(0x4c)
                .setTxPower(-59)
                .build();
        assertTrue("Beacon with a single identifier should match region with all null identifiers", region.matchesBeacon(beacon));
    }

philipp-fischer, 2015-10-20:
Yes sorry you are right. Hopefully finally the real one:

It only fails if there is acutally only one identifer in the beacon:

    ArrayList<Identifier> idlist = new ArrayList<Identifier>(1);
    idlist.add(null);
    Beacon beacon = new Beacon.Builder()
                    .setIdentifiers(idlist)
                    .setManufacturer(0x4c)
                    .setTxPower(-59)
                    .build();

I noticed it, because it happens for my beacon layout which has only one identifier:

    beaconManager.getBeaconParsers().add(new BeaconParser().setBeaconLayout("m:0-0=4c,i:2-3,p:4-4"));

davidgyoung, 2015-10-21:
I don't think the code above makes sense.  The beacon itself has a single identifier with a value of null.  The library simply isn't designed for that.  The rule is that a beacon instance must have 1..n identifiers and each beacon identifier must be non-null.

Only a Region is allowed to have null identifiers in its identifier list.  In the case of Regions, a null identifier is a wildcard.  Having a wildcard on a Beacon instance doesn't make sense.

Perhaps the library needs to enforce this by throwing an `IllegalArgumentException` if you try to call `setIdentifiers()` with a null value in the list.

philipp-fischer, 2015-10-21:
But it also happens if it DOES have an ID.
That's actually the case I was trying to show, getting it wrong about 4 times :)

Try this:
    ArrayList<Identifier> idlist = new ArrayList<Identifier>(1);
    idlist.add(Identifier.fromInt(123));

    Beacon beacon = new Beacon.Builder()
            .setIdentifiers(idlist)
            .setManufacturer(0x4c)
            .setTxPower(-59)
            .build();

    Region region = new Region("com.example.myapp.boostrapRegion", null, null, null);

    Log.d(TAG, "Match region " + region + " with beacon " + beacon + " = " + region.matchesBeacon(beacon));

davidgyoung, 2015-10-22:
@philipp-fischer, made a unit test method out of that, too, but it also passes.  See below.

        ArrayList<Identifier> idlist = new ArrayList<Identifier>(1);
        idlist.add(Identifier.fromInt(123));

        Beacon beacon = new Beacon.Builder()
                .setIdentifiers(idlist)
                .setManufacturer(0x4c)
                .setTxPower(-59)
                .build();

        Region region = new Region("com.example.myapp.boostrapRegion", null, null, null);

        assertTrue("Beacon should match region", region.matchesBeacon(beacon));

I'm not sure what is wrong with your setup, but I simply can't reproduce the error you are reporting, and I've tried five times.  :)  I'm going to close this issue for now.  If you are still having trouble, please clone this project, make a new test case at the end of `RegionTest.java` that does not pass when you run `./gradlew test` (you can try my code above which does pass), then send a pull request, and I'll take a look again.  Thanks!

closed 2015-10-22 by davidgyoung