AltBeacon/android-beacon-library #306: Getting ClassNotFoundException org.altbeacon.beacon.service.StartMRData

ankitgupta100, 2015-10-20:
With the latest Android 6 update (Nexus 5), I am receiving following error on debugging - 

E/Parcel: Class not found when unmarshalling: org.altbeacon.beacon.service.StartRMData
E/Parcel: java.lang.ClassNotFoundException: org.altbeacon.beacon.service.StartRMData
E/Parcel:     at java.lang.Class.classForName(Native Method)
E/Parcel:     at java.lang.Class.forName(Class.java:324)
E/Parcel:     at android.os.Parcel.readParcelableCreator(Parcel.java:2383)
E/Parcel:     at android.os.Parcel.readParcelable(Parcel.java:2337)
E/Parcel:     at android.os.Message.readFromParcel(Message.java:571)
E/Parcel:     at android.os.Message.-wrap0(Message.java)
E/Parcel:     at android.os.Message$1.createFromParcel(Message.java:527)
E/Parcel:     at android.os.Message$1.createFromParcel(Message.java:525)
E/Parcel:     at android.os.IMessenger$Stub.onTransact(IMessenger.java:51)
E/Parcel:     at android.os.Binder.execTransact(Binder.java:453)
E/Parcel:  Caused by: java.lang.ClassNotFoundException: org.altbeacon.beacon.service.StartRMData
E/Parcel:     at java.lang.Class.classForName(Native Method)
E/Parcel:     at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
E/Parcel:     at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
E/Parcel:     at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
E/Parcel:     at java.lang.Class.classForName(Native Method) 
E/Parcel:     at java.lang.Class.forName(Class.java:324) 
E/Parcel:     at android.os.Parcel.readParcelableCreator(Parcel.java:2383) 
E/Parcel:     at android.os.Parcel.readParcelable(Parcel.java:2337) 
E/Parcel:     at android.os.Message.readFromParcel(Message.java:571) 
E/Parcel:     at android.os.Message.-wrap0(Message.java) 
E/Parcel:     at android.os.Message$1.createFromParcel(Message.java:527) 
E/Parcel:     at android.os.Message$1.createFromParcel(Message.java:525) 
E/Parcel:     at android.os.IMessenger$Stub.onTransact(IMessenger.java:51) 
E/Parcel:     at android.os.Binder.execTransact(Binder.java:453) 
E/Parcel:  Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available

Any idea what it means?

davidgyoung, 2015-10-20:
Strange.   The StartRMData class is used when you start ranging/and or monitoring to move data between the BeaconManager and the service.  

A few questions:
1. Do you know when you get this exception?  (Is it on phone boot up, app launch, etc.)
2. Does the app using the library crash when this happens?
3. Does the app using the library detect beacons?

ankitgupta100, 2015-10-20:
1. This happens just a couple of seconds after app launch.
2. The app is not crashing on my phone. Not sure about other devices.
3. The library is working fine and able to detect beacons.

I tested by opening the app with both ble scanning ON and OFF at the time of start, but I can see the same message in the logcat. It has this additional error as well, if helpful.

W/Binder: Caught a RuntimeException from the binder stub implementation.
W/Binder: android.os.BadParcelableException: ClassNotFoundException when unmarshalling: org.altbeacon.beacon.service.StartRMData
W/Binder:     at android.os.Parcel.readParcelableCreator(Parcel.java:2411)
W/Binder:     at android.os.Parcel.readParcelable(Parcel.java:2337)
W/Binder:     at android.os.Message.readFromParcel(Message.java:571)
W/Binder:     at android.os.Message.-wrap0(Message.java)
W/Binder:     at android.os.Message$1.createFromParcel(Message.java:527)
W/Binder:     at android.os.Message$1.createFromParcel(Message.java:525)
W/Binder:     at android.os.IMessenger$Stub.onTransact(IMessenger.java:51)
W/Binder:     at android.os.Binder.execTransact(Binder.java:453)

davidgyoung, 2015-10-20:
Do you have a custom service definition in your AndroidManifest.xml for the BeaconService?

ankitgupta100, 2015-10-20:
No. My app's AndroidManifest has definitions of application, my own custom services (IntentService) and activities. Nothing related to BeaconService.

davidgyoung, 2016-01-07:
Sorry, I don't see any obvious causes of this issue.  I am closing this due to inactivity and the fact that I cannot reproduce the issue.  If you need further help with this, please reopen.

messer00, 2016-02-24:
This happens to me as well. I do have my own service that runs in the background to synchronize data but nothing to do with beacons. I'm led to believe it somehow interferes with these. My stacktrace:

> Caught a RuntimeException from the binder stub implementation.
                                                                     android.os.BadParcelableException: ClassNotFoundException when unmarshalling: org.altbeacon.beacon.service.StartRMData
                                                                         at android.os.Parcel.readParcelableCreator(Parcel.java:2432)
                                                                         at android.os.Parcel.readParcelable(Parcel.java:2358)
                                                                         at android.os.Message.readFromParcel(Message.java:571)
                                                                         at android.os.Message.-wrap0(Message.java)
                                                                         at android.os.Message$1.createFromParcel(Message.java:527)
                                                                         at android.os.Message$1.createFromParcel(Message.java:525)
                                                                         at android.os.IMessenger$Stub.onTransact(IMessenger.java:51)
                                                                         at android.os.Binder.execTransact(Binder.java:453)

davidgyoung, 2016-02-24:
Since both reported cases of this are with a second custom service, it seems reasonable to suspect it is related to adding the other service.  Perhaps we could try to reproduce with the  library's reference application by adding a similar additional service.  This might help track it down

messer00, 2016-02-24:
Mine is just a simple IntentService which gets some data over network and
updates simple static array in the app. Just for your reference. It is run
with AlarmManager every half hour. It seems like this exception likes to
pop up after a few updates like this, but it isn't a rule - sometimes
happens on first update too.

messer00, 2016-02-26:
It seems to occur when I start monitoring for a UUID:

> 
> 02-26 12:20:59.667 32301-32314/XXX E/Parcel: start monitoring on id1: e2c56db5-dffb-48d2-b060-d0f5a71096e0 id2: null id3: null
> 02-26 12:20:59.667 32301-32314/XXX E/Parcel: Class not found when unmarshalling: org.altbeacon.beacon.service.StartRMData

davidgyoung, 2016-02-26:
When you start monitoring, the BeaconManager communicates with the library's scanning service, passing it serialized information about what region you want to monitor in the `StartRMData` class.  The receiving scanning service is supposed to de-serialize that passed information.  When it tries to do so, it fails to find the class for some reason, which causes the error you show.  Something must be wrong with the way the classloaders are accessing the classes from the library's AAR file when you have multiple services set up.  One thing you might try is to copy the source of the library into your application, rather than referencing it in an AAR to see if this makes the problem go away.   Not a good long-term fix, but it might help us figure out why this is happening.

maggud, 2016-02-29:
I also get the same issue, or related. I have implemented a beacon monitor that monitors a region and ranges beacons in that region upon entering a region. When I start my app and do a call to ContentResolver.requestSync(Account account, String authority, Bundle extras) (only the first call), I get the error log shown below:

```
E/Parcel: Class not found when unmarshalling: org.altbeacon.beacon.service.StartRMData
                                                                 java.lang.ClassNotFoundException: org.altbeacon.beacon.service.StartRMData
                                                                     at java.lang.Class.classForName(Native Method)
                                                                     at java.lang.Class.forName(Class.java:324)
                                                                     at android.os.Parcel.readParcelableCreator(Parcel.java:2404)
                                                                     at android.os.Parcel.readParcelable(Parcel.java:2358)
                                                                     at android.os.Message.readFromParcel(Message.java:571)
                                                                     at android.os.Message.-wrap0(Message.java)
                                                                     at android.os.Message$1.createFromParcel(Message.java:527)
                                                                     at android.os.Message$1.createFromParcel(Message.java:525)
                                                                     at android.os.IMessenger$Stub.onTransact(IMessenger.java:51)
                                                                     at android.os.Binder.execTransact(Binder.java:453)
                                                                  Caused by: java.lang.ClassNotFoundException: org.altbeacon.beacon.service.StartRMData
                                                                     at java.lang.Class.classForName(Native Method)
                                                                     at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
                                                                     at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
                                                                     at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
                                                                     at java.lang.Class.classForName(Native Method) 
                                                                     at java.lang.Class.forName(Class.java:324) 
                                                                     at android.os.Parcel.readParcelableCreator(Parcel.java:2404) 
                                                                     at android.os.Parcel.readParcelable(Parcel.java:2358) 
                                                                     at android.os.Message.readFromParcel(Message.java:571) 
                                                                     at android.os.Message.-wrap0(Message.java) 
                                                                     at android.os.Message$1.createFromParcel(Message.java:527) 
                                                                     at android.os.Message$1.createFromParcel(Message.java:525) 
                                                                     at android.os.IMessenger$Stub.onTransact(IMessenger.java:51) 
                                                                     at android.os.Binder.execTransact(Binder.java:453) 
                                                                  Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
W/Binder: Caught a RuntimeException from the binder stub implementation.
                                                                 android.os.BadParcelableException: ClassNotFoundException when unmarshalling: org.altbeacon.beacon.service.StartRMData
                                                                     at android.os.Parcel.readParcelableCreator(Parcel.java:2432)
                                                                     at android.os.Parcel.readParcelable(Parcel.java:2358)
                                                                     at android.os.Message.readFromParcel(Message.java:571)
                                                                     at android.os.Message.-wrap0(Message.java)
                                                                     at android.os.Message$1.createFromParcel(Message.java:527)
                                                                     at android.os.Message$1.createFromParcel(Message.java:525)
                                                                     at android.os.IMessenger$Stub.onTransact(IMessenger.java:51)
                                                                     at android.os.Binder.execTransact(Binder.java:453)
```

messer00, 2016-02-29:
It also seems like the library stops ranging for a while after this happens.

lhunath, 2016-04-19:
AFAIK, the bug is essentially here:

```
        @Override
        public void handleMessage(Message msg) {
            BeaconService service = mService.get();
            StartRMData startRMData = (StartRMData) msg.obj;
```

Note:
https://developer.android.com/reference/android/os/Message.html#obj

The `obj` field should not be used for non-framework-pure objects.  Please use `setData` for this purpose.

lhunath, 2016-04-19:
Likely related:
https://github.com/Estimote/Android-SDK/issues/55

davidgyoung, 2016-04-19:
Yes, it appears that the way the beacon detection data are passed from the service to the client won't work across processes and needs to be updated to work with the service as a separate process.  The original design of the library was intended to allow that, but the recommended practice and usage examples are all based on running it in the same process as the app.

Is there a reason to run the service as a separate process?  If so, how are you setting up the library such that this happens?  Can one of you create a fork of the reference application that demonstrates this problem?

lhunath, 2016-04-19:
My primary motivation was that the service does a lot of work which is blocking the main thread. It would likely be a good idea to move the Service's handler off the main looper. But just putting the service on a different process was a quick fix for me. Except - it doesn't actually work :-)

The only change is putting android:process=":beaconService" or similar on the service's manifest entry.

davidgyoung, 2016-04-19:
If you see that the service is blocking the main thread, please open a separate issue with details.  I don't want to derail this thread with a discussion of that, but all bluetooth callback handling is passed off to a different thread in the `ScanProcessor` class within `BeaconService`, so no heavy lifting should be happening on the main thread.  If this isn't working in some cases, it would be good to know.

quitequick, 2016-04-29:
I'm getting the same thing. I have ACRA - https://github.com/ACRA/acra - in my app. If I remove ACRA the issue disappears. I have no idea why but I thought it may help the debugging process. In a very hand-wavy way ACRA appears to cause services to be run as separate processes, where they are not when ACRA is not there.

TFreudi1, 2016-05-02:
I got this without a separate Service also only when I create a signed release apk.

E/AndroidRuntime: FATAL EXCEPTION: IntentService[BeaconIntentProcessor]
                                                   Process: com.pg.android.meinberglauf, PID: 24609
                                                   android.os.BadParcelableException: Parcelable protocol requires a Parcelable.Creator object called CREATOR on class org.altbeacon.beacon.service.m
                                                       at android.os.Parcel.readParcelableCreator(Parcel.java:2436)
                                                       at android.os.Parcel.readParcelable(Parcel.java:2358)
                                                       at android.os.Parcel.readValue(Parcel.java:2264)
                                                       at android.os.Parcel.readArrayMapInternal(Parcel.java:2614)
                                                       at android.os.BaseBundle.unparcel(BaseBundle.java:221)
                                                       at android.os.BaseBundle.get(BaseBundle.java:281)
                                                       at org.altbeacon.beacon.BeaconIntentProcessor.onHandleIntent(Unknown Source)
                                                       at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:66)
                                                       at android.os.Handler.dispatchMessage(Handler.java:102)
                                                       at android.os.Looper.loop(Looper.java:158)
                                                       at android.os.HandlerThread.run(HandlerThread.java:61)

davidgyoung, 2016-05-02:
@TFreudi1, are you either:

1. Running the BeaconService as a separate process with a custom service manifest declaration as @lhunath and others report above?
2. Using ARCA as @quitequick reports above?

If you are doing either of the following, it may explain why you are having this problem.  If you are not doing this, please post a copy our your AndroidManifest.xml

TFreudi1, 2016-05-02:
I dont do nothing of that 2 Points.I upload my manifest as txt file:
[manifest.txt](https://github.com/AltBeacon/android-beacon-library/files/245608/manifest.txt)

If I switch of proguard completly the error is gone.
I use this commented out release configuration now, but the apk-file is now more than 1 MB "thicker".
buildTypes {
        release {
            //minifyEnabled true
            //pseudoLocalesEnabled true
            //proguardFile 'E:/Development/AndroidStudioProjects/MeinBerglauf/app/proguard-project.txt'
            signingConfig signingConfigs.config

        }
    }
I work with Androidstudio 2.1 under Windows 7 no special things, with the SDK coming with the studio

TFreudi1, 2016-05-02:
I put this in my proguard-project.txt and changed my release-configuration back:
-keep class org.altbeacon.** { *; }
-keepclassmembers class org.altbeacon.** { *; }
That also helps, but proguard still show:
Note: org.altbeacon.beacon.service.BeaconService: can't find dynamically referenced class org.altbeacon.beacon.SimulatedScanData
Note: org.altbeacon.beacon.service.BeaconService accesses a field 'beacons' dynamically
      Maybe this is program field 'org.altbeacon.beacon.service.RangingData { java.util.Collection beacons; }'
      Maybe this is program field 'org.altbeacon.beacon.simulator.StaticBeaconSimulator { java.util.List beacons; }'

ClaudioINc, 2016-10-06:
I`m also getting this, starts when i added the Parse Server push notification service, i also tried Firebase, but same problem, if i remove the service the applications works correctly.

[http://stackoverflow.com/questions/39799846/altbeacon-issue-after-adding-parse-server-push-notification](url)

disheltCiti, 2016-10-07:
Hi!, I have the same problem, Is there a solution?

quitequick, 2017-03-03:
Second time around for me with this issue on another app I am working on with uses AltBeacon. Everything was working fine until I added a required service for Nokia HERE maps and now I get the error : 
E/Parcel: Class not found when unmarshalling: org.altbeacon.beacon.service.StartRMData
E/Parcel: java.lang.ClassNotFoundException: org.altbeacon.beacon.service.StartRMData

What's the status of this issue? I know it's been a while.

davidgyoung, 2017-03-04:
I have not investigated this, but I think there are two paths forward, both involving the  isolated-process attribute of the service declaration in the AndroidManifest.xml https://developer.android.com/guide/topics/manifest/service-element.html

Here is what the service declarations look like by default:

```
        <service android:enabled="true"
            android:exported="false"
            android:isolatedProcess="false"
            android:label="beacon"
            android:name=".service.BeaconService"
            />

        <service android:name=".BeaconIntentProcessor"
            android:enabled="true"
            android:exported="false"
            />
```

1. Test the library with overriding android:isolatedProcess="false" for the BeaconIntentProcessor and see if this makes the problem go away by preventing it from being run in an isolated manner.

2. (Much harder) Test and debug the library with overriding android:isolatedProcess="true" for both declarations, and making changes to the library so that it works.  I know that many static variables used for configuration won't work if the library is in a different process from the main application.  These would all have to be removed and converted to some other way of sharing configuration info between the main app and the services running in a different process.

@quitequick, I think you could easily try #1 by simply overriding those two declarations in your own app manifest.  Perhaps if you add  android:isolatedProcess="false" to the BeaconIntentProcessor declaration everything will start working.  (It's a dream I know!)

davidlomax, 2017-03-06:
I tried method 1. but I still get the same error.

quitequick, 2017-03-06:
I tried suggestion 1. but still get the same error.

davidgyoung, 2017-03-06:
Sorry to hear that @quitequick.  You might try suggestion 2 without any changes to the library.  When running in a different process, you won't be able to use many of the library configuration options that rely on static variables, but it may work OK with the defaults.  

I know it is possible to get this fully working with the service running with full features as a separate process configuration, it just may be a bit time consuming.  If you're lucky, you may get it working good enough for your purposes with little or no effort.

quitequick, 2017-03-06:
I tried suggestion 2. Just setting android:isolatedProcess="true" causes quite a few knock-on effects. When android:isolatedProcess="false" there are no complaints about messaging the service  before it's bound. When the process is isolated there are complaints. e.g. BackgroundPowerSaver does not check for the service being bound so any Activity that causes onResume etc. to be fired an attempt to set the scan period fails.

Also, I imagine there is a lot code out there that would break if the behaviour changed requiring certain methods to succeed only if the service is bound.

I guess it will need someone who know more about the internals of AltBeacon to look at this. But right now unless I can get a fix either AltBeacon or Nokia HERE maps will have to go - so it's a showstopper for me.

davidgyoung, 2017-03-06:
Well, nuts.  Sorry, @quitequick, I realize that this would be a showstopper for your project.  

I'm sure I could fix the library so the service runs successfully in a separate process, but it will certainly require more time than I can put into this in the near term as a volunteer effort.  If your company can sponsor the work needed, perhaps we can work something out depending on your timeline.  Feel free to contact me at tech@davidgyoung.com, if so.

davidgyoung, 2017-03-12:
Candidate fix is here #479

davidgyoung, 2017-03-17:
@ankitgupta100, @messer00, @ankitgupta100, @ClaudioINc, @TFreudi1 

Can you please try out this release and see if it fixes the problems discussed above?  I have prepared a binary aar file with instructions on how to add it to your project in place of the release version available on jCenter:

https://github.com/AltBeacon/android-beacon-library/releases/tag/2.9.2-mpb1