AltBeacon/android-beacon-library #308: Bluetooth broken after long period of scan

henriquemelissopoulos, 2015-10-22:
I left my phone scanning and after 20 hours a problem occurred:

> 10-22 19:16:07.587 8417-8417 D/BeaconParser: Parsing beacon layout: m:2-3=beac,i:4-19,i:20-21,i:22-23,p:24-24,d:25-25
> 10-22 19:16:07.589 8417-8417 D/BeaconParser: Parsing beacon layout: s:0-1=feaa,m:2-2=00,p:3-3:-41,i:4-13,i:14-19
> 10-22 19:16:07.591 8417-8417 D/BeaconParser: Parsing beacon layout: x,s:0-1=feaa,m:2-2=20,d:3-3,d:4-5,d:6-7,d:8-11,d:12-15
> 10-22 19:16:07.593 8417-8417 E/BeaconManager: Cannot contact service to set scan periods
> 10-22 19:16:09.754 8417-8417 E/CycledLeScannerForLollipop: Scan Failed
> 10-22 19:16:10.903 8417-8417 D/BluetoothLeScanner: could not find callback wrapper
> 10-22 19:16:12.952 8417-8417 E/CycledLeScannerForLollipop: Scan Failed
> 10-22 19:16:30.021 8417-8417 D/BluetoothLeScanner: could not find callback wrapper
> 10-22 19:16:32.151 8417-8417 E/CycledLeScannerForLollipop: Scan Failed
> 10-22 19:16:42.170 8417-8417 D/BluetoothLeScanner: could not find callback wrapper
> 10-22 19:16:44.312 8417-8417 E/CycledLeScannerForLollipop: Scan Failed

I'm using ranging with background mode enabled and everything was working fine before the error, now "could not find callback wrapper" and "Scan Failed" goes on forever and never find a beacon again. Already tried turn bluetooth off and on again; force close my app, uninstall and install the app but the only way to make bt scan work again is to reboot the device.

My issue is very similar with #289 and it's the second time I had in this device (Nexus 5, Android 5.1 LMY47I), I'm testing in more devices now and will report if it happens in one of them but maybe it's a frawemork issue? Tried others bt scanners apps like Locate, Beacon Scanner and Estimote and also none of them find any beacon before I restarted the phone.

AlexTrotsenko, 2015-10-28:
Just want to say, that I saw same issue on `Nexus 5` and my colleague saw something similar on `Nexus 7`

davidgyoung, 2016-01-07:
I'm closing this as a duplicate of #289, as I don't see any obvious differences to that issue.  Please put additional comments there.

closed 2016-01-07 by davidgyoung

g4gaj, 2017-09-18:
Hi, i am facing this issue: logs are:  D/BluetoothLeScanner: onClientRegistered() - status=133 clientIf=0 mClientIf=0
09-18 15:35:18.685 18752-18764/org.altbeacon.beaconreference D/BluetoothLeScanner: get call process name: org.altbeacon.beaconreference
09-18 15:35:18.688 18752-18752/org.altbeacon.beaconreference E/CycledLeScannerForLollipop: Scan failed: app cannot be registered
09-18 15:35:19.681 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 998 milliseconds
09-18 15:35:20.029 18752-18752/org.altbeacon.beaconreference E/Beacon: Send beacon time : ,hold:false
09-18 15:35:20.029 18752-18752/org.altbeacon.beaconreference E/Beacon: addDetectedBeaconsIntoMap(),hold:true
09-18 15:35:20.035 18752-18752/org.altbeacon.beaconreference E/Beacon: addDetectedBeaconsIntoMap() end,hold:false
09-18 15:35:20.681 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
09-18 15:35:20.683 18752-18752/org.altbeacon.beaconreference D/ScanHelper: Calling ranging callback
09-18 15:35:20.684 18752-18752/org.altbeacon.beaconreference D/Callback: attempting callback via global broadcast intent: ComponentInfo{org.altbeacon.beaconreference/org.altbeacon.beacon.BeaconIntentProcessor}
09-18 15:35:20.703 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 3975 millisconds.
09-18 15:35:20.704 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
09-18 15:35:20.704 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: We are already scanning and have been for 2028 millis
09-18 15:35:20.705 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 2000 milliseconds
09-18 15:35:20.705 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Scan started
09-18 15:35:20.714 18752-19613/org.altbeacon.beaconreference D/IntentHandler: got ranging data
09-18 15:35:20.714 18752-19613/org.altbeacon.beaconreference E/beacon: Beacon Detected:0
09-18 15:35:21.707 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 998 milliseconds
09-18 15:35:22.707 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Done with scan cycle
09-18 15:35:22.707 18752-18752/org.altbeacon.beaconreference D/ScanHelper: Calling ranging callback
09-18 15:35:22.708 18752-18752/org.altbeacon.beaconreference D/Callback: attempting callback via global broadcast intent: ComponentInfo{org.altbeacon.beaconreference/org.altbeacon.beacon.BeaconIntentProcessor}
09-18 15:35:22.713 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Not stopping scan because this is Android N and we keep scanning for a minimum of 6 seconds at a time. We will stop in 1965 millisconds.
09-18 15:35:22.713 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: starting a new scan cycle
09-18 15:35:22.713 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: We are already scanning and have been for 4037 millis
09-18 15:35:22.713 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Waiting to stop scan cycle for another 2000 milliseconds
09-18 15:35:22.713 18752-18752/org.altbeacon.beaconreference D/CycledLeScanner: Scan started