AltBeacon/android-beacon-library #310: In at least 2.6.1, BeaconManager only monitors most recently added Region.

TheIronMarx, 2015-10-26:
This is within the context of Eddystone monitoring.

I'm retrieving a list of Namespaces and Instances of beacons that I want to listen for out in the wild. I make that call, then start monitoring for those regions, but the only time the `MonitorNotifier` or `RangeNotifier` callbacks get invoked is when a beacon within the last `Region` added is found.

I bind the following BeaconConsumer. Note that the `Beacon` here is a custom POJO effectively holding the Namespace and Instance, not `org.altbeacon.beacon.Beacon`.
```java
private BeaconConsumer beaconConsumer = new BeaconConsumer() {
        @Override
        public void onBeaconServiceConnect() {
            beaconManager.setMonitorNotifier(monitorNotifier);

            for (Beacon beacon : beacons) {
                try {
                    beaconManager.startMonitoringBeaconsInRegion(new Region(REGION_ID, Identifier.parse(beacon.getNamespace()), Identifier.parse(beacon.getInstance()), null));
                } catch (Exception e) {
                    Log.e("auto", "BeaconService > onBeaconServiceConnect() > Error: Could not start monitoring beacon : +" + beacon.getId());
                }
            }
        }
```

My `BeaconManager` is provisioned and provided by Dagger as follows.
```java
@Provides
    @Singleton
    public BeaconManager provideBeaconManager(Application app) {
        BeaconManager manager = BeaconManager.getInstanceForApplication(app.getApplicationContext());
        manager.getBeaconParsers().add(new BeaconParser().setBeaconLayout(BeaconParser.EDDYSTONE_UID_LAYOUT));
        return manager;
    }
```

Every Region seems to be added successfully, but again, the only time my `MonitorNotifier` is called is when a beacon of the most recently added `Region`. The Namespace and Instance values are fully filled. As an example, `7b0ca750095477cb3e77` and `000000000007` respectively.

Any ideas as to what's going on here?

TheIronMarx, 2015-10-26:
A quick iteration through `beaconManager.getMonitoredRegions()` reveals all Namespaces and Instance values expected.

davidgyoung, 2015-10-26:
The problem is that you cannot use a constant value of REGION_ID for the identifier for the Region object.  That identifier is supposed to be unique with the purpose calling up previously configured Regions.  If you use the same value for multiple regions, you end up replacing pre-registered regions, which is exactly what you are experiencing.

davidgyoung, 2015-10-26:
The fact that the `beaconManager.getMonitoredRegions()` returns multiple items in this case is a bug that is misleading you.  It should return only a single entry for the most recently registered Region with that identifier.

TheIronMarx, 2015-10-26:
I see! That clears everything up. Thanks for the help.

closed 2015-10-26 by TheIronMarx