AltBeacon/android-beacon-library #313: The library is making my app freeze on Android 6.0 (Marshmallow)

FrancoSabadini, 2015-11-05:
When I start the app without starting the service that uses the beacons library it works well, but when I start the service the app has a lot of lag. This is only happening on Android 6.0 running on a Nexus 6. For other devices running Android 5.x it works well. Any idea what this could be?

davidgyoung, 2015-11-06:
Can you try an unmodified copy of the reference app or the [Locate app](https://play.google.com/store/apps/details?id=com.radiusnetworks.locate&hl=en) and see if you see the same thing?

fvink, 2015-11-06:
I've also experienced lag after denying location permission from the app. On Android Marshmallow, ACCESS_COARSE_LOCATION and ACCESS_FINE_LOCATION are needed for Bluetooth scanning in the background. If the permission is denied the app starts lagging and SecurityException will keep getting thrown.

davidgyoung, 2015-11-06:
See here for instructions on adding the permissions check needed for Android 6 to your app.

    http://altbeacon.github.io/android-beacon-library/requesting_permission.html

FrancoSabadini, 2015-11-08:
Thanks @davidgyoung and @fvink, adding the permission check made it work fine :).

closed 2015-11-08 by FrancoSabadini

eckoj433, 2016-11-21:
I am having the same issue, however it persists despite asking for permissions. Is there some other permission that might be missing that is causing this lagg, except coarse_location?