AltBeacon/android-beacon-library #320: The library is blocking network requests, slowing the network performance.

pkarc, 2015-11-25:
When the app start ranging and monitoring for beacons, the library block the network requests(Mobile or Wifi), making them too slow, I'm using volley to make the requests, but it blocks other applications that uses internet too. does anyone have this issue, or this is a isolated case?

jyz0025338, 2015-11-26:
It's not a isolated case. I've seen this issue in other projects...maybe it will be better if you increase the scan interval.

davidgyoung, 2015-11-26:
@pkarc,  on some mobile devices including the Moto G, Moto XX, Nexus 4 and Nexus 7,  the combined Bluetoot/WiFi chip prevents simultaneous operation of both radios. Scanning for beacons can make WiFi unreliable and using WiFi can make beacon detections get dropped.  What device  model are you using?

I would test with WiFi off to see if the symptoms remain. 

If you do see the same with WiFi off using mobile data, please run the reference app for this project, modify it to set `beaconManager.setDebug(true)` and attach a LogCat excerpt to this issue during a time when network requests are not working.

pkarc, 2015-11-26:
I'm testing this on Nexus 4, if I deactivate the WiFi, all works smoothly. 

So I propose.. let BeaconManager stop Bluetooth scanning, I mean expose a method that give us the control over the scanning process, this way we can stop scanning before making a network request and then make scanning available again once the request finish. 

Obviously this wont make a solution for the devices described above, but maybe its a beginning, so then we can identify those devices(Chip maybe, WiFi in use...)

So, what do you thing about this guys?

davidgyoung, 2015-11-26:
@pkarc, I think that is a fine idea, but I'm not sure it will work.  It may take some time between stopping scanning and when the WiFi connection will recover.  There actually is a way (albeit not super convenient) to make the library stop scanning.  You can simply call `beaconManager.stopRangingBeaconsInRegion(region)` and `beaconManager.stopMonitoringBeaconsInRegion(region)` for each of your regions before making the web service call.  Scanning will stop if no more regions are being ranged/monitored.

Please report back how this works.  If it works well, we could probably add a method that makes it easier to temporarily pause and later restart scanning.

davidgyoung, 2015-12-01:
@pkarc, would you be up for doing these tests?  If we can prove the strategy is viable, then I'd like to leave this issue open to drive a workaround as you suggest.

pkarc, 2015-12-01:
I did it, it works but the change is not significant as far as saw, The thing here is that I have to call ```beaconManager.startRangingBeaconsInRegion``` for each of the regions I had active.

So lets assume that the web service call takes as most 1.5s. How long takes the regionManager to stop scanning? 1, 2, 4 seconds? so the scanning is not stopping at all because I already start monitoring regions again before the regionManager "stop scanning"

davidgyoung, 2015-12-01:
The default scan interval in the foreground is 1.1 seconds.  So if you stop ranging/monitoring, the scan stop should happen within 1.1 seconds.  I would think you would want your logic to be something like the following:

1. Stop scanning/monitoring
2. Delay for some time to allow WiFi to recover -- at a minimum 1.1 seconds, for scanning to stop, but perhaps longer to allow recovery.
3. Make the web service call
4. When the web service call completes, restart scanning/monitoring

davidgyoung, 2016-01-07:
Closing because it is inactive.  @pkarc please reopen if you think the proposed workaround it viable.

closed 2016-01-07 by davidgyoung